esphome/issues #6038: ESP32-C6 ADC issue / error: 'class esphome::adc::ADCSensor' has no member named 'channel2_';

lboue, 2024-07-11:
### The problem

**ESP32-C6 ADC issue**

error: 'class esphome::adc::ADCSensor' has no member named 'channel2_'; 

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.2

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
esphome:
  name: c6-can
  friendly_name: C6 CAN

esp32:
  board: esp32-c6-devkitc-1
  flash_size: 4MB
  variant: esp32c6
  framework:
    type: esp-idf
    version: "5.2.2"
    platform_version: 6.7.0
    sdkconfig_options:
      CONFIG_OPENTHREAD_ENABLED: n
      CONFIG_ENABLE_WIFI_STATION: y
      CONFIG_USE_MINIMAL_MDNS: y
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

sensor:
  - platform: adc
    id: source_sensor
    pin: GPIO0
```

lboue, 2024-07-13:
I made a clone and commented everything about ADC2 for the test. There are still other errors preventing compilation at the moment:

[logs_c6-adc_compile.txt](https://github.com/user-attachments/files/16201033/logs_c6-adc_compile.txt)

```
Compiling .pioenvs/c6-adc-test/src/esphome/components/adc/adc_sensor.cpp.o
In file included from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/data/cache/platformio/packages/framework-espidf/components/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
In file included from src/esphome/components/adc/adc_sensor.h:11:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
src/esphome/components/adc/adc_sensor.h:85:3: error: 'esp_adc_cal_characteristics_t' does not name a type
   85 |   esp_adc_cal_characteristics_t cal_characteristics_[SOC_ADC_ATTEN_NUM] = {};
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp: In member function 'virtual void esphome::adc::ADCSensor::setup()':
src/esphome/components/adc/adc_sensor.cpp:71:54: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
   71 |                                               &this->cal_characteristics_[i]);
      |                                                      ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:69:22: error: 'esp_adc_cal_characterize' was not declared in this scope
   69 |     auto cal_value = esp_adc_cal_characterize(adc_unit, (adc_atten_t) i, ADC_WIDTH_MAX_SOC_BITS,
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:73:12: error: 'ESP_ADC_CAL_VAL_EFUSE_VREF' was not declared in this scope
   73 |       case ESP_ADC_CAL_VAL_EFUSE_VREF:
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:76:12: error: 'ESP_ADC_CAL_VAL_EFUSE_TP' was not declared in this scope
   76 |       case ESP_ADC_CAL_VAL_EFUSE_TP:
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:79:12: error: 'ESP_ADC_CAL_VAL_DEFAULT_VREF' was not declared in this scope
   79 |       case ESP_ADC_CAL_VAL_DEFAULT_VREF:
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp: In member function 'virtual float esphome::adc::ADCSensor::sample()':
src/esphome/components/adc/adc_sensor.cpp:200:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  200 |     uint32_t mv = esp_adc_cal_raw_to_voltage(sum, &this->cal_characteristics_[(int32_t) this->attenuation_]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:200:19: error: 'esp_adc_cal_raw_to_voltage' was not declared in this scope
  200 |     uint32_t mv = esp_adc_cal_raw_to_voltage(sum, &this->cal_characteristics_[(int32_t) this->attenuation_]);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:244:60: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  244 |   uint32_t mv12 = esp_adc_cal_raw_to_voltage(raw12, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_12_COMPAT]);
      |                                                            ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:244:19: error: 'esp_adc_cal_raw_to_voltage' was not declared in this scope
  244 |   uint32_t mv12 = esp_adc_cal_raw_to_voltage(raw12, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_12_COMPAT]);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:245:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  245 |   uint32_t mv6 = esp_adc_cal_raw_to_voltage(raw6, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_6]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:246:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  246 |   uint32_t mv2 = esp_adc_cal_raw_to_voltage(raw2, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_2_5]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:247:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  247 |   uint32_t mv0 = esp_adc_cal_raw_to_voltage(raw0, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_0]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/c6-adc-test/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
========================= [FAILED] Took 10.25 seconds =========================
```

lboue, 2024-07-20:
This branch from @ssieb brings a few fixes, but does not correct all the errors:
https://github.com/ssieb/esphome/commits/adc/esphome/components/adc

lboue, 2024-08-06:
@lucas2lab @jkremlacek
I've spent several hours coding and I have a static example that works but is limited to GPIO0.

ESPHome config

external_components:
  - source:
      # https://github.com/lboue/esphome/blob/adc_oneshot/esphome/components/adc
      type: git
      url: https://github.com/lboue/esphome
      ref: adc_oneshot
    components: [ adc ]
    refresh: 0s

sensor:
  - name: "ADC1_CH0 GPIO0"
    platform: adc
    pin: GPIO0
    update_interval: 15s

yin, 2025-05-12:
I am also watching this issue to be resolved. Thank you!