esphome/issues #6039: https://web.esphome.io/?pico can not connect to pico

kolokolt, 2024-07-12:
### The problem

My platform win 11 and Chrome. 
Follow the instruction from https://web.esphome.io/?pico. 
Uploading esphome-web-rp2040.uf2 to picco W. After apload completed, drive disaperes as described, but esphome can not see picco.

### Which version of ESPHome has the issue?

lates at 12 Jul 2024

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

RP2040

### Board

RAspberry PIcoo

### Component causing the issue

https://web.esphome.io/?pico

jasonehines, 2024-07-25:
I'm seeing the same thing when using Edge on my M1 MacBook Air. However if I go to the page for the esp devices and select connect, I see "Pico W (cu.usbmodem1101) - Paired".

si458, 2024-07-31:
same here, Chrome, OSX 14.5 (23F79), Intel Macbook Pro

bsimmo, 2024-08-14:
Same here  Windows 11, both Edge and Chrome with PicoW
Pico page doesn't see it.
ESP page does see
ESP logs showing
`[15:28:28][I][app:100]: ESPHome version 2024.7.3 compiled on Aug  4 2024, 21:14:19
[15:28:28][I][app:102]: Project esphome.web version 24.8.3
[15:28:28][C][logger:185]: Logger:
[15:28:28][C][logger:186]:   Level: DEBUG
[15:28:28][C][logger:188]:   Log Baud Rate: 115200
[15:28:28][C][logger:189]:   Hardware UART: USB_CDC
[15:28:28][C][wifi:599]: WiFi:
[15:28:28][C][wifi:427]:   Local MAC: 28:CD:C1:0F:5C:43
[15:28:28][C][wifi:432]:   SSID: [redacted]
[15:28:28][C][wifi:438]:   BSSID: [redacted]
[15:28:28][C][wifi:440]:   Hostname: 'esphome-web-0f5c43'
[15:28:28][C][wifi:442]:   Signal strength: 1 dB ▂▄▆█
[15:28:28][C][wifi:446]:   Channel: -1
[15:28:28][C][wifi:447]:   Subnet: 0.0.0.0
[15:28:28][C][wifi:448]:   Gateway: 0.0.0.0
[15:28:28][C][wifi:449]:   DNS1: 0.0.0.0
[15:28:28][C][wifi:450]:   DNS2: 0.0.0.0
[15:28:28][C][mdns:116]: mDNS:
[15:28:28][C][mdns:117]:   Hostname: esphome-web-0f5c43
[15:28:28][C][esphome.ota:073]: Over-The-Air updates:
[15:28:28][C][esphome.ota:074]:   Address: esphome-web.local:2040
[15:28:28][C][esphome.ota:075]:   Version: 2
[15:28:28][C][safe_mode:018]: Safe Mode:
[15:28:28][C][safe_mode:019]:   Boot considered successful after 60 seconds
[15:28:28][C][safe_mode:021]:   Invoke after 10 boot attempts
[15:28:28][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[15:28:28][W][safe_mode:028]: Last reset occurred too quickly; safe mode will be invoked in 5 restarts
[15:28:28][C][api:139]: API Server:
[15:28:28][C][api:140]:   Address: esphome-web.local:6053
[15:28:28][C][api:144]:   Using noise encryption: NO
[15:28:28][C][improv_serial:032]: Improv Serial:`

codyc1515, 2024-09-30:
Echoing the same. Copy the file to the USB. The Pico disconnects immediately thereafter. I cannot detect it on ESPHome web but can if I click to the ESP page. On the ESP page I can even see logs. But I cannot configure Wi-Fi.

Artafinde, 2024-11-15:
Same issue here.
1. Downloaded ESPhome for Pico
2. Dragged the downloaded file to the RPI-RP2 USB drive. Drive disappeared as expected.
3. But can't connect to the Pico and therefore can't enter wifi credentials

edwinspies, 2024-11-19:
I encounter the same problem as everybody else.

closed 2026-03-12 by bdraco