esphome/issues #6042: HA Crash during compile. 2024.7.0b2 with micro word.

karllinder, 2024-07-13:
### The problem

During compiling phase the complete HA crash.
This is related to the micro word, when removing micro_wake_word and voice assistant yaml the compiling is done without any issues.
Usually it´s crash during the TensorFlow compile.
Same issue was with Beta1.

```
Archiving .pioenvs/audio1/esp-idf/efuse/libefuse.a
Compiling .pioenvs/audio1/components/esp-tflite-micro/tensorflow/lite/micro/flatbuffer_utils.o
Compiling .pioenvs/audio1/components/esp-tflite-micro/tensorflow/lite/micro/memory_helpers.o
Compiling .pioenvs/audio1/components/esp-tflite-micro/tensorflow/lite/micro/micro_allocation_info.o
```

### Which version of ESPHome has the issue?

2024.7.0b2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.2

### What platform are you using?

ESP32-IDF

### Board

ESP32Dev

### Component causing the issue

microwakeword

### Example YAML snippet

```yaml
i2s_audio:
  id: audio_in
  i2s_lrclk_pin: GPIO32 #WSEL - WS pin
  i2s_bclk_pin: GPIO25 #BCLK - SCK pin

microphone:
  - platform: i2s_audio
    id: mic_1
    pdm: false
    channel: right
    i2s_audio_id: audio_in
    adc_type: external
    i2s_din_pin: GPIO26 #SD

micro_wake_word:
  models:
    - model: github://esphome/micro-wake-word-models/models/v2/hey_jarvis.json@main
#  vad:
 #  - model: github://esphome/micro-wake-word-models/models/v2/vad.json@main
  on_wake_word_detected:
    then:
      - voice_assistant.start:
          wake_word: !lambda return wake_word;

voice_assistant:
  microphone: mic_1
  use_wake_word: true
```

[... 252 lines cut ...]
Compiling .pioenvs/audio1/driver/mcpwm.o
Compiling .pioenvs/audio1/driver/sdio_slave.o
Compiling .pioenvs/audio1/driver/sdmmc_host.o
Compiling .pioenvs/audio1/driver/sdmmc_transaction.o
Compiling .pioenvs/audio1/driver/touch_sensor_common.o
Compiling .pioenvs/audio1/driver/esp32/touch_sensor.o
Compiling .pioenvs/audio1/driver/adc.o
Compiling .pioenvs/audio1/driver/esp32/adc.o
Compiling .pioenvs/audio1/driver/adc_deprecated.o
Compiling .pioenvs/audio1/driver/esp32/dac.o
Compiling .pioenvs/audio1/efuse/esp32/esp_efuse_table.o
Compiling .pioenvs/audio1/efuse/esp32/esp_efuse_fields.o
Compiling .pioenvs/audio1/efuse/esp32/esp_efuse_utility.o
Compiling .pioenvs/audio1/efuse/src/esp_efuse_api.o
Archiving .pioenvs/audio1/esp-idf/driver/libdriver.a
Compiling .pioenvs/audio1/efuse/src/esp_efuse_fields.o
Compiling .pioenvs/audio1/efuse/src/esp_efuse_utility.o
Compiling .pioenvs/audio1/efuse/src/esp_efuse_api_key_esp32.o
Compiling .pioenvs/audio1/components/esp-tflite-micro/tensorflow/lite/micro/debug_log.o
Compiling .pioenvs/audio1/components/esp-tflite-micro/tensorflow/lite/micro/fake_micro_context.o
Archiving .pioenvs/audio1/esp-idf/efuse/libefuse.a
Compiling .pioenvs/audio1/components/esp-tflite-micro/tensorflow/lite/micro/flatbuffer_utils.o
Compiling .pioenvs/audio1/components/esp-tflite-micro/tensorflow/lite/micro/memory_helpers.o
Compiling .pioenvs/audio1/components/esp-tflite-micro/tensorflow/lite/micro/micro_allocation_info.o
`

ssieb, 2024-07-13:
You probably need more RAM.  You could try restricting the compile processes to 1, but that might still not be enough.

darth-hp, 2024-07-18:
See https://github.com/esphome/issues/issues/5964#issuecomment-2236759711

closed 2024-08-15 by karllinder