esphome/issues #6045: 1Wire problem after 2024.6 update

sebsebseb1982, 2024-07-16:
### The problem

Hi,

For some time now, I have been having problems with my DS18B20 probes which no longer report temperature.

I saw several issues relating to the same problem, I,ve tried the proposed solutions:
 - [https://github.com/ssieb/esphome](https://github.com/esphome/issues/issues/5922#issuecomment-2180432464)
 - [delay between updates](https://github.com/esphome/issues/issues/5908#issuecomment-2196033203)
 - ...

But none of them seem to work... does anyone have any idea what might be causing the problem?
What have I missing ?

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.2

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: domo
  friendly_name: Domo

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "[redacted]"

ota:
  - platform: esphome
    password: "[redacted]"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Domo Fallback Hotspot"
    password: "[redacted]"

captive_portal:

# Example configuration entry
one_wire:
  - platform: gpio
    pin: 17

[... 216 lines cut ...]
[12:53:34][C][safe_mode:021]:   Invoke after 10 boot attempts
[12:53:34][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[12:53:34][C][api:139]: API Server:
[12:53:34][C][api:140]:   Address: domo.local:6053
[12:53:34][C][api:142]:   Using noise encryption: YES
**[12:53:48][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[12:53:49][W][dallas.temp.sensor:074]: 'Extérieur' - reading scratch pad failed bus reset
[12:53:49][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[12:53:55][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[12:53:55][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[12:53:55][W][dallas.temp.sensor:074]: 'Rez-de-chaussée' - reading scratch pad failed bus reset
[12:53:55][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[12:53:56][W][dallas.temp.sensor:074]: 'Etage' - reading scratch pad failed bus reset
[12:53:56][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[12:53:59][D][sensor:094]: 'Uptime Sensor': Sending state 271.92801 s with 0 decimals of accuracy
[12:54:18][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[12:54:19][W][dallas.temp.sensor:074]: 'Extérieur' - reading scratch pad failed bus reset
[12:54:19][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[12:54:25][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[12:54:25][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[12:54:25][W][dallas.temp.sensor:074]: 'Rez-de-chaussée' - reading scratch pad failed bus reset
[12:54:25][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[12:54:26][W][dallas.temp.sensor:074]: 'Etage' - reading scratch pad failed bus reset
[12:54:26][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed**
```

ssieb, 2024-07-17:
Was this ever working?  If so, how long ago?
The `[12:53:49][W][dallas.temp.sensor:074]: 'Extérieur' - reading scratch pad failed bus reset` indicates that the sensor is not responding to the presence detect pulse or else something else is pulling down the bus line.

sakrlog, 2024-07-17:
I'm facing the same problem as well. I had this Tuya [MS-103 device](https://www.expert4house.com/en/smart-home/sensors-and-detectors/temperature-and-humidity/tuya-smart-switch-wifi-temperature-and-humidity-sensor) that I flashed to use esphome. It was working fine for over 6 months. On the latest update that change dallas->1wire, the temperature stopped working. An interesting fact, When I first deployed the updated code, it was working intermediately, now it is not at all. 

`esphome:
  name: azan
  friendly_name: Azan

bk72xx:
  board: generic-bk7231t-qfn32-tuya

# Enable logging
logger:

mdns:

# Enable Home Assistant API
api:

ota:
  platform: esphome
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Azan Fallback Hotspot"
    password: "RmVSNDtwUSDH"

captive_portal:
  
text_sensor:
  - platform: libretiny
    version:
      name: LibreTiny Version

one_wire:
  - platform: gpio
    pin: 7

sensor:
  - platform: dallas_temp
    address: 0x8e3c01f096982e28
    name: "Azan Temperature"

output:
  - platform: libretiny_pwm
    id: output_led_1
    pin:
      number: P9
      inverted: true
  - platform: libretiny_pwm
    id: output_led_2
    pin:
      number: P8
      inverted: true

light:
  - platform: monochromatic
    id: light_switch_1
    output: output_led_1
  - platform: monochromatic
    id: light_switch_2
    output: output_led_2

switch:
  - platform: gpio
    id: switch_1
    name: Relay 1
    pin: P6
    on_turn_on:
      - light.turn_on: light_switch_1
    on_turn_off:
      - light.turn_off: light_switch_1
  - platform: gpio
    id: switch_2
    name: Relay 2
    pin: P24
    on_turn_on:
      - light.turn_on: light_switch_2
    on_turn_off:
      - light.turn_off: light_switch_2

binary_sensor:
  - platform: gpio
    id: binary_switch_all
    pin:
      number: P1
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1
        - switch.toggle: switch_2

status_led:
  pin:
    number: P10
    inverted: true
` 

Not sure what to do at this point.

sebsebseb1982, 2024-07-17:
> Was this ever working? If so, how long ago? The `[12:53:49][W][dallas.temp.sensor:074]: 'Extérieur' - reading scratch pad failed bus reset` indicates that the sensor is not responding to the presence detect pulse or else something else is pulling down the bus line.

Yes it was working since at least two years but it seems to be erratic since the last june updates. This night the measurements were resumed but it seems erratic :

[image]

sakrlog, 2024-07-17:
@sebsebseb1982 In my logs I was seeing "Scratch pad checksum invalid!". I unplugged the sensor and reinserted it. Now it is working. Waiting for it to stop

ltvanderkrogt, 2024-07-22:
ESPHome 2024.7.1 fixed my ds18b20 issue! I have 4 sensors on 1 wire connected to a Wemos d1 mini. 
[image]

kevindebosscher, 2024-07-22:
same problem over here since last update...
Worked for years,

sebsebseb1982, 2024-07-22:
Worked again after unplug & plug sensors

closed 2024-07-22 by sebsebseb1982

sakrlog, 2024-08-15:
@sebsebseb1982 this case shouldn't have been closed. Sorry I didn't have the time to report again. Sometimes, unplugging and plugging again, makes the temp readable, but in any case, it will stop working again after a few hours

ESP32andmore, 2024-08-15:
I report the same issue in [#6119](https://github.com/esphome/issues/issues/6119).  One-Wire/Dallas_temp stopped working after 2024.6.6 with a ESP32 arduino build.  I did get it to work two 2 weeks ago with 2024.7.4 by deleting the device, and adding it again with the original YAML reported [here](https://github.com/esphome/issues/issues/6119#issuecomment-2272984684). But Monday I recompiled and it was not working again.  I again deleted the device from the dashboard and re-added, but NOT working now. It does in fact work with another ESP32-S3 board with all versions. Samuel indicated there were no changes from 2024.6.6 to the code. The fact that it works with certain builds and not others leads one to the conclusion it might be build related.  In all cases the hardware that it fails on is our testing rig for this board that has been confirmed to be fully functional as well the target board that is a mature design. Flashing 2024.6.6 the sensors are found and highly stable.

sakrlog, 2024-08-19:
@zimbo86 this is 100% similar to the behavior I'm seeing

sysquadrat, 2024-08-24:
Same issue here for one (1) out of six dallas temp-sensors. It failed since June 20th. The sensor is directly connected like the others. esphome see's the sensor.

sakrlog, 2024-08-24:
Its interestingthat only one of the 6 is failing. Should we consider changing the senenor? > Same issue here for one (1) out of six dallas temp-sensors. It failed since June 20th. The sensor is directly connected like the others. esphome see's the sensor.

sysquadrat, 2024-08-26:
Well I ordered 5 more now. Will check the sensor on another device to see if it works there. Sorry for the trouble.

SofaSurfer76, 2024-11-29:
Hi everyone, I have 2 DS18B20 connected to my ESP dev board, and recently I updated to latest ESPHOME version with this code:

```
one_wire:
  - platform: gpio
    pin: GPIO13

#dallas:
#  - pin: 13 #18
#    update_interval: 10s

interval:
  - interval: 60s
    then:
      - component.update: camino
      - delay: 1s
      - component.update: ricircolo_camino
      - delay: 1s

sensor:
  - platform: dallas_temp
    address: 0xb2838a4b1f64ff28
    name: "Camino"
    id: 'camino'
    filters:
      - heartbeat: 10s
    on_value:
      then:
        - script.execute: avvio_camino
        - script.execute: allarme_temperatura

  - platform: dallas_temp
    address: 0x3d178f4b1f64ff28
    name: "Ricircolo camino"
    id: 'ricircolo_camino'
    filters:
      - heartbeat: 10s
    on_value:
      then:
        - script.execute: avvio_ricircolo_radiatori
        - script.execute: acqua_sanitari_con_camino
```

my sensor sometimes failed to be detected at boot, start running after a few secs/minutes, but no values read, and errors..
```
[18:16:26][W][gpio.one_wire:078]:   Found no devices!
...
...
[18:16:44][D][sensor:094]: 'Camino': Sending state nan °C with 1 decimals of accuracy
[18:16:45][D][sensor:094]: 'Ricircolo camino': Sending state nan °C with 1 decimals of accuracy
[18:16:49][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[18:16:49][W][dallas.temp.sensor:074]: 'Ricircolo camino' - reading scratch pad failed bus reset
[18:16:49][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[18:16:51][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[18:16:52][W][dallas.temp.sensor:074]: 'Camino' - reading scratch pad failed bus reset
[18:16:52][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[18:16:52][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[18:16:53][W][dallas.temp.sensor:074]: 'Ricircolo camino' - reading scratch pad failed bus reset
[18:16:53][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
```

I've checker all soldering and connections, then I ended up on this issue, but I still can't understand how to fix it/workaround, can someone be so kind to explain, since the issue is closed?

ESP32andmore, 2024-11-29:
Hi,
Do you have a pullup resistor to 3.3v connected to pin 13. ESPHome may or may not enable the internal ESP pullup, BUT it is a arbitrary weak pullup (GPIO internal pull-up and pull-down resistance value - ESP32 Forum<https://esp32.com/viewtopic.php?t=5111>) and it is outside the usable spec for one-wire.  1k to 4.7k is acceptable for one-wire (depends on the parasitic capacitive load of the wires, aka longer wire require a lower value).  Put a 4.7k pullup resistor from 3.3v to Pin13 and it will be fine.

Frank
***@***.***
www.esp32andmore.com<http://www.esp32andmore.com>

SofaSurfer76, 2024-11-29:
Thank you Frank for the quick reply, I have a 5k pullup restistor, and the cables of the sensors are about 3 meters, I will try with lower values and post the result.

Cheers,

ESP32andmore, 2024-11-29:
That should work fine!!

SofaSurfer76, 2024-12-02:
Yes, but it's not, and before the update my board did work for about 2 years. How could I consider esphome reliable with such this problem? I'm wondering why there have been this breaking change on 2024.6!!

ESP32andmore, 2024-12-02:
Sorry, I answered with no context since I just replied to your one email without looking at the thread. You don't want to get me started on the June, 2024 change to One-wire that went from a super reliable interface to a defeatured one that has all kind of issues.  I even got the "on you are using ESPHome to leverage your board sales" from the developers when on the submitted here it asks what do you love about ESPhome, and I answered "untested releases". But some of us makers do use ESPhome for products all within their license, they forgot to mention that my board sales promote their OS which is owned by for profit Nuba Casa, some developers work there, and probably will make millions when they go Public.
I did not know this board was work fine and this broke as a software update.  So now you are in the murky waters of trying to get the "HUB" version of working. If you have not fix it yet, post again and I have some steps you can try.

ssieb, 2024-12-03:
@SofaSurfer76 `reading scratch pad failed bus reset` indicates a wiring or sensor issue.
@ESP32andmore the one-wire change fixed most of the issues people were having.  It was tested by people that had been having issues and it worked for them.  Apparently some other people have setups that are a bit borderline and now don't work.  If you're building hardware, then I assume you have a logic analyzer.  I would love to get traces from a non-working setup.  But so far, I don't know of any people that have had issues that weren't resolved by fixing the hardware in some way.

ESP32andmore, 2024-12-03:
Its been working reliably now and I do agree many times its a sensor wiring issue, bad sensor or a knock-off

SofaSurfer76, 2024-12-03:
@ssieb if you say it's a wiring problem I fumly trust you and I will find out where is the problem! 💪

ssieb, 2024-12-03:
I can't say absolutely for sure, but that's the most likely issue.  Maybe come find me on the esphome discord server and we can work on it there.

SofaSurfer76, 2024-12-07:
@ssieb the problem is that my esp board does not supply 3.3V anymore, it's about 1.5V. Thanks for your advice