esphome/issues #6046: Setting MQTT log topic to null produces error

IetIesAai, 2024-07-16:
### The problem

In documentation(https://esphome.io/components/mqtt) and #3047 it is mentioned setting 
```
mqtt:
  log_topic: null
```
would disable logging via mqtt. However, when I do that and install the firmware wirelessly or attempt to retrieve the logs wirelessly, the error below appears. The upload itseld and the device do seem to work.

What might be relevant is I also disabled the api. 

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

issue not relevant to home assistant

### What platform are you using?

ESP32

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
mqtt:
  id: mqtt_broker
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  log_topic: null
```

### Anything in the logs that might be useful for us?

```txt
Traceback (most recent call last):
  File "[...]/esphome/bin/esphome", line 8, in <module>
    sys.exit(main())
  File "[...]/esphome/lib/python3.10/site-packages/esphome/__main__.py", line 1079, in main
    return run_esphome(sys.argv)
  File "[...]/esphome/lib/python3.10/site-packages/esphome/__main__.py", line 1066, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "[...]/esphome/lib/python3.10/site-packages/esphome/__main__.py", line 480, in command_logs
    return show_logs(config, args, port)
  File "[...]/esphome/lib/python3.10/site-packages/esphome/__main__.py", line 388, in show_logs
    return mqtt.show_logs(
  File "[...]/esphome/lib/python3.10/site-packages/esphome/mqtt.py", line 216, in show_logs
    topic = config[CONF_MQTT][CONF_LOG_TOPIC][CONF_TOPIC]
TypeError: 'NoneType' object is not subscriptable
```

IetIesAai, 2024-07-22:
also happens in 2024.7

spuder, 2024-10-11:
Do you get the same error with empty string? The following works for me. Notice that empty string is null, not string literal `null`

```
mqtt:
  # broker: broker.emqx.io # This server gives many TLS errors. 
  broker: 192.168.1.42
  port: 9993
  client_id: ${name}
  username: bblp
  password: 12345678
  log_topic:
  discover_ip: false
  discovery: false
  discovery_retain: false
  discovery_prefix:
  use_abbreviations: false
  topic_prefix: testtopic/1234
  ```

solarkennedy, 2024-10-11:
And empty string gives
```
  not a valid value.
  log_topic: ''
```

Note that `log_topic: null` is _not_ a string literal. It is a real null.
A string literal would look like `'null'` (quotes to force the string)

The validation for this in the code is here:
https://github.com/esphome/esphome/blob/efe4c5e3bc496d9f1f4d152c28a27e505b9f3024/esphome/components/mqtt/__init__.py#L249

But this will fail because the topic is required:
https://github.com/esphome/esphome/blob/dev/esphome/components/mqtt/__init__.py#L75

And when it does get parsed it will become empty string anyway:
https://github.com/esphome/esphome/blob/efe4c5e3bc496d9f1f4d152c28a27e505b9f3024/esphome/config_validation.py#L1181

But this is just validation. The `show_logs` function isn't using this at all, and imo we are getting too far too late if it thinks it can use mqtt:
https://github.com/esphome/esphome/blob/dev/esphome/__main__.py#L369

The reason it things we should be using mqtt is because it is the only available when we get to show_logs.
It doesn't pick `api` because you probably don't have `api:` in your config (like I don't) because the documentation says to remove it: https://esphome.io/components/mqtt.html

At this point I'm not sure what to do. Maybe it could print a friendlier error message, but fundamentally there are no show_logs options that will work. Too bad it can't use whatever method the web interface uses?

closed 2024-10-22 by jesserockz