esphome/issues #6051: SDL display fails with linker error on M1 Mac

andrein, 2024-07-17:
### The problem

I'm trying to use the ESPhome SDL display to rapidly prototype an UI on a M1 Macbook pro. 
Linking fails with the following error:
```
INFO Compiling app...
DEBUG Running:  platformio run -d .esphome/build/sdl-display -v
Processing sdl-display (build_flags: -DESPHOME_LOG_LEVEL=ESPHOME_LOG_LEVEL_DEBUG, -DSDL_BYTEORDER=4321, -DUSE_HOST, -D_THREAD_SAFE, -I/opt/homebrew/include/SDL2, -L/opt/homebrew/lib, -Wno-sign-compare, -Wno-unused-but-set-variable, -Wno-unused-variable, -fno-exceptions, -lSDL2, -lsodium, -std=c++17; lib_deps: esphome/noise-c@0.1.4; platform: platformio/native)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- noise-c @ 0.1.4 (License: MIT, Path: /Users/andrein/Projects/andrein/esphome/.esphome/build/sdl-display/.piolibdeps/sdl-display/noise-c)
|   |-- libsodium @ 1.10018.1 (License: MIT, Path: /Users/andrein/Projects/andrein/esphome/.esphome/build/sdl-display/.piolibdeps/sdl-display/libsodium)
Building in release mode
g++ -o .pioenvs/sdl-display/program .pioenvs/sdl-display/src/esphome/components/api/api_connection.o .pioenvs/sdl-display/src/esphome/components/api/api_frame_helper.o .pioenvs/sdl-display/src/esphome/components/api/api_pb2.o .pioenvs/sdl-display/src/esphome/components/api/api_pb2_service.o .pioenvs/sdl-display/src/esphome/components/api/api_server.o .pioenvs/sdl-display/src/esphome/components/api/list_entities.o .pioenvs/sdl-display/src/esphome/components/api/proto.o .pioenvs/sdl-display/src/esphome/components/api/subscribe_state.o .pioenvs/sdl-display/src/esphome/components/api/user_services.o .pioenvs/sdl-display/src/esphome/components/display/display.o .pioenvs/sdl-display/src/esphome/components/display/display_buffer.o .pioenvs/sdl-display/src/esphome/components/display/rect.o .pioenvs/sdl-display/src/esphome/components/host/core.o .pioenvs/sdl-display/src/esphome/components/host/gpio.o .pioenvs/sdl-display/src/esphome/components/host/preferences.o .pioenvs/sdl-display/src/esphome/components/logger/logger.o .pioenvs/sdl-display/src/esphome/components/logger/logger_esp32.o .pioenvs/sdl-display/src/esphome/components/logger/logger_esp8266.o .pioenvs/sdl-display/src/esphome/components/logger/logger_host.o .pioenvs/sdl-display/src/esphome/components/logger/logger_libretiny.o .pioenvs/sdl-display/src/esphome/components/logger/logger_rp2040.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_component.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_esp32.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_esp8266.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_host.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_libretiny.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_rp2040.o .pioenvs/sdl-display/src/esphome/components/network/util.o .pioenvs/sdl-display/src/esphome/components/sdl/sdl_esphome.o .pioenvs/sdl-display/src/esphome/components/socket/bsd_sockets_impl.o .pioenvs/sdl-display/src/esphome/components/socket/lwip_raw_tcp_impl.o .pioenvs/sdl-display/src/esphome/components/socket/lwip_sockets_impl.o .pioenvs/sdl-display/src/esphome/components/socket/socket.o .pioenvs/sdl-display/src/esphome/core/application.o .pioenvs/sdl-display/src/esphome/core/color.o .pioenvs/sdl-display/src/esphome/core/component.o .pioenvs/sdl-display/src/esphome/core/component_iterator.o .pioenvs/sdl-display/src/esphome/core/controller.o .pioenvs/sdl-display/src/esphome/core/entity_base.o .pioenvs/sdl-display/src/esphome/core/helpers.o .pioenvs/sdl-display/src/esphome/core/log.o .pioenvs/sdl-display/src/esphome/core/ring_buffer.o .pioenvs/sdl-display/src/esphome/core/scheduler.o .pioenvs/sdl-display/src/esphome/core/string_ref.o .pioenvs/sdl-display/src/esphome/core/time.o .pioenvs/sdl-display/src/esphome/core/util.o .pioenvs/sdl-display/src/main.o -L.pioenvs/sdl-display -L/opt/homebrew/lib .pioenvs/sdl-display/lib393/libsodium.a .pioenvs/sdl-display/lib498/libnoise-c.a -lSDL2 -lsodium
Undefined symbols for architecture arm64:
  "__crypto_generichash_blake2b_pick_best_implementation", referenced from:
      _sodium_init in libsodium.a[20](core.o)
  "__crypto_pwhash_argon2_pick_best_implementation", referenced from:
      _sodium_init in libsodium.a[20](core.o)
  "__crypto_stream_salsa20_pick_best_implementation", referenced from:
      _sodium_init in libsodium.a[20](core.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
*** [.pioenvs/sdl-display/program] Error 1
=============================================================================================================================== [FAILED] Took 0.62 seconds ===============================================================================================================================
```

The docs point out that libsodium must be installed via homebrew, but esphome seems to use the `esphome/libsodium` platformio package instead of the one installed via homebrew, see the `.pioenvs/sdl-display/lib393/libsodium.a` in the command line.

If I remove that and run the command manually, it builds successfully:
```
❯ cd .esphome/build/sdl-display/src/esphome/
❯ g++ -o .pioenvs/sdl-display/program .pioenvs/sdl-display/src/esphome/components/api/api_connection.o .pioenvs/sdl-display/src/esphome/components/api/api_frame_helper.o .pioenvs/sdl-display/src/esphome/components/api/api_pb2.o .pioenvs/sdl-display/src/esphome/components/api/api_pb2_service.o .pioenvs/sdl-display/src/esphome/components/api/api_server.o .pioenvs/sdl-display/src/esphome/components/api/list_entities.o .pioenvs/sdl-display/src/esphome/components/api/proto.o .pioenvs/sdl-display/src/esphome/components/api/subscribe_state.o .pioenvs/sdl-display/src/esphome/components/api/user_services.o .pioenvs/sdl-display/src/esphome/components/display/display.o .pioenvs/sdl-display/src/esphome/components/display/display_buffer.o .pioenvs/sdl-display/src/esphome/components/display/rect.o .pioenvs/sdl-display/src/esphome/components/host/core.o .pioenvs/sdl-display/src/esphome/components/host/gpio.o .pioenvs/sdl-display/src/esphome/components/host/preferences.o .pioenvs/sdl-display/src/esphome/components/logger/logger.o .pioenvs/sdl-display/src/esphome/components/logger/logger_esp32.o .pioenvs/sdl-display/src/esphome/components/logger/logger_esp8266.o .pioenvs/sdl-display/src/esphome/components/logger/logger_host.o .pioenvs/sdl-display/src/esphome/components/logger/logger_libretiny.o .pioenvs/sdl-display/src/esphome/components/logger/logger_rp2040.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_component.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_esp32.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_esp8266.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_host.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_libretiny.o .pioenvs/sdl-display/src/esphome/components/mdns/mdns_rp2040.o .pioenvs/sdl-display/src/esphome/components/network/util.o .pioenvs/sdl-display/src/esphome/components/sdl/sdl_esphome.o .pioenvs/sdl-display/src/esphome/components/socket/bsd_sockets_impl.o .pioenvs/sdl-display/src/esphome/components/socket/lwip_raw_tcp_impl.o .pioenvs/sdl-display/src/esphome/components/socket/lwip_sockets_impl.o .pioenvs/sdl-display/src/esphome/components/socket/socket.o .pioenvs/sdl-display/src/esphome/core/application.o .pioenvs/sdl-display/src/esphome/core/color.o .pioenvs/sdl-display/src/esphome/core/component.o .pioenvs/sdl-display/src/esphome/core/component_iterator.o .pioenvs/sdl-display/src/esphome/core/controller.o .pioenvs/sdl-display/src/esphome/core/entity_base.o .pioenvs/sdl-display/src/esphome/core/helpers.o .pioenvs/sdl-display/src/esphome/core/log.o .pioenvs/sdl-display/src/esphome/core/ring_buffer.o .pioenvs/sdl-display/src/esphome/core/scheduler.o .pioenvs/sdl-display/src/esphome/core/string_ref.o .pioenvs/sdl-display/src/esphome/core/time.o .pioenvs/sdl-display/src/esphome/core/util.o .pioenvs/sdl-display/src/main.o -L.pioenvs/sdl-display -L/opt/homebrew/lib .pioenvs/sdl-display/lib498/libnoise-c.a -lSDL2 -lsodium
[... cut ...]
[16:21:14][W][component:238]: Components should block for at most 30 ms.
```

### Which version of ESPHome has the issue?

2024.7.0

### What type of installation are you using?

pip
[... 10 lines cut ...]

```yaml
:
  name: sdl-display
  friendly_name: SDL Display

host:

logger:
  level: DEBUG

display:
  - platform: sdl
    id: screen    
    update_interval: 1s
    auto_clear_enabled: false
    show_test_card: true
    dimensions:
      width: 800
      height: 480

api:
  encryption:
    key: "xxx"
```

ngdio, 2024-07-26:
Experiencing the same issue.

clydebarrow, 2024-08-15:
Fixed by https://github.com/esphome/esphome/pull/7288

closed 2024-08-15 by clydebarrow