esphome/issues #6053: ESPHome 2024.7.0 Micro Wake Word on ESP32-S3-Box-3

NightHawkATL, 2024-07-17:
### The problem

I just updated to the latest ESPHome and I tried to update the micro wake word to what is required and I keep getting an error that it is not right. There isn't a great explanation on the website or here to get it to take. I have tried the example on the website and I get an error, I try the example in the error and I get an error.
[image]
[image]
I try to remove the micro wake word setup and I get an error that it is required.
[image]
[image]

It would be nice to have some great examples on breaking changes or links to what works for everyone or what to change before upgrading. I did read over the breaking changes last night and tried to make the change before the upgrade and it wouldn't take because the code wasn't supported in that manner yet until after the update. Then after the update, I get stuck in a loop of not being able to get it right to the liking for the add-on.

### Which version of ESPHome has the issue?

2024.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.2

### What platform are you using?

ESP32

### Board

ESP32-S3-Box-3

### Component causing the issue

Voice Assistant/Micro Wake Word

### Example YAML snippet

```yaml
see provided screen shots in description
```

### Anything in the logs that might be useful for us?

```txt
2024-07-17 10:34:44,964 INFO Running command 'esphome --dashboard run /config/esphome/esp32-s3-box-3-05ac44.yaml --device 10.0.3.96'
2024-07-17 10:34:47,328 INFO Process exited with return code 2
2024-07-17 10:34:56,303 WARNING 404 GET /static/schema/model.json (0.0.0.0) 1.55ms
2024-07-17 10:34:57,488 WARNING 404 GET /static/schema/model.json (0.0.0.0) 1.90ms
2024-07-17 10:35:01,217 WARNING 404 GET /static/schema/model.json (0.0.0.0) 1.43ms
2024-07-17 10:35:01,622 WARNING 404 GET /static/schema/model.json (0.0.0.0) 1.74ms
2024-07-17 10:35:02,178 WARNING 404 GET /static/schema/model.json (0.0.0.0) 1.32ms
2024-07-17 10:35:02,385 WARNING 404 GET /static/schema/model.json (0.0.0.0) 1.87ms
2024-07-17 10:35:02,910 WARNING 404 GET /static/schema/model.json (0.0.0.0) 1.34ms
2024-07-17 10:35:22,208 INFO 304 GET /static/schema/substitutions.json (0.0.0.0) 2.03ms
2024-07-17 10:35:38,119 INFO 200 POST /edit?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 3.64ms
2024-07-17 10:35:39,874 INFO 200 POST /edit?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 3.78ms
2024-07-17 10:35:39,909 INFO 304 GET /info?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 2.04ms
2024-07-17 10:35:40,979 INFO 101 GET /run (0.0.0.0) 1.24ms
2024-07-17 10:35:41,055 INFO Running command 'esphome --dashboard run /config/esphome/esp32-s3-box-3-05ac44.yaml --device 10.0.3.96'
2024-07-17 10:35:42,864 INFO Process exited with return code 2
2024-07-17 10:36:09,588 INFO 200 POST /edit?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 3.61ms
2024-07-17 10:36:09,618 INFO 304 GET /info?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 1.73ms
2024-07-17 10:36:10,597 INFO 101 GET /run (0.0.0.0) 0.55ms
2024-07-17 10:36:10,606 INFO Running command 'esphome --dashboard run /config/esphome/esp32-s3-box-3-05ac44.yaml --device 10.0.3.96'
2024-07-17 10:36:12,707 INFO Process exited with return code 2
2024-07-17 10:37:01,699 WARNING 404 GET /static/schema/micro_wake_word.json (0.0.0.0) 2.65ms
2024-07-17 10:37:01,795 WARNING 404 GET /static/schema/micro_wake_word.json (0.0.0.0) 1.38ms
2024-07-17 10:37:02,061 WARNING 404 GET /static/schema/micro_wake_word.json (0.0.0.0) 1.48ms
2024-07-17 10:37:02,642 WARNING 404 GET /static/schema/micro_wake_word.json (0.0.0.0) 1.69ms
2024-07-17 10:37:02,773 WARNING 404 GET /static/schema/micro_wake_word.json (0.0.0.0) 1.44ms
[... 57 lines cut ...]
2024-07-17 10:47:44,170 INFO 200 POST /edit?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 4.57ms
2024-07-17 10:47:44,200 INFO 304 GET /info?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 2.13ms
2024-07-17 10:47:45,291 INFO 101 GET /run (0.0.0.0) 1.33ms
2024-07-17 10:47:45,295 INFO Running command 'esphome --dashboard run /config/esphome/esp32-s3-box-3-05ac44.yaml --device 10.0.3.96'
2024-07-17 10:47:47,702 INFO Process exited with return code 2
2024-07-17 10:49:39,651 INFO 200 GET /devices (0.0.0.0) 5.46ms
2024-07-17 10:53:24,324 INFO 304 GET /static/schema/packages.json (0.0.0.0) 3.06ms
2024-07-17 10:53:39,963 INFO 200 POST /edit?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 4.60ms
2024-07-17 10:53:39,995 INFO 304 GET /info?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 3.34ms
2024-07-17 10:53:41,354 INFO 101 GET /run (0.0.0.0) 1.32ms
2024-07-17 10:53:41,553 INFO Running command 'esphome --dashboard run /config/esphome/esp32-s3-box-3-05ac44.yaml --device 10.0.3.96'
2024-07-17 10:53:44,315 INFO Process exited with return code 2
2024-07-17 10:54:39,645 INFO 200 GET /devices (0.0.0.0) 3.09ms
2024-07-17 10:55:00,258 INFO 200 POST /edit?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 5.03ms
2024-07-17 10:55:00,287 INFO 304 GET /info?configuration=esp32-s3-box-3-05ac44.yaml (0.0.0.0) 2.31ms
2024-07-17 10:55:01,550 INFO 101 GET /run (0.0.0.0) 1.16ms
2024-07-17 10:55:01,552 INFO Running command 'esphome --dashboard run /config/esphome/esp32-s3-box-3-05ac44.yaml --device 10.0.3.96'
2024-07-17 10:55:03,954 INFO Process exited with return code 2
```

### Additional information

As an aside and further troubleshooting, I did upgrade the board firmware as it was shown to need an update in HA. 
[image]
[image]

nwootton, 2024-07-18:
+1 for this

```
INFO ESPHome 2024.7.0
INFO Reading configuration /config/esphome/esp32-s3-box-3-5a9fc8.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

micro_wake_word: [source /data/packages/c46f54c1/wake-word-voice-assistant/esp32-s3-box-3.yaml:186]
  
  'models' is a required option for [micro_wake_word].
  
  The model parameter has moved to be a list element under the models parameter.
  model: okay_nabu
  on_wake_word_detected: 
    - voice_assistant.start: 
        wake_word: !lambda |-
          return wake_word;
```

Also tried re-installing from scratch via ESPHome.io/projects page. Same thing happens.

reisueber, 2024-07-18:
I had the same problem. I've attempted to delete my ESP32-S3-Box-3 from ESPHome and from the Devices section in Home Assistant. I flashed the ESP32-S3-Box-3 from scratch, which worked fine. My device runs now on the new version, but when I tried to adopt this device in ESPHome, I receive the same error message: "'models' is a required option for [micro_wake_word]".
I think that the generated yaml is wrong. 
I checked the YAML files once via FTP. During the time I deleted the device, it moved the esp32-s3-box YAML to the trash folder and subsequently created a new one, which is identical to the one in the trash folder.

nwootton, 2024-07-18:
I've just basically followed the same path: deleted device from HA, deleted the .yaml files in `/config/esphome`, but I then re-started HA. 

I then installed from scratch, adopted and after an excessive 800 seconds, the whole thing compiled correctly and runs. I've then tweaked the config in ESPHome to include my customisations (mostly fixed IP and changing images) and after a much more reasonible 65 seconds, it has compiled and runs fine.

My working assumption is that the compiled code (in .picoenvs?) is cached and NOT being updated with new model structure. By deleting and re-starting HA perhaps this code got dumped? Would explain the 800 second initial compile time. 

Anyway - way out of my depth here, just glad it appears to be working again.

reisueber, 2024-07-18:
@nwootton The same thing is happening to me now. Somehow, my device was adopted in ESPHome, and I had the option to update it. After three unsuccessful attempts, it eventually worked, and then it took a while to compile everything and install the update to the device.

NightHawkATL, 2024-07-18:
Looks like after a reboot, I was able to update as well. Just a note to all trying this to make sure that you remove, then reboot before updating if you run into the same issue.

EelcoAV, 2024-07-26:
On a hunch I changed the word "model" in this section:
'micro_wake_word:
  model: ' 
to "models". And the device would update again.

NightHawkATL, 2024-07-26:
> On a hunch I changed the word "model" in this section: 'micro_wake_word: model: ' to "models". And the device would update again.

I have that as well.

Clasper101, 2024-08-12:
I have just replaced my wired router with another from the same manufacturer, Mikrotik, and quite a bit of trouble setting it up to access the internet using the same settings as on my previous one. But I knew I would have issues with ESPhome devices that had specific ip addresses in their yaml files. But the only one that refused to play ball was the ESP32 S3 box 3 and the voice assistant. I added the models section, although I used 'Model' until I read the earlier post above. This still didn't work as I couldn't get rid of mapping errors when validating the new yaml file. So I am now copying the code from Github, as several starting from scratch efforts have just resulted in a blank screen. Fingers crossed.

Clasper101, 2024-08-12:
Well after copying the raw file from Github, substituting the friendly name and installing the yaml, my ESP32 S3 box 3 is up and running again.
There are still lots of yellow warning notifications during the compiling.
[image]