esphome/issues #6054: M5stack POESP32

BeFygo, 2024-07-17:
### The problem

Defining   "framework:    type: esp-idf " with the POESP32 makes it not boot on POE

When the "framework" is not defined the device boots on POE just fine but then doesnt work as BLEProxy (config just fine but nog connections)

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.6.1

### What platform are you using?

ESP32

### Board

M5STACK POESP32

### Example YAML snippet

```yaml
esphome:
  name: m5stack-bleproxy
  friendly_name: M5Stack BLEproxy

esp32:
  board: m5stack-core-esp32
#  framework:
#    type: esp-idf
    

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "***"

ota:
  - platform: esphome
    password: "***"

ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  power_pin: GPIO5  
  phy_addr: 1    

switch:
  - platform: gpio
    id: gate_relay16
    pin: GPIO16
    name: "Gate Switch 16"

#  - platform: gpio
#    id: gate_relay17
#    pin: GPIO17
#    name: "Gate Switch 17"  

esp32_ble_tracker:
  scan_parameters:
  #  interval: 1100ms
  #  window: 1100ms
    active: true

bluetooth_proxy:
  active: true

button:
  - platform: safe_mode
    id: button_safe_mode
    name: Safe Mode Boot

  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset
```

BeFygo, 2024-07-20:
I managed to get the esp32-idf framework running.
Now the issue that it doesnt bluetooth proxy, the antenna works ( tested with esphome over wifi instead of ethernet)

BeFygo, 2024-07-21:
Bluetooth proxy now works, needed to add the sensors manually

BeFygo, 2024-08-01:
```
  board: m5stack-core-esp32
  framework:
    type: esp-idf
```

flashing this combo seems to be the culprit, this way the device wont boot.

 ```
   board: esp32dev
  framework:
    type: esp-idf

```
flashing this combo seems to work, will this have the full performance and functionalities the board can offer?

klim123123, 2024-08-20:
Are you talking about this device?
[PoESP32](https://shop.m5stack.com/products/esp32-ethernet-unit-with-poe)
On m5stack website it says "does not support WiFi&BLE wireless function"

BeFygo, 2024-08-20:
> Are you talking about this device?
> [PoESP32](https://shop.m5stack.com/products/esp32-ethernet-unit-with-poe)
> On m5stack website it says "does not support WiFi&BLE wireless function"

Yes that device. 
As stock it doesn't but with some hardware adjustments it does.

WiFi and ble work.

klim123123, 2024-08-20:
> Yes that device. As stock it doesn't but with some hardware adjustments it does.
> 
> WiFi and ble work.

Could you please give us some details. A lot of people looking for this, on forums.

klim123123, 2024-09-08:
> As stock it doesn't but with some hardware adjustments it does.

@BeFygo please, share what adjustments to do to enable wifi&ble on PoESP32.
PLEASE.

BeFygo, 2024-10-06:
Hi,

Sorry for the late reply.

I made a short github page about the adjustments, hopefully its helpfull enough.

https://github.com/BeFygo/PoESP32-M5/tree/main

klim123123, 2024-10-06:
Thank you. So hardware modification is just put antenna 👍 
I'm using:
```
esp32:
  board: m5stack-core-esp32
```
framework is not defined.
and bluetooth_proxy works for me.

BeFygo, 2024-10-22:
> Thank you. So hardware modification is just put antenna 👍 I'm using:
> 
> ```
> esp32:
>   board: m5stack-core-esp32
> ```
> 
> framework is not defined. and bluetooth_proxy works for me.

If the framework is not defined, what framework is used then?

edestecd, 2024-12-26:
Arduino is default: https://esphome.io/components/esp32.html#arduino-framework