esphome/issues #6057: My PC reboots when updating esphome devices

Greali, 2024-07-18:
### The problem

Hello esphome community!

I have to warn first that maybe this problem is not related directly with esphome.
I have like 15 esphome devices added to esphome addon in Home Assistant.
Normally, after an update of esphome version, 15 messages appears telling me about the update of each device.
I click on update 1 by 1, and then wait until all updates are done.
The thing is almost since 2 months or so, my PC reboots during the process and sometimes I have to use SSH to connect and remove corrupted compiled data and then try again.
Happens with minor versions and mayor versions of esphome.
Sometimes my PC reboots randomly, but like 1 time per week, not that much.
In the windows event viewer it only says Id 41 kernel power critical error as the reason of the reboot, no more errors or blue screen.
I had RAM memory errors couple months before and I've changed the 2 modules I have (16x2) for new ones and I did Memtest86 test and shows no errors now. But maybe the problem could be related to this.
I'm desperated because of the restarts.
Do you know guys what could be the problem?
What should I test to know what the problem is?
The thing is I can replicate the situation compiling multiple code devices at the same time so it's gonna get restarted.

Thank you in advance.

### Which version of ESPHome has the issue?

2024.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7

### What platform are you using?

ESP32

### Board

Shelly PM Plus

### Component causing the issue

N/A

Greali, 2024-07-18:
I did a prime95 test. Have max 100 Cº temp and stable voltages, no restarts.

[image]

Greali, 2024-07-18:
I did another test and after the OTA successful message appeared the PC rebooted.

darth-hp, 2024-07-18:
I suppose this is related to #5964

Try for every affected component
```yaml
esphome:
  compile_process_limit: 1
```

or set compile_process_limit in the addon itself

Greali, 2024-07-18:
> I suppose this is related to #5964
> 
> Try for every affected component
> 
> ```yaml
> esphome:
>   compile_process_limit: 1
> ```
> 
> or set compile_process_limit in the addon itself

Thank you very much!
I'll try. I think I have some problem related with the RAM so that's why is crashing the whole PC.

How can I set compile_process_limit in the addon itself?

ssieb, 2024-07-19:
It's in the addon settings.

Greali, 2024-07-23:
> It's in the addon settings.

Thank you, I've found the setting.
For now I'm still getting reboots when updating the devices, almost each time.
When the PC is on idle it's not getting rebooted. I'm still investigating.

Greali, 2024-11-15:
> Maybe you have connected an ESP to the server and when the esp reboots disconnects the server from electricity?

Wow, that was the issue, I've installed a shelly 2 months ago to measure the power consumption and when I update the shelly the relay turn off and on causing the PC turning off :(
I'll try to use reboot_timeout: 0s in order to avoid the reset that the shelly does normally when not have WiFi or API connection.
Thank you!

closed 2024-11-15 by Greali