esphome/issues #6059: M5Stack Atom Echo

lufetico, 2024-07-19:
### The problem

A have a M5Stack Atom Echo as a device to control Home Assistant with my voice. Project Voice Assistant (https://esphome.io/projects/)

A while ago I updated the firmware of my M5Stack Atom Echo to the latest version available (2024.7.0) and I thought everything was broken but the problem is that now I have to talk very loud for the device to hear me and wake up. Is there any way to increase the microphone sensitivity? I have to speak practically glued to it or it won't hear me.

I have no technical knowledge of ESPHome. What information can I provide to help resolve the problem?

### Which version of ESPHome has the issue?

2024.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.2

### What platform are you using?

ESP32

### Board

M5Stack Atom Echo

HaraldGithub, 2024-07-19:
Same problem here with the same system configuration.

niklasmoeller, 2024-07-19:
Same for me. I have to get very close for it to pick up my wake word. But the voice command actually still works, when i'm further away.

niklasmoeller, 2024-07-20:
Little update for you guys. I talked to Jesse in their discord and he told me it will be fixed again in 2024.7.1. This is the MR that does the trick: https://github.com/esphome/esphome/pull/7109

If you can't wait until then: Just edit the file yourself (if you use docker, you can just open a console inside the docker container and edit the file there inside /esphome/esphome/...), clear the build and reinstall. Works for me now. :)

HaraldGithub, 2024-07-20:
@niklasmoeller 

Thank you for this information.
BTW: Have you also talked to Jesse about this issue: #6061  ?

niklasmoeller, 2024-07-20:
No, but there's a workaround in another issue. I'll mention it in 6061.

HaraldGithub, 2024-07-20:
Yes, i know.
The partial rollback works for me too, but with the next ESP-Update, the problem will occur again.

reflectivecode, 2024-07-20:
Using that PR fixed the problem I had with my Atom Echos were voice activity detection wasn't correctly detecting the end of speech.

```
external_components:
  - source: github://pr#7109
    components:
      - i2s_audio
```

ucdailoi, 2024-07-22:
Thanks for the fix in 2024.7.1

The issue I'm now seeing is a very loooonnnngggg wait when it's receiving the response, and this its spat out in the logs of the M5Stack.

54 seconds of waiting then "Cannot receive audio, buffer is full"

> [20:16:00][D][voice_assistant:504]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[20:16:54][D][voice_assistant:627]: Event Type: 4
[20:16:54][D][voice_assistant:655]: Speech recognised as: "Turn off the palm light."
[20:16:54][D][voice_assistant:627]: Event Type: 5
[20:16:54][D][voice_assistant:660]: Intent started
[20:16:54][D][voice_assistant:627]: Event Type: 6
[20:16:54][D][voice_assistant:627]: Event Type: 7
[20:16:54][D][voice_assistant:683]: Response: "Turned off the switch"
[20:16:54][D][light:036]: 'M5Stack Atom Echo 0ae870' Setting:
[20:16:54][D][light:051]:   Brightness: 100%
[20:16:54][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[20:16:54][D][light:109]:   Effect: 'None'
[20:16:54][D][voice_assistant:627]: Event Type: 98
[20:16:54][D][voice_assistant:770]: TTS stream start
[20:16:54][D][esp-idf:000][speaker_task]: I (361606) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=8
[20:16:54]
[20:16:54][D][i2s_audio.speaker:206]: Started I2S Audio Speaker
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][E][voice_assistant:806]: Cannot receive audio, buffer is full
[20:16:54][D][voice_assistant:627]: Event Type: 99
[20:16:54][D][voice_assistant:778]: TTS stream end
[20:16:54][D][voice_assistant:375]: End of audio stream received
[20:16:54][D][voice_assistant:504]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[20:16:54][D][voice_assistant:510]: Desired state set to RESPONSE_FINISHED
[20:16:55][D][i2s_audio.speaker:210]: Stopping I2S Audio Speaker
[20:16:55][D][i2s_audio.speaker:222]: Stopped I2S Audio Speaker

reflectivecode, 2024-07-22:
The long response time could just be how long your text-to-speech pipeline takes. Are you running Piper locally? I switched to "Google Translate text-to-speech" because Piper was unbearably slow on my hardware.

As for the "Cannot receive audio, buffer is full" error, I occasionally see that error accompanied by garbled audio output. Somewhere I saw someone recommend using a higher-quality/higher-power USB power adapter to fix that problem. I tried that myself and had better results, but I can't confirm that the power adapter was the only change that contributed to the improvement.

asherw, 2024-07-23:
(sorry that was me posting as ucdailoi when I was visiting) The full pipeline completed successfully but it was having trouble getting the response audio and playing it.

I've moved to using Micro Wake Word and that is working a lot smoother.  Will tell if its stable over the next few days.

MWW example here: https://github.com/esphome/firmware/pull/242

lufetico, 2024-07-23:
> Little update for you guys. I talked to Jesse in their discord and he told me it will be fixed again in 2024.7.1. This is the MR that does the trick: [esphome/esphome#7109](https://github.com/esphome/esphome/pull/7109)

I do not have the knowledge to make this change manually. Is it possible to downgrade the firmware?

lufetico, 2024-07-26:
I upgraded to the new firmware yesterday and the problem seems to have been fixed. Thank you very much!

qmiloo, 2024-08-21:
Hey, unfortunately with 2024.7.3 have still issue with low sensitivity. Have to almost put it to my mouth to get this work ;(

fredebec, 2024-08-22:
I have also the same issue with 2024.7.3 : low sensitivity with the wake word, but voice command is OK.

sharoon65, 2024-09-20:
I also have this issue! I just set up my atom echo today and I have to be very close to the device to pickup the wake word. Has anyone solved this issue yet?

palKrzysiek, 2024-12-06:
same problem here. low sensitivity with the wake word, but voice command is OK

janbrueggenjuergen, 2025-09-30:
I noticed the same issue and you can inspect the recorded sound samples by inserting:

> assist_pipeline:
> debug_recording_dir: /share/assist_pipeline

into your: /homeassistant/configuration.yaml

They were dead quiet to me with a faint command altough i was sitting 1m from the atom echo.
The current solution i use that works okish is to alter the mic and voice components from the default config to boost the signals. This also amplifies noise and im suspecting a hardware bug or something still wrong with the firmware as the quality is rather bad compared to other diy mems microphones i deployed.

The changes i used:

> voice_assistant:
>   id: va
>   microphone:
>     microphone: echo_microphone
>     channels: 0
>     gain_factor: 64
>   noise_suppression_level: 4
>   auto_gain: 0dBFS
>   volume_multiplier: 6

you can also alter the wakeword detection, im currently using a custom alexa one:

> micro_wake_word:
>   id: mww
>   vad:
>     probability_cutoff: 0.50
>   models:
>     # CUSTOM
>     - model: https://github.com/kahrendt/microWakeWord/releases/download/v2.1_models/alexa.json
>       id: alexa
>       probability_cutoff: 0.50
>       sliding_window_size: 5