esphome/issues #6060: Unable to Compile LibreTiny on ESPHome 2024.7.0

mattjacobson6, 2024-07-19:
### The problem

Hi - since updating to 2024.7.0, I cannot compile any LibreTiny devices.  Appears to be an issue with installing dependencies associated to LibreTiny - all other ESP-based devices compile fine.

### Which version of ESPHome has the issue?

2024.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

BK72XX

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.7.0
INFO Reading configuration /config/esphome/master-bath-shower-fan.yaml...
INFO Detected timezone 'America/New_York'
INFO Generating C++ source...
INFO Compiling app...
Processing [XXXXX] (board: generic-bk7231n-qfn32-tuya; framework: arduino; platform: https://github.com/libretiny-eu/libretiny.git)
--------------------------------------------------------------------------------
LibreTiny: Installing Python dependencies
Collecting ltchiptool<5.0,>=4.5.1
  Downloading ltchiptool-4.11.1-py3-none-any.whl.metadata (4.1 kB)
Collecting bitstruct<9.0.0,>=8.1.1 (from ltchiptool<5.0,>=4.5.1)
  Downloading bitstruct-8.19.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (7.4 kB)
Collecting bk7231tools<3.0.0,>=2.0.0 (from ltchiptool<5.0,>=4.5.1)
  Downloading bk7231tools-2.0.2-py3-none-any.whl.metadata (790 bytes)
Collecting click<9.0.0,>=8.1.3 (from ltchiptool<5.0,>=4.5.1)
  Downloading click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting colorama<0.5.0,>=0.4.5 (from ltchiptool<5.0,>=4.5.1)
  Downloading colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Collecting hexdump<4.0,>=3.3 (from ltchiptool<5.0,>=4.5.1)
  Downloading hexdump-3.3.zip (12 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      /tmp/pip-install-ld8ep3y2/hexdump_73729c9cfd5c46ca9c1c3066ef722551/setup.py:46: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
        from distutils.core import setup
      usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
         or: setup.py --help [cmd1 cmd2 ...]
         or: setup.py --help-commands
         or: setup.py cmd --help
      
      error: invalid command 'egg_info'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
*** Error 1
ModuleNotFoundError: No module named 'ltchiptool':
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/libretiny/builder/main.py", line 29:
    env.SConscript("utils/cores.py", exports="env")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/libretiny/builder/utils/cores.py", line 5:
    from ltchiptool import Family
========================== [FAILED] Took 7.37 seconds ==========================
```

Cossid, 2024-07-22:
I have had no issue compiling LibreTiny from 2024.7.0, including after cleaning and re-downloading all dependencies.

I'm guessing you'll need to clean out your cache and try again.

It does appear you're specifically using the dev version instead of recommended?  However, I was also able to compile against dev.

stefan-lsx, 2024-07-25:
I faced the same issue, but I was doing some networking changes at about the same time and turned off internet access for HA/Esphome machine, so it couldn't download new package and  proceed with build