esphome/issues #6063: 2024.7.0 fatal error: uptime_sensor.h

djrm05, 2024-07-20:
### The problem

Unable to upgrade to 2024.7.0 as it seems that uptime_sensor.h component is missing

### Which version of ESPHome has the issue?

2024.7.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_sensor.h: No such file or directory
```

djrm05, 2024-07-20:
Not ESPHOME issue

closed 2024-07-20 by djrm05

Tadimsky, 2024-07-21:
For others who find this issue searching for this error, this error is from this component: https://github.com/echavet/MitsubishiCN105ESPHome
And it's fixed in v1.3.6