esphome/issues #6069: ESP32-POE-ISO-EA boot looping and not responding to ESPHome adoption.

lightmaster, 2024-07-22:
### The problem

My Olimex ESP32-POE-ISO-EA is boot looping after updating to 2024.7.0. I initially updated through ESPHome and it never came back online. I then tried to flash it from scratch again using https://esphome.io/projects/?type=bluetooth and even with a fresh flash from the site the console shows it's boot looping. Home Assistant can see the Bluetooth Proxy as a device it suggests adding, and ESPHome can see it on the network, however when either of them try to connect it never responds. 

### Which version of ESPHome has the issue?

2024.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.3

### What platform are you using?

ESP32

### Board

Olimex ESP32-POE-ISO-EA

### Component causing the issue

Unknown

### Example YAML snippet

```yaml
Occurs with a fresh install from https://esphome.io/projects/?type=bluetooth, without having configured any YAML through the ESPHome addon.
```

### Anything in the logs that might be useful for us?

```txt
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6652
ho 0 tail 12 room 4
load:0x40078000,len:15052
load:0x40080400,len:3836
entry 0x4008069c
I (29) boot: ESP-IDF 4.4.7 2nd stage bootloader
I (29) boot: compile time 08:59:26
I (29) boot: Multicore bootloader
I (32) boot: chip revision: v3.1
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (74) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (81) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (89) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (96) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (104) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=44734h (280372) map
I (218) esp_image: segment 1: paddr=0005475c vaddr=3ffbdb60 size=04bech ( 19436) load
I (226) esp_image: segment 2: paddr=00059350 vaddr=40080000 size=06cc8h ( 27848) load
I (237) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=dacf8h (896248) map
I (562) esp_image: segment 4: paddr=0013ad20 vaddr=40086cc8 size=12210h ( 74256) load
I (605) boot: Loaded app from partition at offset 0x10000
[... 121 lines cut ...]
I (607) cpu_start: App cpu up.
I (637) cpu_start: Pro cpu start user code
I (637) cpu_start: cpu freq: 160000000
I (637) cpu_start: Application information:
I (642) cpu_start: Project name:     olimex-bluetooth-proxy
I (648) cpu_start: App version:      2024.7.0
I (653) cpu_start: Compile time:     Jul 17 2024 08:58:27
I (659) cpu_start: ELF file SHA256:  5e741f992656a0dd...
I (665) cpu_start: ESP-IDF:          4.4.7
I (670) cpu_start: Min chip rev:     v0.0
I (674) cpu_start: Max chip rev:     v3.99 
I (679) cpu_start: Chip rev:         v3.1
I (684) heap_init: Initializing. RAM available for dynamic allocation:
I (691) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (697) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (703) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (709) heap_init: At 3FFCAAB0 len 00015550 (85 KiB): DRAM
I (716) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (722) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (728) heap_init: At 40098ED8 len 00007128 (28 KiB): IRAM
I (736) spi_flash: detected chip: generic
I (739) spi_flash: flash io: dio
I (746) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

kblin, 2024-07-22:
I have a very similar-looking issue on 2024.7.1 with an esp32-based everything presence 1. Lewis, the EP1 creator, mentioned that moving the device to a network with internet access works around the bug, and I can confirm that this works for me.

lightmaster, 2024-07-22:
@kblin I don't let devices have internet access that don't need it. Do you know the specific URL or IP that it needs to access to be able to function?

lightmaster, 2024-07-22:
I tried moving the ethernet port to the untagged LAN network so it has full internet access, flashing it from scratch with the ESPHome site as a Omilex Bluetooth Proxy, adopting it to ESPHome addon, and it still will not connect to HA as a Bluetooth Proxy.

I was able to get it working by flashing it using the June 20th yaml config at https://github.com/esphome/firmware/blob/87a7b4b54ecd6588b7942417230bfdb8fdbad113/bluetooth-proxy/olimex-esp32-poe-iso.yaml and not adopting it into ESPHome so it didn't get updated.

jesserockz, 2024-07-22:
I am looking into this today. I personally have an NoT network and I am not getting any crashes, just graceful failures so it is a little harder to try find the exact source of the bug.

```
[08:38:47][D][esp-idf:000]: E (6894) esp-tls: couldn't get hostname for :everythingsmarthome.github.io: getaddrinfo() returns 202, addrinfo=0x0
[08:38:47]
[08:38:47][D][esp-idf:000]: E (6894) esp-tls: Failed to open new connection
[08:38:47]
[08:38:47][D][esp-idf:000]: E (6903) TRANSPORT_BASE: Failed to open a new connection
[08:38:47]
[08:38:47][D][esp-idf:000]: E (6913) HTTP_CLIENT: Connection failed, sock < 0
[08:38:47]
[08:38:47][E][component:164]: Component http_request set Error flag: unspecified
[08:38:47][E][http_request.idf:085]: HTTP Request failed: ESP_ERR_HTTP_CONNECT
[08:38:47][E][component:164]: Component http_request.update set Error flag: Failed to fetch manifest from https://everythingsmarthome.github.io/everything-presence-one/everything-presence-one-manifest.json
```

jesserockz, 2024-07-22:
Ok the issue is the default timeout of the `http_request` is set to 5s which is the same as the wdt timeout so it is getting caught there if the firewall drops packets (mine was set to reject instead of drop but changing it has allowed me to reproduce)

closed 2024-07-23 by kbx81

lightmaster, 2024-07-25:
Just to follow up, I updated the addon to 2024.7.2 this morning and I set up the BL Proxy from scratch again using the ESPHome website. Everything went smoothly and it's working fine. I saw the message in the logs about not being able to connect to `firmware.esphome.io` so I did give it access to that domain, though it did work beforehand too.

Thanks for jumping on this so quickly.