esphome/issues #6071: SML Crashing ESP8266 after 2024.7.1

Leopoldpaul, 2024-07-22:
### The problem

After 2024.7.1, my ESP8266 used for monitoring my Smartmeter is constantly rebooting, and it is caused by the SML component because if I disconnect the SML reader, the ESP stops rebooting. Unfortunately, I can't log some errors because the ESP is constantly rebooting. 

### Which version of ESPHome has the issue?

2024.7.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

d1

### Component causing the issue

SML

### Example YAML snippet

```yaml
uart:
  id: uart_bus
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600

sml:
  id: mysml
  uart_id: uart_bus

sensor:
  - platform: sml
    name: "Zahlerstand"
    sml_id: mysml
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0000001

  - platform: sml
    name: "Zahlerstand HT"
    sml_id: mysml
    obis_code: "1-0:1.8.1"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0000001

  - platform: sml
    name: "Zahlerstand NT"
    sml_id: mysml
    obis_code: "1-0:1.8.2"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0000001

  - platform: sml
    name: "Leistung"
    sml_id: mysml
    obis_code: "1-0:16.7.0"
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: power
    state_class: measurement
    filters:
      - multiply: 0.01

  - platform: sml
    name: "Leistung L1"
    sml_id: mysml
    obis_code: "1-0:36.7.0"
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: power
    state_class: measurement
    filters:
      - multiply: 0.01

  - platform: sml
    name: "Leistung L2"
    sml_id: mysml
    obis_code: "1-0:56.7.0"
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: power
    state_class: measurement
    filters:
      - multiply: 0.01

  - platform: sml
    name: "Leistung L3"
    sml_id: mysml
    obis_code: "1-0:76.7.0"
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: power
    state_class: measurement
    filters:
      - multiply: 0.01
```

### Additional information

2024.7.0 was working fine.

TWuerdemann, 2024-07-22:
Same issue here. Although my YAML is a bit more complex, because I watch 2 different smartmeters (one with an external lib and one with the SML component).

Nethertheless after I installed the backup of 2014.7.0 it worked again.

awptechnologies, 2024-07-24:
I am Rebooting over and over again as well. Started out of no where. on latest image of esphome

Leopoldpaul, 2024-07-25:
Version 2024.7.2 did not fix the issue.

eNBeWe, 2024-08-08:
I have been banging my head against the sml parser code for two days now, and I think I found the culprit.
I will test with my reader tomorrow and then send a PR.

closed 2024-08-12 by jesserockz