esphome/issues #6072: Uptade ota failed

SebastienMrc, 2024-07-22:
### The problem

Hello, since the update of ESPHome 2024.6 I can no longer update my ESPs... 
Currently, all my ESPs is in version 2024.5.5 and it is impossible for me to update them or modify their codes and upload it.

When I try to update, I have the following error messages:
```
INFO ESPHome 2024.7.1
INFO Reading configuration /config/esphome/esp-m1-lessive.yaml...
Failed config

ota.unknown: [source /config/esphome/esp-m1-lessive.yaml:16]
  
  'ota' requires a 'platform' key but it was not specified.
  password: xxXXxXxXxxxxxXxxXXxXxXxxxXXXXxxX
```

And : 

"Failure to call the update/install service. Error compiling esp-m6-assouplissant.yaml; Try again in ESPHome dashboard for more information."

### Which version of ESPHome has the issue?

2024.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.7.3 Supervisor 2024.06.2 Operating System 12.4 Interface utilisateur 20240710.0

### What platform are you using?

ESP8266

### Board

d1_mini

### Example YAML snippet

```yaml
esphome:
  name: esp-m1-lessive
 
esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "97sXd+D/+pI8pLfkN9LpIYBHXWDtGeAeKw7rniBlQkU="

ota:
  password: "xxXXxXxXxxxxxXxxXXxXxXxxxXXXXxxX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-M1-Lessive Fallback Hotspot"
    password: "Rk51rqCFTVE8"
```

### Anything in the logs that might be useful for us?

```txt
'ota' requires a 'platform' key but it was not specified.
```

randybb, 2024-07-22:
The error is telling you where the problem is. OTA component needs platform, in your case https://esphome.io/components/ota/esphome . Next time please read release notes and check all breaking changes first, in this case it is this one https://esphome.io/changelog/2024.6.0#ota-platforms

closed 2024-07-22 by randybb

SebastienMrc, 2024-07-22:
Thank you very much for your help, indeed it was good!