esphome/issues #6075: Mark Modbus Sensors as Unavailable When Slave is Offline

iezhkv, 2024-07-22:
### The problem

I am using ESPHome to integrate Modbus sensors. Currently, there is no built-in functionality to mark Modbus sensors as "unavailable" when the Modbus slave is offline or not responding. This makes it difficult to identify when the sensors are not updating due to a slave issue.

I have multiple Modbus sensors configured in my ESPHome setup. When the Modbus slave goes offline or fails to respond, the sensors still display the last known value, which can be misleading. I have tried using lambda functions and interval checks to manually set the sensor state to NaN, but this approach is not reliable and adds complexity to the configuration.

I propose adding a feature to the Modbus integration in ESPHome that would automatically mark sensors as unavailable when the Modbus slave is offline. This could be achieved by:

Introducing a status check within the Modbus component that sets the sensor state to NaN when no response is received after a certain number of retries.
Providing a configuration option to enable or customize this behavior.
This feature would greatly enhance the usability and reliability of Modbus integrations in ESPHome.

### Which version of ESPHome has the issue?

2024.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

Olimex ESP-POE

### Component causing the issue

Modbus

### Example YAML snippet

```yaml
uart:
  id: uart_modbus_client
  tx_pin: 4
  rx_pin: 36
  baud_rate: 9600

modbus:
  uart_id: uart_modbus_client
  id: modbus_client

modbus_controller:
  - id: modbus_slave
    modbus_id: modbus_client
    address: 0x1
    update_interval: 1s
    offline_skip_updates: 1
    

sensor:
  - platform: modbus_controller
    modbus_controller_id: modbus_slave
    name: "Slave Temperature"
    id: slave_temp
    register_type: holding
    address: 0x0001
    value_type: FP32
    unit_of_measurement: "°C"

# Read humidity from Modbus slave
  - platform: modbus_controller
    modbus_controller_id: modbus_slave
    name: "Slave Humidity"
    id: slave_hum
    register_type: holding
    address: 0x0002
    value_type: FP32
    unit_of_measurement: "%"
```

ssieb, 2024-07-22:
There is no way to mark a sensor as unavailable. https://github.com/esphome/feature-requests/issues/1568
You could use a timeout filter to change the value to nan, which will make it "unknown" in HA.

closed 2024-07-22 by ssieb

nagyrobi, 2024-10-13:
Fr: https://github.com/esphome/feature-requests/issues/2920