esphome/issues #6077: 1-Wire not working with Nodemcu ESP8266. Getting: "Scratch pad checksum invalid!"

WasuMrTomass0, 2024-07-24:
### The problem

Hi experts,

I am facing an issue with 1-Wire and dallas_temp sensors. I have created a YAML config to read data from two temperature sensors DS18B20 (waterproof). At this very moment, I receive warnings from both sensors:

```log
[W] [component:170]          Component dallas_temp.sensor cleared Warning flag
[W] [dallas.temp.sensor:139] 'Temperatura solary' - Scratch pad checksum invalid!
[W] [component:157]          Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
```
My current config in this scope is:
```yaml
esp8266:
  board: nodemcuv2

# Other

one_wire:
  - platform: gpio
    pin: D2
    id: one_wire_pin_D2

sensor:
  
  # Temperature sensor #1
  - platform: dallas_temp
    name: "Temperatura bojler"
    address: 0x281AA646D4606CD7
    resolution: 9
    update_interval: 10s
    one_wire_id: one_wire_pin_D2
    unit_of_measurement: "°C"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 2
    filters:
    - delta: 0.25
    on_value_range:
      - above: 80.0
        then:
          - switch.turn_off: switch_bojler
    
  # Temperature sensor #2
  - platform: dallas_temp
    name: "Temperatura solary"
    address: 0x2880BF46D4B17AFA
    resolution: 9
    update_interval: 10s
    one_wire_id: one_wire_pin_D2
    unit_of_measurement: "°C"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 2
    filters:
    - delta: 0.25
```

Worth noting: 
* I am aware of issues in 2024.6 related to 1-wire. I have tried to use multiple versions of ESPHome (2024.5.5, 2024.6, 2024.7.1) - all of them where having issues.
* At first I tried to get readings from 1-Wire bus to read sensor adresses. There were no log messages related to 1-Wire. Even "no devices found" was not printed.
* I have used Arduino Nano to read adresses of my sensors (running on 5V). I used those adresses in my ESP config. Then I got error messages shown above.
* When sensor was disconected from D2 pin (1-Wire bus) message was a bit different - something about reseting the bus.

**What I am looking for:**
* Is there any ongoing issue with 1-Wire component? or is it all fixed in current version - I was able to find issues only related to 2024.6.x versions. 
* I have confirmed that sensors are working OK with Arduino Uno (5V). Is there a chance it is not working correctly on 3.3V? (Datasheet said 3.3V is ok)
* Do you have any ideas what could be causing this?

If this summary should contain more information, please let me know. I will try to fix it.
Thank you for you time (in advance)

### Which version of ESPHome has the issue?

 2024.7.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Nodemcu v3

### Component causing the issue

1-Wire

### Example YAML snippet

```yaml
# YAML variables. Dereferenced with $
substitutions:
  esphome_name: esp-kotlownia
  esphome_friendly_name: ESP Kotłownia

# Main settings
esphome:
  name: $esphome_name
  friendly_name: $esphome_friendly_name
  # # Automatically add the mac address to the name
  # # so you can use a single firmware for all devices
  # name_add_mac_suffix: true

  # This will allow for (future) project identification,
  # configuration and updates.
  project:
    name: esphome.$esphome_name
    version: "1.1"
  on_boot:
    priority: 600
    then:
      - script.execute: switch_bojler_on_period_from_ha_number

esp8266:
  board: nodemcuv2

# To be able to get logs from the device via serial and api.
logger:
  level: DEBUG

# Enable Web server.
web_server:
  port: 80

# API is a requirement of the dashboard import.
api:
  encryption:
    key: !secret esphome_api_encrypt_key
  reboot_timeout: 15min
  services:
    - service: bojler_on_for_period
      then:
        - script.execute: switch_bojler_on_period_from_ha_number
    - service: bojler_on_for_15min
      then:
        - script.execute:
            id: switch_bojler_on_parameter_period
            delay_minutes: 15
    - service: bojler_on_for_20min
      then:
        - script.execute:
            id: switch_bojler_on_parameter_period
            delay_minutes: 20
    - service: bojler_on_for_30min
      then:
        - script.execute:
            id: switch_bojler_on_parameter_period
            delay_minutes: 30

ota:
  - platform: esphome
    password: !secret esphome_ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $esphome_name
    password: !secret esphome_ap_password

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device.
captive_portal:

time:
  # Sync time with Home Assistant
  - platform: homeassistant
    id: homeassistant_time

switch:
  # Kontroler bojlera (który jest zabezpieczony wyłącznikiem temperatury, 
  # także nie potrzeba kontrolować temperatury wody)
  - platform: gpio
    id: switch_bojler
    pin: GPIO5
    restore_mode: ALWAYS_OFF
    name: "Bojler"
  # Restart mikrokontrolera
  - platform: restart
    name: Restart

script:
  # Script with delay parameter - low level script
  - id: switch_bojler_on_parameter_period
    mode: restart
    parameters:
      delay_minutes: int
    then:
      - if:
          condition:
            switch.is_on: switch_bojler
          # If already on, turn off
          then:
            - switch.turn_off: switch_bojler
          # If already off, turn on for some time
          else:
            - switch.turn_on: switch_bojler
            - delay: !lambda "return delay_minutes * 60 * 1000;" 
            - switch.turn_off: switch_bojler
            
    # Script with delay read from HA's number
  - id: switch_bojler_on_period_from_ha_number
    mode: restart
    then:
      - script.execute:
          id: switch_bojler_on_parameter_period
          delay_minutes: !lambda "return id(bojler_on_period_min).state;" 

number:
  - platform: template
    id: bojler_on_period_min
    name: Czas włączenia bojlera
    min_value: 1
    max_value: 60
    step: 1
    initial_value: 15
    unit_of_measurement: Min
    icon: "mdi:timer-outline"
    mode: box
    optimistic: true

one_wire:
  - platform: gpio
    pin: D2
    id: one_wire_pin_D2

sensor:
  
  # Temperature sensor #1
  - platform: dallas_temp
    name: "Temperatura bojler"
    address: 0x281AA646D4606CD7
    resolution: 9
    update_interval: 10s
    one_wire_id: one_wire_pin_D2
    unit_of_measurement: "°C"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 2
    filters:
    - delta: 0.25
    on_value_range:
      - above: 80.0
        then:
          - switch.turn_off: switch_bojler
    
  # Temperature sensor #2
  - platform: dallas_temp
    name: "Temperatura solary"
    address: 0x2880BF46D4B17AFA
    resolution: 9
    update_interval: 10s
    one_wire_id: one_wire_pin_D2
    unit_of_measurement: "°C"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 2
    filters:
    - delta: 0.25
    
    # Reports the WiFi signal strength/RSSI in dB
  - platform: wifi_signal
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 300s
    entity_category: "diagnostic"
  # Reports the WiFi signal strength in %
  - platform: copy
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
    device_class: ""
  - platform: uptime
    name: Uptime
    disabled_by_default: false
    force_update: false
    filters:
      - lambda: return x / 3600;
    unit_of_measurement: "h"
    icon: mdi:timer-outline
    accuracy_decimals: 0
    device_class: duration
    state_class: total_increasing
    entity_category: diagnostic
    update_interval: 3600s

text_sensor:
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
  - platform: version
    name: Version
    hide_timestamp: true
    disabled_by_default: false
    icon: mdi:new-box
    entity_category: diagnostic
```

ssieb, 2024-07-24:
How are you reading the logs?  There will be some message from the one_wire bus.  Either a list of found devices or saying it didn't find any.  If you use the ota logs, it will be in the config dump when you connect.
Your sensor ids are reversed.  esphome and whatever you're using on Arduino format them differently.  For esphome, the 28 should be at the end.

WasuMrTomass0, 2024-07-24:
> _How are you reading the logs?_

I read them from web page of my ESP.

> _If you use the ota logs, it will be in the config dump when you connect._

Could you point me to docs which describe this? I can't find any info in ota, nor logger section. I don't know how what is config dump - tried to google it. I can't use USB uart to read logs :c It might be helpful fot future debug.

> _Your sensor ids are reversed_

**That was it!** Thank you for your help! Big kudos to you!
Arduino Nano code was just a sample I got from internet and I didnt expect it would reverse byte order.

closed 2024-07-24 by WasuMrTomass0