esphome/issues #6082: ESP32-POE-ISO-EA boot looping

doonga, 2024-07-25:
### The problem

I tried flashing a ESP32-POE-ISO-EA as a bluetooth proxy using the web installer. The flash succeeds but then the device starts to boot loop. If I go back to the older yaml that I was using (from the old firmware repo) it starts working again. I don't see anything particularly noteworthy in the logs but I will attach them.

I have 2 of these and tried it on both with the same result so I don't believe I have a hardware problem.

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Olimex

### Example YAML snippet

```yaml
substitutions:
  name: bluetooth-proxy-office
  friendly_name: bluetooth-proxy-office

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.7.2
  name_add_mac_suffix: false
  project:
    name: esphome.bluetooth-proxy
    version: "24.7.4.1"

esp32:
  board: esp32-poe-iso
  framework:
    type: esp-idf

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12

api:
  encryption:
    key: !secret bluetooth_proxy_office_api_key
logger:
ota:
  - platform: esphome
    id: ota_esphome
  - platform: http_request
    id: ota_http_request

update:
  - platform: http_request
    id: update_http_request
    name: Firmware
    source: https://firmware.esphome.io/bluetooth-proxy/olimex-esp32-poe-iso/manifest.json

http_request:

dashboard_import:
[... 1028 lines cut ...]

api:
  encryption:
    key: !secret bluetooth_proxy_office_api_key
logger:
ota:
  - platform: esphome
    id: ota_esphome

dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/olimex-esp32-poe-iso.yaml@main

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic

ssieb, 2024-07-25:
If you can plug the device directly into what is running HA and get the serial logs that way, then esphome will decode that stack trace.

ssieb, 2024-07-27:
It's running out of RAM.

doonga, 2024-07-27:
Oh, hah ok. I'll buy the ones with more RAM on them. Thank you!

closed 2024-07-27 by doonga

ssieb, 2024-07-28:
Every ESP32 of the same type has the same amount of RAM.  I haven't checked if the newer variants have more RAM than the original ESP32.  If you're using BLE, then getting one with PSRAM might help.

doonga, 2024-07-28:
Ya you're right, I got confused with the storage space, not ram. I'll check into that, thank you again!

ashleysmithgpu, 2024-09-16:
I was having the same issue. For anyone else, you can try disabling mdns and using a static IP. This allowed just enough RAM for my board to not crash:

```
  manual_ip:
    static_ip: 192.168.1.???
    gateway: 192.168.1.1
    subnet: 255.255.255.0

mdns:
  disabled: true
```