esphome/issues #6083: LED Strip flickers randomly with light effects

deCodeIt, 2024-07-26:
### The problem

[image]

https://github.com/user-attachments/assets/5b75c7c8-2423-495d-81ae-bcc3b3bf1d44

When using a Neopixel WS2812B strip with Raspberry pi pico w running Esphome, the led strip flickers intermittently. I'm not sure what's the root cause of the issue. I tried using a different frequency (8MHz) but it didn't work either.

Is this a common issue? I also tried with a different strip and faced same issue. I used another pico w running micropython and verified that the strip works fine with it so not sure what's wrong with using via esphome.

### Which version of ESPHome has the issue?

2024.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.3

### What platform are you using?

RP2040

### Board

pico w

### Component causing the issue

rp2040_pio_led_strip

### Example YAML snippet

```yaml
light:
  - platform: rp2040_pio_led_strip
    name: "Neopixel Strip"
    id: led_strip
    pin: GPIO21
    num_leds: 60
    pio: 0
    rgb_order: GRB
    chipset: WS2812B
    effects:
      - pulse:
      - pulse:
          name: "Fast Pulse"
          transition_length: 0.5s
          update_interval: 0.5s
          min_brightness: 0%
          max_brightness: 100%
      # Use default parameters:
      - random:
      # Customize parameters
      - random:
          name: "My Slow Random Effect"
          transition_length: 30s
          update_interval: 30s
      - random:
          name: "My Fast Random Effect"
          transition_length: 4s
          update_interval: 5s
      - strobe:
      - strobe:
          name: Strobe Effect With Custom Values
          colors:
            - state: true
              brightness: 100%
              red: 100%
              green: 90%
              blue: 0%
              duration: 500ms
            - state: false
              duration: 250ms
            - state: true
              brightness: 100%
              red: 0%
              green: 100%
              blue: 0%
              duration: 500ms
      - addressable_rainbow:
      - addressable_rainbow:
          name: Rainbow Effect With Custom Values
          speed: 10
          width: 50
```

### Anything in the logs that might be useful for us?

```txt
There's no error message in the logs.
```

deCodeIt, 2024-07-26:
Tagging @jesserockz @Papa-DMan

zencow, 2024-08-18:
I was lucky enough to have a previous UF2 image for one of my Pico's.  An update a few months back had someone's "fix" for flickering, and basically made my neopixels non-function on the Pico.  Maybe their changes worked for them, but not for the rest of us.

closed 2024-08-19 by jesserockz