esphome/issues #6085: Pulse Counter Sensor on nodemcu-32s - can't disable pullup

Heiko-L, 2024-07-26:
### The problem

When using the Pulse Counter Sensor pullups are active and can't be disabled (pullup: false does not do anything either).
If I configure the same Pins as simple binary sensor pullups can be disabled:
binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO14
      mode:
        input: true
    id: pulse_counter_1
    name: ${friendly_name} Pulse Counter 1

  - platform: gpio
    pin: 
      number: GPIO27
      mode:
        input: true
    id: pulse_counter_2
    name: ${friendly_name} Pulse Counter 2

Can someone confirm this?

TIA
Heiko

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

Pulse Counter Sensor

### Example YAML snippet

```yaml
substitutions:
  friendly_name: Test
  
esphome:
  name: test

preferences:
  flash_write_interval: 15min

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

logger:
  level: WARN

api:
  encryption:
    key: !secret api_key

ota:
 - platform: esphome
   password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Tests Fallback Hotspot"
    password: !secret wifi_fallback_password

captive_portal:
    
web_server:
  port: 80

sensor:
  - platform: pulse_counter
    pin:
      number: GPIO27
      inverted: true
      mode:
        input: true
    name: ${friendly_name} RPM1
    filters:
      - multiply: 0.3333333333333
    update_interval: 20s
    
  - platform: pulse_counter
    pin:
      number: GPIO14
      inverted: true
      mode:
        input: true
    name: ${friendly_name} RPM2
    filters:
      - multiply: 0.3333333333333
    update_interval: 20s
```

ssieb, 2024-07-27:
Why do you think the pullup is enabled?

Heiko-L, 2024-07-27:
> Why do you think the pullup is enabled?

I have an external circuit with pullup, series resistor and schottky diodes and it can't be pulled down because of the enabled internal pullups (only to about 1,8V - 33k external series resistor against the 45k internal pullup).
Defining the GPIO as binary sensor everything works fine because internal pullups are disabled then.

ssieb, 2024-07-28:
There is nothing in the code that would enable the pullup.

Heiko-L, 2024-07-29:
But they are enabled.

These options I tried:
```
  - platform: pulse_counter
    pin: GPIO27
    name: ${friendly_name} RPM1
    
  - platform: pulse_counter
    pin:
      number: GPIO14
      mode:
        input: true
        pullup: false
    name: ${friendly_name} RPM2
```
```
  - platform: pulse_counter
    pin:
      number: GPIO27
      mode:
        input: true
        pullup: true
    name: ${friendly_name} RPM1
    
  - platform: pulse_counter
    pin:
      number: GPIO14
      inverted: true
      mode:
        input: true
        pullup: false
    name: ${friendly_name} RPM2
```
Then I tried to enable pulldowns and that doesn't work either:
```
  - platform: pulse_counter
    pin:
      number: GPIO27
      mode:
        input: true
        pulldown: true 
    name: ${friendly_name} RPM1
    
  - platform: pulse_counter
    pin:
      number: GPIO14
      inverted: true
      mode:
        input: true
        pulldown: true
    name: ${friendly_name} RPM2
```
The circuit (FanXTachoOut is connected to ESP32's GPIO):
[image]

With all of the above options I get ~1,8-1,9V at FanXTachoOut if FanXTachoIn is tied to ground.

Using:
```
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      mode:
        input: true
    name: ${friendly_name} Pulse Counter 1

  - platform: gpio
    pin:
      number: GPIO27
      mode:
        input: true
        pullup: false
    name: ${friendly_name} Pulse Counter 2
```
I get <2mV at FanXTachoOut if FanXTachoIn is tied to ground.

That is the output after esphome run:

> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
>  - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
> Dependency Graph
> |-- AsyncTCP-esphome @ 2.1.3
> |-- WiFi @ 2.0.0
> |-- FS @ 2.0.0
> |-- Update @ 2.0.0
> |-- ESPAsyncWebServer-esphome @ 3.2.2
> |-- DNSServer @ 2.0.0
> |-- ESPmDNS @ 2.0.0
> |-- noise-c @ 0.1.4
> |-- ArduinoJson @ 6.18.5
>

Heiko-L, 2024-07-30:
AAAARGH, it is hard coded in esp-idf... Why???
[pulse_cnt.c from esp-idf's GitHub](https://github.com/espressif/esp-idf/blob/23e4823f17a8349b5e03536ff7653e3e584c9351/components/esp_driver_pcnt/src/pulse_cnt.c#L725)

Heiko-L, 2024-07-31:
@ssieb is it possible to alter the IO_MUX_x_REG after creating the pulse counter to set the Pin to the desired input mode?

ssieb, 2024-07-31:
You can try.
Why do you have such a circuit?

Heiko-L, 2024-07-31:
> You can try.

How?

> Why do you have such a circuit?

Because the NodeMCU Boards I have are pus***s... I have already three with some defect GPIOs (I think because of weak/non existent clamping diodes)- so I decided to pull up, clamp and limit the current externally.

Heiko-L, 2024-07-31:
Adding a pulse counter with setup_priority: 601 and a binary sensor with setup_priority: 599 on the same GPIO pin let me change the pin configuration via the binary sensor (allow_other_uses must be configured on both components)