esphome/issues #6088: esp32_camera causing bootloops

illuzn, 2024-07-26:
### The problem

Enabling esp32_camera causes a bootloop as follows:

```log
[01:46:07]E (10219) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[01:46:07]E (10219) task_wdt:  - loopTask (CPU 1)
[01:46:07]E (10219) task_wdt: Tasks currently running:
[01:46:07]E (10219) task_wdt: CPU 0: IDLE
[01:46:07]E (10219) task_wdt: CPU 1: IDLE
[01:46:07]E (10219) task_wdt: Aborting.
```

It may be that I have configured the pins incorrectly however, a boot loop should still not arise in this situation.

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Not Applicable

### What platform are you using?

ESP32

### Board

esp32s3box

### Component causing the issue

esp32_camera

### Example YAML snippet

```yaml
esphome:
  name: doorbell

esp32:
  board: esp32s3box
  variant: esp32s3
  flash_size: 16MB
  framework:
# There are build issues with esp32_camera and esp-idf as at July 2024. This is above my paygrade; use arduino for now.
    type: arduino

psram:
  mode: octal
  speed: 80MHz

# T-SIMCAM Setup
esp32_camera:
  external_clock:
    pin: GPIO14
    frequency: 20MHz
  i2c_pins:
    sda: GPIO4
    scl: GPIO5
  data_pins: [GPIO11, GPIO9, GPIO8, GPIO10, GPIO12, GPIO17, GPIO16, GPIO15]
  vsync_pin: GPIO6
  href_pin: GPIO7
  pixel_clock_pin: GPIO13
  name: Camera
  resolution: SVGA
```

### Anything in the logs that might be useful for us?

```txt
[01:46:07]E (10219) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[01:46:07]E (10219) task_wdt:  - loopTask (CPU 1)
[01:46:07]E (10219) task_wdt: Tasks currently running:
[01:46:07]E (10219) task_wdt: CPU 0: IDLE
[01:46:07]E (10219) task_wdt: CPU 1: IDLE
[01:46:07]E (10219) task_wdt: Aborting.
```

### Additional information

There appears to be a seperate issue with esp32_camera which causes a complete device lockup when using esp-idf

hughrawlinson, 2024-08-08:
In case it's useful to anyone, I have this issue too, but with the arduino framework as well as the esp-idf framework, and on the ai-thinker esp32cam (from az-delivery)

illuzn, 2024-08-14:
Did you double check that you are not using the PSRAM pin (see the esphome docs)?