esphome/issues #6091: SML esphome devie after update to 2024.7.2 don't work. device over network not reachable

linuxdep, 2024-07-27:
### The problem

after update from 2024.6.6 to 7.2 esphome device (ESP32) not reachable over network, ip is configured.

I have done some tests and I have noticed that when I do not have the IR read head on the counter, the esphome runs, as soon as I put the read head on the counter, everything seems to hang up.

Restore to 6.6 and everything runs fine again.

I have the same problem on other devices, one that uses BT, one that queries data from the Seplos BMS via Modbus, only the Modbus queries that query the Deye WR work. 

What other data do you need to investigate this in more detail?

Translated with DeepL.com (free version)

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.3 and 2024.7.2 

### What platform are you using?

ESP32

### Component causing the issue

SML and BT and modbus

### Example YAML snippet

```yaml
esphome:
  name: "esp32-stromzaehler"
  friendly_name: esp32-stromzaehler

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xXx"

ota:
  platform: esphome
  password: !secret HA_OTA_Password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-9E17C4"
    password: "xXx"

captive_portal:

web_server:
  port: 80

time:
   - platform: sntp
     id: sntp_time
     timezone: Europe/Berlin
     servers:
     - 192.168.123.1

uart:
  id: uart_1
  rx_pin: GPIO9
  tx_pin: GPIO10
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1

sml:
  id: haus
  uart_id: uart_1

binary_sensor:
  - platform: status
    name: "{name} status"

sensor:
  - platform: sml
    name: "Haus Total energy"
    id: Haus_Total_energy
    sml_id: haus
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 6
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0000001
  - platform: sml
    name: "Haus Total power"
    id: Haus_Total_power
    sml_id: haus
    obis_code: "1-0:16.7.0"
    unit_of_measurement: W
    accuracy_decimals: 2
    device_class: power
    state_class: measurement
    filters:
      - multiply: 0.01
  - platform: sml
    name: "Haus L1 power"
    id: Haus_L1_power
    sml_id: haus
    obis_code: "1-0:36.7.0"
    unit_of_measurement: W
    accuracy_decimals: 2
    device_class: power
    state_class: measurement
    filters:
      - multiply: 0.01 
  - platform: sml
    name: "Haus L2 power"
    id: Haus_L2_power
    sml_id: haus
    obis_code: "1-0:56.7.0"
    unit_of_measurement: W
    accuracy_decimals: 2
    device_class: power
    state_class: measurement
    filters:
      - multiply: 0.01 
  - platform: sml
    name: "Haus L3 power"
    id: Haus_L3_power
    sml_id: haus
    obis_code: "1-0:76.7.0"
    unit_of_measurement: W
    accuracy_decimals: 2
    device_class: power
    state_class: measurement
    filters:
      - multiply: 0.01 
  - platform: sml
    name: "Haus L1 voltage"
    id: Haus_L1_voltage
    sml_id: haus
    obis_code: "1-0:32.7.0"
    unit_of_measurement: V
    accuracy_decimals: 1
    device_class: voltage
    state_class: measurement
    filters:
      - multiply: 0.1
  - platform: sml
    name: "Haus L2 voltage"
    id: Haus_L2_voltage
    sml_id: haus
    obis_code: "1-0:52.7.0"
    unit_of_measurement: V
    accuracy_decimals: 1
    device_class: voltage
    state_class: measurement
    filters:
      - multiply: 0.1
  - platform: sml
    name: "Haus L3 voltage"
    id: Haus_L3_voltage
    sml_id: haus
    obis_code: "1-0:72.7.0"
    unit_of_measurement: V
    accuracy_decimals: 1
    device_class: voltage
    state_class: measurement
    filters:
      - multiply: 0.1
```

Leopoldpaul, 2024-07-29:
This is related to https://github.com/esphome/issues/issues/6071 and https://github.com/esphome/esphome/pull/6148

linuxdep, 2024-07-30:
> This is related to #6071 and [esphome/esphome#6148](https://github.com/esphome/esphome/pull/6148)

Thanks, but i don't found it on open Time.
But i have also in other device Problem with WLAN connect in 7.2, add Moment i use 6.6 and hope that would be fixed.

closed 2024-08-12 by jesserockz