esphome/issues #6094: waveshare_epaper driver 2.90inv2-r2 is displaying text from right to left

Soleil01, 2024-07-28:
### The problem

Waveshare 2.9 version 2 rel 2.1 e-paper driven by ESP32 and using driver 2.90inv2-r2 is writing all text from right to left instead of left to right.

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.7.3

### What platform are you using?

ESP32

### Component causing the issue

Waveshare 2.9 inch e-Paper Module V2 rel 2.1

### Example YAML snippet

```yaml
substitutions:
  api_key: "ol5BL0nEm35XD5kXwyeVaWfBNos4oMnStb5lcbkz55k="
  device_entity_name: epaper
  device_name: esphome_${device_entity_name}
  device_wifi_name: esphome_${device_entity_name}_wifi
  device_friendly_name: Esphome ${device_entity_name}
  device_ip_address: 192.168.1.135
  device_sampling_time: 120s
  device_frequency: 40000Hz
  light_transition_time_default: 1s

esphome:
  name: ${device_entity_name}
  friendly_name: ${device_friendly_name}

esp32:
  board: esp32dev
  framework:
    type: arduino

font:
    # gfonts://family[@weight]
  - file: "gfonts://Roboto"
    id: roboto_20
    size: 20

  - file:
      type: gfonts
      family: Roboto
      weight: 900
    id: roboto_16
    size: 16

  - file: "https://github.com/IdreesInc/Monocraft/releases/download/v3.0/Monocraft.ttf"
    id: web_font
    size: 20

  - file:
      url: "https://github.com/IdreesInc/Monocraft/releases/download/v3.0/Monocraft.ttf"
      type: web
    id: web_font2
    size: 24
spi:
  clk_pin: 26
  mosi_pin: 25

logger:
  #logs:
    #sensor: INFO
    #binary_sensor: INFO
    #text_sensor: INFO

# Enable Home Assistant API
api:
  reboot_timeout: 60min
  encryption:
    key: ${api_key}

ota:
  platform: esphome
  password: !secret ota_password

safe_mode:
  reboot_timeout: 10min
  num_attempts: 5

wifi:
  fast_connect: True
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: ${device_ip_address}
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  ap:
    ssid: ${device_wifi_name}
    password: !secret wifi_password

web_server:
  port: 80
  auth:
    username: !secret web_server_user
    password: !secret web_server_password

time:
  - platform: homeassistant
    id: time_homeassistant

captive_portal:

binary_sensor:
  - platform: status
    name: status
    id: ${device_name}_status

one_wire:
  - platform: gpio
    pin: 32

display:
  - platform: waveshare_epaper
    cs_pin: 27
    dc_pin: 14
#    busy_pin: D4
    reset_pin: 12
    model: 2.90inv2-r2
#    model: 2.90inv2
    full_update_every: 100
    update_interval: 10s
    reset_duration: 2ms
    rotation: 90°
    lambda: |-
      it.printf(150, 6, id(roboto_20), TextAlign::TOP_CENTER, "%s", "Hello World!");
      it.strftime(150, 116, id(roboto_20), TextAlign::BOTTOM_CENTER, "%Y-%m-%d %H:%M:%S", id(time_homeassistant).now());

switch:
  - platform: gpio
    name: "Onboard LED"
    pin: 2
    inverted: False

output:

light:

sensor:
  - platform: uptime
    name: uptime

  - platform: wifi_signal
    name: senseur_signal_wifi
    update_interval: ${device_sampling_time}
```

### Anything in the logs that might be useful for us?

```txt
Had those repeating messages depending on some settings:
[16:22:43] [Wl [component: 2371: Component display took a long time for an operation (52 ms).
[16:22:431 [Wl [component: 238]: Components should block for at most 30 ms.
[16:22:531 [Wl [component: 2371: Component display took a long time
for an operation (51 ms).
[16:22:531 [Wl [component: 238]: Components should block for at most 30 ms.
```

### Additional information

[image]

I tried the other drivers (ex: 2.90inv2) and the text was OK but got scrambled at the refresh time.

rufinus, 2024-11-06:
same display. dont have this problem. (but also the display block warning)

```
spi:
  clk_pin: 18
  mosi_pin: 23

display:
  - platform: waveshare_epaper
    cs_pin: 5
    dc_pin: 17
    busy_pin: 4
    reset_pin: 16
    model: 2.90inv2-r2
    full_update_every: 600
    update_interval: 15s
    rotation: 90°
    lambda: |-
      it.print(it.get_width()/2, 0, id(header), TextAlign::TOP_CENTER, "AMS Box Climate");
      it.printf(5, 50, id(roboto_30), "%.1f°C", id(temp_left).state);
      it.printf(5, 80, id(roboto_24), "  %.1f%%", id(hum_left).state);

      it.printf(it.get_width()-5, 50, id(roboto_30), TextAlign::TOP_RIGHT, "%.1f°C", id(temp_right).state);
      it.printf(it.get_width()-5, 80, id(roboto_24), TextAlign::TOP_RIGHT, "%.1f%%  ", id(hum_right).state);
      it.printf(it.get_width()/2, 110, id(roboto_16), TextAlign::TOP_CENTER, "Heizung ist %s (%.1f Watt)", id(my_heater_status).state.c_str(), id(heater_power).state );
    
    ```
![image](https://github.com/user-attachments/assets/67a4ab24-f074-4ca0-a037-194b99554c72)

BeeSci, 2024-12-09:
Same display here without mirrored text.
Display section:
```
display:
  - platform: waveshare_epaper
    cs_pin: D8
    dc_pin: D3
    busy_pin: D2
    reset_pin: D1
    model: 2.90inv2-r2
    full_update_every: 86400 # 86400 # 24 Stunden 
    rotation: 90°
    reset_duration: 2ms
    update_interval: 10s
    id: MAC_5CCF7F62D39D_Epaper
    pages:
      - id: page1 # große Uhr anzeigen
        lambda: |-
                  // Print time in HH:MM format
                    it.strftime((it.get_width()/2), (it.get_height()/2), id(font5), TextAlign::CENTER, "%H:%M", id(esptime).now());
```
Your font section looks unfamiliar to me.
Font files are located in config/esphome/fonts
Maybe you can try mine:
```
font:
  - file: 'fonts/LucidaSansDemiBold.ttf'
    id: font1
    glyphs:
      ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'Ä', 'ä', 'Ö', 'ö', 'Ü', 'ü', 'ß', '/']
    size: 15

  - file: 'fonts/LucidaSansDemiBold.ttf'
    id: font2
    glyphs:
      ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'Ä', 'ä', 'Ö', 'ö', 'Ü', 'ü', 'ß', '/']
    size: 12

  - file: 'fonts/LucidaSansDemiBold.ttf'
    id: font3
    glyphs:
      ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'Ä', 'ä', 'Ö', 'ö', 'Ü', 'ü', 'ß', '/']
    size: 30

  - file: 'fonts/LucidaSansDemiBold.ttf'
    id: font4
    glyphs:
      ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'Ä', 'ä', 'Ö', 'ö', 'Ü', 'ü', 'ß', '/']
    size: 40

  - file: 'fonts/LucidaSansDemiBold.ttf'
    id: font5
    glyphs:
      ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'Ä', 'ä', 'Ö', 'ö', 'Ü', 'ü', 'ß', '/']
    size: 80
```

dsamarin1, 2025-04-17:
Same problem here! Using a Xiao ESP32-C3. Text and graphics draw from the top right corner instead of top left.

I thought the problem was here but am having trouble editing the source in situ and recompiling on my HA server (proxmox container):
https://github.com/esphome/esphome/blob/af9b56877843638a815b834309a9e85d8a16e694/esphome/components/waveshare_epaper/waveshare_epaper.cpp#L1668-L1669

See page 19 ("Init Code" block) and page 20 (Command 11) of datasheet:
https://files.waveshare.com/upload/7/79/2.9inch-e-paper-v2-specification.pdf

I wonder if something on the waveshare hardware changed?

```
font:
  - file: "gfonts://Roboto"
    id: roboto16
    size: 16

spi:
  clk_pin: 8
  mosi_pin: 10
  # miso_pin: 9

display:
  - platform: waveshare_epaper
    cs_pin: 3
    dc_pin: 2
    # busy_pin: 5
    reset_pin: 4
    model: 2.90inv2-r2
    update_interval: 6s
    full_update_every: 30
    # rotation: 180
    
    lambda: |-
      it.print(0, 0, id(roboto16), "Hello World!");
      it.line(0, 0, 50, 50);
```

[image]

BeeSci, 2025-04-19:
I use the very same display with a ESP8266. Asides from other problems - if I set this " # busy_pin: 5 " to disable the busy pin like you do, I also see the mirrored display. As soon as I enable the busy pin, the orientation is correct. My display is about 2 years old. So I do not think the hardware has changed since then.

dsamarin1, 2025-05-24:
Picked this back up. Working correctly now. Turns out I had to set `display: model: 2.90inv2` (was using `2.90inv2-r2`) and switch my `busy_pin` to another pin on the XIAO-ESP32-C3 (from pin D5 to D9). Hope this can help you @Soleil01

MaximeCheramy, 2025-07-04:
I can reproduce this issue if I comment out the `busy_pin` setting.