esphome/issues #6095: GROW R503 Fingerprint reader keep restarting once I scan a finger

azen0k, 2024-07-28:
### The problem

Hi!
I'm encourtering an issue with the GROW Fingerprint reader R503.

I tried 4 boards (ESP32 WROOM 32 USBC or USB micro / Nodemcu v2 / Nodemcu V3) and i always have the same behaviour.

The fingerprint is recognized and seems to be correctly wired (Aura LED working as intended)

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.2

### What platform are you using?

ESP32

### Board

Generic AliExpress ESP32 WROOM 32 USBC

### Component causing the issue

UART?

### Example YAML snippet

```yaml
substitutions:
  name: cabane-a-velo
  friendly_name: Cabane-a-velo

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework: 
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  services:
  - service: enroll
    variables:
      finger_id: int
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
  - service: cancel_enroll
    then:
[... 341 lines cut ...]
[12:12:55][D][fingerprint_grow:263]: Got parameters
[12:12:55][D][sensor:094]: 'Cabane-a-velo Status': Sending state 4.00000  with 0 decimals of accuracy
[12:12:55][D][sensor:094]: 'Cabane-a-velo Capacity': Sending state 200.00000  with 0 decimals of accuracy
[12:12:56][D][sensor:094]: 'Cabane-a-velo Security Level': Sending state 3.00000  with 0 decimals of accuracy
[12:12:56][D][binary_sensor:036]: 'Cabane-a-velo Enrolling': Sending state OFF
[12:12:56][D][fingerprint_grow:285]: Getting fingerprint count
[12:12:56][D][fingerprint_grow:288]: Got fingerprint count
[12:12:56][D][sensor:094]: 'Cabane-a-velo Count': Sending state 0.00000  with 0 decimals of accuracy
[12:12:56][D][text_sensor:064]: 'Cabane-a-velo ESPHome Version': Sending state '2024.7.2 Jul 28 2024, 11:52:05'
[12:12:56][D][binary_sensor:034]: 'Cabane-a-velo Status': Sending initial state OFF
[12:12:56][C][wifi:047]: Setting up WiFi...
[12:12:56][C][wifi:060]: Starting WiFi...
[12:12:56][C][wifi:061]:   Local MAC: 08:D1:F9:E1:64:B0
[12:12:56][D][wifi:481]: Starting scan...
[12:12:56][D][fingerprint_grow:348]: Setting Aura LED
[12:12:56][D][fingerprint_grow:352]: Aura LED set
[12:12:56][D][fingerprint_grow:125]: Scan and match
[12:12:56][D][fingerprint_grow:156]: Getting image 1
[12:12:56][D][fingerprint_grow:168]: Finger Misplaced
[12:12:56][D][text_sensor:064]: 'Cabane-a-velo State': Sending state 'Misplaced finger'
```

### Additional information

With the latest board I have also a continuous fingerprint read. The Grow R503 seems to detect a finger on its own.

azen0k, 2024-07-28:
On every board I get the "CORRUPTED" log

azen0k, 2024-07-28:
I just tried to compile firmware with older version 2024.3.2 ([HA Community topic](https://community.home-assistant.io/t/esp32-crashes-with-grow-r503-fingerprint-sensor/702025/2))

Same issue

nguyenquangduc, 2024-07-28:
i also faced the same problem.i am using r303 fingerprint which has no touch detection output and my esp hangs very often and won't restart

azen0k, 2024-07-30:
Just tried to compile with ESP-IDF. Same result.

log attached.
[logs (1).txt](https://github.com/user-attachments/files/16428001/logs.1.txt)

azen0k, 2024-08-07:
Just received an ESP32-S3-WROOM-1 and it works.

26tajeen, 2024-12-05:
Did this situation ever resolve?

azen0k, 2024-12-05:
> Did this situation ever resolve?

I don't think so. I found my way around this issue by adding a delay corresponding to the blinking/flashing time. It works.