esphome/issues #6096: select causing bootloops

EinRainerZufall, 2024-07-28:
### The problem

I wanted to use a `select` sensor to assign a color to a single LED from a WS2812 LED chain. When I install the select I always get a boot loop. I have tested 5 different ESP32 boards and the error occurred on all of them.

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

select

### Example YAML snippet

```yaml
esphome:
  name: led-test
  friendly_name: led-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

light:
  - platform: esp32_rmt_led_strip
    id: led_chain_2
    name: "RGB LED"
    rgb_order: GRB
    num_leds: 10
    rmt_channel: 1
    pin: GPIO19
    chipset: WS2812
    restore_mode: RESTORE_DEFAULT_OFF

select:
  - platform: template
    id: b_led_1
    name: "Button 1 LED mode"
    icon: mdi:led-variant-on
    options: 
      - "off"
      - "normal"
      - "warn"
      - "error"
    initial_option: "off"
    optimistic: True
    on_value: 
      then:
        - if: 
            condition: 
              - lambda: 'return id(b_led_1).state == "off";'
            then: 
              - light.addressable_set: 
                  id: led_chain_2
                  range_from: 0
                  range_to: 1
                  red: 0
                  green: 0
                  blue: 0
                  color_brightness: 0
        - if: 
            condition: 
              - lambda: 'return id(b_led_1).state == "normal";'
            then: 
              - light.addressable_set: 
                  id: led_chain_2
                  range_from: 0
                  range_to: 1
                  red: 0
                  green: 1
                  blue: 0
                  color_brightness: 0.75
        - if: 
            condition: 
              - lambda: 'return id(b_led_1).state == "warn";'
            then: 
              - light.addressable_set: 
                  id: led_chain_2
                  range_from: 0
                  range_to: 1
                  red: 1
                  green: 0.6
                  blue: 0.15
                  color_brightness: 0.75
        - if: 
            condition: 
              - lambda: 'return id(b_led_1).state == "error";'
            then: 
              - light.addressable_set: 
                  id: led_chain_2
                  range_from: 0
                  range_to: 1
                  red: 1
                  green: 0
                  blue: 0
                  color_brightness: 0.75
```

### Anything in the logs that might be useful for us?

```txt
[12:03:35]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[12:03:35]
[12:03:35]Core  1 register dump:
[12:03:35]PC      : 0x400e448c  PS      : 0x00060030  A0      : 0x800e339a  A1      : 0x3ffb22d0  
[12:03:35]A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffb28b8  
[12:03:35]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffc3204  A9      : 0x3ffb22b0  
[12:03:35]A10     : 0x3ffb231c  A11     : 0x00000000  A12     : 0x00000002  A13     : 0x5eabd4db  
[12:03:35]A14     : 0x3ffc3204  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
[12:03:35]EXCVADDR: 0x00000090  LBEG    : 0x40089734  LEND    : 0x4008974a  LCOUNT  : 0xffffffff  
[12:03:35]
[12:03:35]
[12:03:35]Backtrace:0x400e4489:0x3ffb22d00x400e3397:0x3ffb2350 0x400e2d70:0x3ffb2390 0x400e3489:0x3ffb23d0 0x400e2d70:0x3ffb2410 0x400e2dc6:0x3ffb2450 0x400dd315:0x3ffb24c0 0x400ddd98:0x3ffb2540 0x400ddde6:0x3ffb25a0 0x400e1bf0:0x3ffb25c0 0x40160425:0x3ffb25e0 0x400e16d4:0x3ffb2600 0x400e418e:0x3ffb2630 0x400e815e:0x3ffb2820 
[12:03:35]
[12:03:35]
[12:03:35]
[12:03:35]
[12:03:35]ELF file SHA256: 0000000000000000
[12:03:35]
[12:03:35]Rebooting...
[12:03:35]ets Jun  8 2016 00:22:57
[12:03:35]
[12:03:35]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:03:35]configsip: 0, SPIWP:0xee
[12:03:35]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:03:35]mode:DIO, clock div:2
[12:03:35]load:0x3fff0030,len:1184
[12:03:35]load:0x40078000,len:13132
[12:03:35]load:0x40080400,len:3036
[12:03:35]entry 0x400805e4
```

ssieb, 2024-07-28:
What were the lines before the exception?

EinRainerZufall, 2024-07-28:
sorry for not copying everything

> [22:49:00][I][logger:156]: Log initialized
[22:49:00][C][safe_mode:079]: There have been 7 suspected unsuccessful boot attempts
[22:49:00][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:49:00][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:49:00][I][app:029]: Running through setup()...
[22:49:00][C][esp32_rmt_led_strip:021]: Setting up ESP32 LED Strip...
[22:49:00][D][template.select:014]: Setting up Template Select
[22:49:00][D][template.select:017]: State from initial: off
[22:49:00][D][select:015]: 'Button 1 LED mode': Sending state off (index 0)
[22:49:00]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[22:49:00]
[22:49:00]Core  1 register dump:
[22:49:00]PC      : 0x400e448c  PS      : 0x00060030  A0      : 0x800e339a  A1      : 0x3ffb22d0  
[22:49:00]A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffb28b8  
[22:49:00]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffc3204  A9      : 0x3ffb22b0  
[22:49:00]A10     : 0x3ffb231c  A11     : 0x00000000  A12     : 0x00000002  A13     : 0xaf4bea9d  
[22:49:00]A14     : 0x3ffc3204  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
[22:49:00]EXCVADDR: 0x00000090  LBEG    : 0x40089734  LEND    : 0x4008974a  LCOUNT  : 0xffffffff  
[22:49:00]
[22:49:00]
[22:49:00]Backtrace:0x400e4489:0x3ffb22d00x400e3397:0x3ffb2350 0x400e2d70:0x3ffb2390 0x400e3489:0x3ffb23d0 0x400e2d70:0x3ffb2410 0x400e2dc6:0x3ffb2450 0x400dd315:0x3ffb24c0 0x400ddd98:0x3ffb2540 0x400ddde6:0x3ffb25a0 0x400e1bf0:0x3ffb25c0 0x40160425:0x3ffb25e0 0x400e16d4:0x3ffb2600 0x400e418e:0x3ffb2630 0x400e815e:0x3ffb2820 
[22:49:00]
[22:49:00]
[22:49:00]
[22:49:00]
[22:49:00]ELF file SHA256: 0000000000000000
[22:49:00]
[22:49:00]Rebooting...
[22:49:00]ets Jun  8 2016 00:22:57
[22:49:00]
[22:49:00]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:49:00]configsip: 0, SPIWP:0xee
[22:49:00]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:49:00]mode:DIO, clock div:2
[22:49:00]load:0x3fff0030,len:1184
[22:49:00]load:0x40078000,len:13132
[22:49:00]load:0x40080400,len:3036
[22:49:00]entry 0x400805e4

This is the whole LOG that is repeated over and over again

ssieb, 2024-07-28:
Can you plug the device into the HA server, so that esphome can decode the stack trace?

EinRainerZufall, 2024-07-29:
this is the LOG from the HA server

```txt
[15:06:16][I][logger:156]: Log initialized
[15:06:16][C][safe_mode:079]: There have been 1 suspected unsuccessful boot attempts
[15:06:16][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:06:16][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[15:06:16][I][app:029]: Running through setup()...
[15:06:16][C][esp32_rmt_led_strip:021]: Setting up ESP32 LED Strip...
[15:06:16][D][template.select:014]: Setting up Template Select
[15:06:16][D][template.select:017]: State from initial: off
[15:06:16][D][select:015]: 'Button 1 LED mode': Sending state off (index 0)
[15:06:16]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[15:06:16]
[15:06:16]Core  1 register dump:
[15:06:16]PC      : 0x400e448c  PS      : 0x00060030  A0      : 0x800e339a  A1      : 0x3ffb22d0  
WARNING Decoded 0x400e448c: esphome::light::AddressableLight::schedule_show() at /data/build/led-test/src/esphome/components/light/addressable_light.h:76
 (inlined by) esphome::light::AddressableSet<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int>::play(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int) at /data/build/led-test/src/esphome/components/light/automation.h:188
[15:06:16]A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffb28b8  
[15:06:16]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffc3204  A9      : 0x3ffb22b0  
[15:06:16]A10     : 0x3ffb231c  A11     : 0x00000000  A12     : 0x00000002  A13     : 0xb962128f  
[15:06:16]A14     : 0x3ffc3204  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
[15:06:16]EXCVADDR: 0x00000090  LBEG    : 0x40089734  LEND    : 0x4008974a  LCOUNT  : 0xffffffff  
[15:06:16]
[15:06:16]
[15:06:16]Backtrace:0x400e4489:0x3ffb22d00x400e3397:0x3ffb2350 0x400e2d70:0x3ffb2390 0x400e3489:0x3ffb23d0 0x400e2d70:0x3ffb2410 0x400e2dc6:0x3ffb2450 0x400dd315:0x3ffb24c0 0x400ddd98:0x3ffb2540 0x400ddde6:0x3ffb25a0 0x400e1bf0:0x3ffb25c0 0x40160425:0x3ffb25e0 0x400e16d4:0x3ffb2600 0x400e418e:0x3ffb2630 0x400e815e:0x3ffb2820 
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400e4489: esphome::light::AddressableSet<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int>::play(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int) at /data/build/led-test/src/esphome/components/light/automation.h:189
WARNING Decoded 0x400e3397: esphome::Action<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int>::play_complex(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int) at /data/build/led-test/src/esphome/core/automation.h:125
WARNING Decoded 0x400e2d70: esphome::ActionList<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int>::play(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int) at /data/build/led-test/src/esphome/core/automation.h:203
WARNING Decoded 0x400e3489: esphome::IfAction<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int>::play_complex(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int) at /data/build/led-test/src/esphome/core/base_automation.h:207
WARNING Decoded 0x400e2d70: esphome::ActionList<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int>::play(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int) at /data/build/led-test/src/esphome/core/automation.h:203
WARNING Decoded 0x400e2dc6: esphome::Automation<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int>::trigger(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int) at /data/build/led-test/src/esphome/core/automation.h:241
 (inlined by) esphome::Trigger<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int>::trigger(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int) at /data/build/led-test/src/esphome/core/automation.h:98
 (inlined by) esphome::select::SelectStateTrigger::SelectStateTrigger(esphome::select::Select*)::{lambda(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)#1}::operator()(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int) const at /data/build/led-test/src/esphome/components/select/automation.h:13
 (inlined by) std::_Function_handler<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int), esphome::select::SelectStateTrigger::SelectStateTrigger(esphome::select::Select*)::{lambda(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)#1}>::_M_invoke(std::_Any_data const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, unsigned int&&) at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
[... cut ...]
[15:06:16]
[15:06:16]
[15:06:16]
[15:06:16]ELF file SHA256: 0000000000000000
[15:06:16]
[15:06:16]Rebooting...
[15:06:16]ets Jun  8 2016 00:22:57
[15:06:16]
[15:06:16]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:06:16]configsip: 0, SPIWP:0xee
[15:06:16]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:06:16]mode:DIO, clock div:2
[15:06:16]load:0x3fff0030,len:1184
[15:06:16]load:0x40078000,len:13132
[15:06:16]load:0x40080400,len:3036
[15:06:16]entry 0x400805e4
```

ssieb, 2024-07-29:
That's what I expected.  The select is trying to control the light before it's properly initialized.  Try adding `setup_priority: 600` to the select.

EinRainerZufall, 2024-07-29:
Thanks, it works now. Is it the case that the select should be loaded before the light? And I didn't find anything about `setup_priority` in the select docu. Does this work for all components and is there a list in which order what is loaded?

ssieb, 2024-07-29:
The select needs to be initialized after the light.  That option is not documented because it can cause a lot of weird issues and crashing if you don't know what you're doing.  The default priorities are set to work in most cases, although I'm thinking the template select shouldn't be so high, because it's probably going to used in situations like this.

closed 2024-07-29 by ssieb