esphome/issues #6097: what "clean" do exactly ?

phil31, 2024-07-28:
### The problem

Hi, 
i try to buil a new ESP32 device with HA addon .
with the simple yaml generated, i can't compile right.
notice that is working for several others targets, but not for the simple ESP32.

i would like to know what does exactlty the "clean" command ?
and where are physicaly stored the source/compilation files ?

i notice this breaking change : 
"The /config/esphome/.esphome folder has been removed and replaced with /data when running ESPHome as a Home Assistant add-on. This was done because it is exactly what the add-on /data folder was made for and there were many files in the .esphome folder that were making the Home Assistant Backup larger for no reason at all. All files in there are generated or sourced from online based on your configurations files which have not been touched."

==> but can't manage to find the /data directory with my compiled files !!! (using SSH or file navigator )

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

custom

### Example YAML snippet

```yaml
esphome:
  name: solar-test
  friendly_name: solar_test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "gLgF///gPyzTK8bWLI1AMb+ynb75q+dw/2JGgM42eC0="

ota:
  - platform: esphome
    password: "e78a4b6ee93bf4cb48b15cfedb2b1277"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Solar-Test Fallback Hotspot"
    password: "xWSlVHKafbLz"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
[... 218 lines cut ...]
 ! \xa0           _lock.release(\xa9;
 ^
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/mutex:473:4: error: stray '\240' in program
 ! \xa0           _lock.release(\xa9;
    ^
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/mutex:473:16: error: stray '\177' in program
 ! \xa0           _lock.release(\xa9;
                ^
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/mutex:473:40: error: stray '\177' in program
 ! \xa0           _lock.release(\xa9;
                                        ^
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/mutex:475:6: error: missing terminating " character
  }
      ^                                 
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/mutex:476:3: error: stray '\240' in program
Compiling .pioenvs/solar-test/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/solar-test/src/esphome/components/esp32/gpio.cpp.o
```

### Additional information

it seem clearly a problem of setup ..
what are my options please ?

thanks for any help

ssieb, 2024-07-28:
You need the advanced ssh option so you can access the host.  But you have some serious file corruption.  Re-installing the addon might fix it, but it's an indication of a system problem.
But this is not something for a github issue.  Come ask on the esphome discord server if you need more help.

phil31, 2024-07-28:
ok thanks for you advice

closed 2024-12-10 by github-actions[bot]