esphome/issues #6100: uart debug after delimiter doesn't work as expected for escaped characters (\r, \n etc)

cccr, 2024-07-30:
### The problem

I'm using example from [uart documentation](https://esphome.io/components/uart.html#debugging) to log incoming message after delimiter "\n" (just changed _dummy_receiver_ to _true_ and _delimiter_ to _"\r"_)

```
# Example configuration entry
uart:
  baud_rate: 115200
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

However it doesn't work. I see every character I send through uart ignoring delimiter "\r", because after->timeout is in effect with default 100ms

The workaround is to use list value:

```
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  id: uart_1
  data_bits: 7
  parity: EVEN
  stop_bits: 1
  baud_rate: 1200
  debug:
    direction: RX
    dummy_receiver: true
    after:
      timeout: 0.5min
      delimiter:
        - 13
    sequence:
      - lambda: |-
          id(uart_1).write_str("\n");
          std::string str(bytes.begin(), bytes.end());
          id(mqtt_client).publish("ti707/tx", str.c_str());
```

(with increased timeout like 0.5 minutes it works as intended -- message published to mqtt right after I press "\r" on the uart device)

I'm not a Python engineer and cannot offer a reliable fix, but looks like this is the place, where it was broken (5 years ago ?)
https://github.com/esphome/esphome/blob/83bb7d0266459ebae70d663addabc45a1f112092/esphome/components/uart/__init__.py#L101

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

uart

closed 2024-12-11 by github-actions[bot]

xaviergr, 2025-01-02:
I can also confirm the above problem. The delimiter option (as well as the bytes option) is ignored if written like it is suggested in the documentation.

ssieb, 2025-01-03:
The `after:` options are "or"ed.  Whichever one happens first will trigger the logging.

ssieb, 2025-01-03:
It generally shouldn't unless your data is arriving quite slowly.  That timeout is supposed to be inter-character, not for the whole thing, so unless you're trying to type to it, the data should arrive before the timeout.