esphome/issues #6101: Incorrect operation of MODBU

0IA, 2024-07-30:
### The problem

An incorrect request is sent to the bus 4 times in a row instead of once. At the same time, the bytes of the number of registers are incorrectly formed in the request.
I tried to form the request in different ways (see the commented lines), but the result is always the same.
This was working a long time ago (about a year) when I first did the tests. But when I came back to the project after a long pause, I am getting an wrong result.

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.3

### What platform are you using?

ESP32

### Board

esp32dev on KinCony  KC868-A16 

### Component causing the issue

modbus_controller

### Example YAML snippet

```yaml
uart:
  id: mbus
  tx_pin: GPIO13
  rx_pin: GPIO16
  baud_rate: 9600
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
     delimiter: "\n"
    sequence:
     - lambda: UARTDebug::log_string(direction, bytes);

modbus:
  id: modbus1
  uart_id: mbus
  send_wait_time: 200ms

modbus_controller:
- id: a6
  address: 1
  modbus_id: modbus1
  update_interval: 10s

sensor:
#- platform: modbus_controller
#  id: pzem_voltage_l1
#  modbus_controller_id: pzem_test
#  name: "EVPZEM voltage L1"
#  register_type: read
#  address: 0x0000
#  value_type: U_DWORD
- platform: modbus_controller
  modbus_controller_id: a6
  name: "voltage_1"
  id: pzem_custom_command
  #    address: 0x0000
  #    register_type: "read"
  ## reimplement using custom_command
  # 0xF8 : modbus device address
  # 0x4 : modbus function code
  # 0x0 : high byte of modbus register address
  # 0x0: low byte of modbus register address
  # 0x00: high byte of total number of registers requested
  # 0x01: low byte of total number of registers requested
  custom_command: [0xF8, 0x4, 0x0, 0x0, 0x00, 0x01]
  value_type: U_DWORD
  unit_of_measurement: V
  accuracy_decimals: 1
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.7.2
INFO Reading configuration /config/esphome/dbc.yaml...
INFO Starting log output from 192.168.1.199 using esphome API
INFO Successfully connected to distribution-box-controller-1 @ 192.168.1.199 in 0.003s
INFO Successful handshake with distribution-box-controller-1 @ 192.168.1.199 in 0.127s
[12:00:02][I][app:100]: ESPHome version 2024.7.2 compiled on Jul 30 2024, 11:42:50
[12:00:02][C][logger:185]: Logger:
[12:00:03][C][logger:186]:   Level: VERY_VERBOSE
[12:00:03][C][logger:188]:   Log Baud Rate: 115200
[12:00:03][C][logger:189]:   Hardware UART: UART0
[12:00:03][C][uart.arduino_esp32:151]: UART Bus 1:
[12:00:03][C][uart.arduino_esp32:152]:   TX Pin: GPIO13
[12:00:03][C][uart.arduino_esp32:153]:   RX Pin: GPIO16
[12:00:03][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[12:00:03][C][uart.arduino_esp32:157]:   Baud Rate: 9600 baud
[12:00:03][C][uart.arduino_esp32:158]:   Data Bits: 8
[12:00:03][C][uart.arduino_esp32:159]:   Parity: NONE
[12:00:03][C][uart.arduino_esp32:160]:   Stop bits: 1
[12:00:03][C][modbus:152]: Modbus:
[12:00:03][C][modbus:154]:   Send Wait Time: 200 ms
[12:00:03][C][modbus:155]:   CRC Disabled: NO
[12:00:03][C][modbus_controller:341]: ModbusController:
[12:00:03][C][modbus_controller:342]:   Address: 0x01
[12:00:03][C][modbus_controller:344]: sensormap
[12:00:03][C][modbus_controller:348]:  Sensor type=0 start=0xF47F offset=0x0 count=2 size=4
[12:00:03][C][modbus_controller:350]: ranges
[12:00:03][C][modbus_controller:353]:   Range type=0 start=0xF47F count=2 skip_updates=0
[12:00:03][C][modbus_controller:355]: server registers
[12:00:03][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'voltage_1'
[12:00:03][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[12:00:03][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: 'V'
[12:00:03][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 1
[12:00:03][C][ethernet:310]: Ethernet:
[12:00:03][C][ethernet:492]:   IP Address: 192.168.1.199
[12:00:03][C][ethernet:493]:   Hostname: 'distribution-box-controller-1'
[12:00:03][C][ethernet:494]:   Subnet: 255.255.255.0
[12:00:03][C][ethernet:495]:   Gateway: 192.168.1.1
[12:00:03][C][ethernet:500]:   DNS1: 0.0.0.0
[12:00:03][C][ethernet:501]:   DNS2: 0.0.0.0
[12:00:03][C][ethernet:513]:   MAC Address: B0:A7:32:80:E4:94
[12:00:03][C][ethernet:514]:   Is Full Duplex: YES
[12:00:03][C][ethernet:515]:   Link Speed: 100
[12:00:03][C][ethernet:324]:   MDC Pin: 23
[12:00:03][C][ethernet:325]:   MDIO Pin: 18
[12:00:03][C][ethernet:326]:   PHY addr: 0
[12:00:03][C][ethernet:328]:   Type: LAN8720
[12:00:03][C][mdns:116]: mDNS:
[12:00:03][C][mdns:117]:   Hostname: distribution-box-controller-1
[12:00:03][V][mdns:118]:   Services:
[12:00:03][V][mdns:120]:   - _esphomelib, _tcp, 6053
[12:00:03][V][mdns:122]:     TXT: friendly_name = distribution box controller 1
[12:00:03][V][mdns:122]:     TXT: version = 2024.7.2
[12:00:03][V][mdns:122]:     TXT: mac = b0a73280e494
[12:00:03][V][mdns:122]:     TXT: platform = ESP32
[12:00:03][V][mdns:122]:     TXT: board = esp32dev
[12:00:03][V][mdns:122]:     TXT: network = ethernet
[12:00:03][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[12:00:03][C][esphome.ota:073]: Over-The-Air updates:
[12:00:03][C][esphome.ota:074]:   Address: 192.168.1.199:3232
[12:00:03][C][esphome.ota:075]:   Version: 2
[12:00:03][C][esphome.ota:078]:   Password configured
[12:00:03][C][safe_mode:018]: Safe Mode:
[12:00:03][C][safe_mode:020]:   Boot considered successful after 60 seconds
[12:00:03][C][safe_mode:021]:   Invoke after 10 boot attempts
[12:00:03][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[12:00:03][C][api:139]: API Server:
[12:00:03][C][api:140]:   Address: 192.168.1.199:6053
[12:00:03][C][api:142]:   Using noise encryption: YES
[12:00:06][VV][scheduler:226]: Running interval 'update' with interval=10000 last_execution=936565 (now=946565)
[12:00:06][V][modbus_controller:224]: Updating modbus component
[12:00:06][VV][modbus_controller:228]: Updating range 0xF47F
[12:00:06][V][modbus_controller:191]: Range : F47F Size: 2 (0) skip: 0
[12:00:06][V][modbus_controller:044]: Sending next modbus command to device 1 register 0xF47F count 2
[12:00:06][VV][uart.arduino_esp32:194]:     Flushing...
[12:00:06][V][modbus:231]: Modbus write raw: F8.04.00.00.00.01 (6)
[12:00:06][V][modbus_controller:557]: Command sent 0 0xF47F 2
[12:00:06][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[12:00:07][V][modbus_controller:044]: Sending next modbus command to device 1 register 0xF47F count 2
[12:00:07][VV][uart.arduino_esp32:194]:     Flushing...
[12:00:07][V][modbus:231]: Modbus write raw: F8.04.00.00.00.01 (6)
[12:00:07][V][modbus_controller:557]: Command sent 0 0xF47F 2
[12:00:07][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[12:00:07][V][modbus_controller:044]: Sending next modbus command to device 1 register 0xF47F count 2
[12:00:07][VV][uart.arduino_esp32:194]:     Flushing...
[12:00:07][V][modbus:231]: Modbus write raw: F8.04.00.00.00.01 (6)
[12:00:07][V][modbus_controller:557]: Command sent 0 0xF47F 2
[12:00:07][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[12:00:07][V][modbus_controller:044]: Sending next modbus command to device 1 register 0xF47F count 2
[12:00:07][VV][uart.arduino_esp32:194]:     Flushing...
[12:00:07][V][modbus:231]: Modbus write raw: F8.04.00.00.00.01 (6)
[12:00:07][V][modbus_controller:557]: Command sent 0 0xF47F 2
[12:00:07][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[12:00:07][V][modbus_controller:044]: Sending next modbus command to device 1 register 0xF47F count 2
[12:00:07][VV][uart.arduino_esp32:194]:     Flushing...
[12:00:07][V][modbus:231]: Modbus write raw: F8.04.00.00.00.01 (6)
[12:00:07][V][modbus_controller:557]: Command sent 0 0xF47F 2
[12:00:07][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[12:00:07][D][modbus_controller:040]: Modbus command to device=1 register=0xF47F countdown=0 no response received - removed from send queu
```

ssieb, 2024-07-30:
Why are you overriding the command?  It's sending what you're telling it to.
And for the uart, just add `debug:`, not the other options.  It will be easier to read.

0IA, 2024-07-31:
This is a test configuration, I tried several ways to execute the same request. The result is always the same. 
For convenience, I lowered the debug_level now there is less information in the log.

Anyway, I'm looking for answers to the following questions:
Why is the request repeated 4 times?
Why is the request in debug output not as expected?
In debug output I see this data: 
` \xF8\x04\x00\x00\x00\x01%\xA3`
Whereas the expected request should contain other values:
```
    Request: [F8] [04] [0000] [0001] [25A3]
             |    |    |      |      |-> CRC16 (41765)
             |    |    |      |->Number Of Registers (1)
             |    |    |-> Register Offset (0 = 30001)
             |    |-> Function Code (4)
             |-> Slave ID (-8)
```

Debug log:
```
[11:58:30][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:30][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:30][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:31][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:31][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:31][D][modbus_controller:040]: Modbus command to device=1 register=0xF47F countdown=0 no response received - removed from send queue
[11:58:40][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:40][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:40][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:41][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:41][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:41][D][modbus_controller:040]: Modbus command to device=1 register=0xF47F countdown=0 no response received - removed from send queue
[11:58:50][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:50][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:50][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:51][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:51][D][uart_debug:158]: >>> "\xF8\x04\x00\x00\x00\x01%\xA3"
[11:58:51][D][modbus_controller:040]: Modbus command to device=1 register=0xF47F countdown=0 no response received - removed from send queue

```

ssieb, 2024-07-31:
It repeats because it doesn't get a response.  It doesn't get a response because you're sending an invalid message.
Your uart debug output is bad because you're trying to print it as a string.  The `%` is your "missing" `0x25`.

ssieb, 2024-07-31:
If you need more help, then come ask on the esphome discord server.

closed 2024-07-31 by ssieb

0IA, 2024-07-31:
Thank you!
So the problem is elsewhere.

ssieb, 2024-07-31:
The problem is that you're sending an invalid message.  Don't override it and let the component send it properly.