esphome/issues #6102: HTTP_REQUEST : Certificate verification

PaulBiod, 2024-07-30:
### The problem

Hi, since last esphome update, I have an error preventing me from compiling.

The message is : 

```
Failed config

http_request: None
  {}
ESPHome supports certificate verification only via ESP-IDF. Set 'verify_ssl: false' to skip certificate validation and allow less secure HTTPS connections.
```

As far as I don't use HTTP_request, I suppose it's used from the PR
Thanks for helping

### Which version of ESPHome has the issue?

2024.7.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.3

### What platform are you using?

ESP32

### Board

ESP32 S3

### Component causing the issue

ONLINE_IMAGE PR

guillempages, 2024-08-01:
Online Image has been refactor to rely on the http_request component for downloading, instead of reimplementing the wheel.

As such, I updated the example on that PR (https://github.com/esphome/esphome/pull/4710), to make it clear that http_request is now needed.

For arduino frameworks, you need to explicitly add
```
http_request:
  verify_ssl: false
```
into your configuration; since arduino does not support SSL verification (at least yet).

Also, this is not really an issue in the core esphome (since that PR has not been merged yet), but an issue with that PR, and should have been commented there 😉

closed 2024-08-01 by guillempages

PaulBiod, 2024-08-01:
Hi, Thanks, I fixed it by using IDF but I appreciate the solution for arduino. Thanks a lot

guillempages, 2024-08-01:
FYI: with the latest version, ESP-IDF will also need to explicitly configure "http_request:" (but the SSL validation can be left as default.

PaulBiod, 2024-08-01:
Thanks.
I have a weird error now : 

[image]

guillempages, 2024-08-01:
Hm. This is because of a change that was needed to fix some lint/clang issue in the PR.
There was a change in core in the latest version,  and I had to adapt. Unfortunately this means that in the meantime the pr cannot be used without the dev version of esphome. ☹️

I am hoping that the PR can be merged soon,  and it is then available directly on esphome.

I fear you will have to wait until the 2024.8 release (~3 weeks) to get that error fixed 😢 or at least the beta, if you are brave enough 😉

PaulBiod, 2024-08-01:
It's not a problem, I just won't compile cause it's working. Hope it will get merged soon.
Thanks for all