esphome/issues #6106: mics_4514 sensor reading way off

erkr, 2024-07-31:
### The problem

I noticed that the sensor readings for the mics_4514 are way off. Low concentrations are ignored, higher values are calculated wrong. 

Here is the graph for the datasheet for the RED sensor (the OX based sensors have the same issue):

[image]

I marked the thresholds used in code for the various sensors with colored dots. Two issues for each sensor:
- only concentrations on the right side of these dot are reported! Especially dangerous with the CO sensor that ignores values below 11ppm (red dot). Please note 6ppm is the max CO concentration for 24h, totally ignored by this implementation
- The value calculations are wrong! The datasheet graph is exponential, but the value interpolation is  done linear (only correct for the max concentrations).

### Which version of ESPHome has the issue?

Firmware: 2024.6.6 (Jul 8 2024, 07:53:53)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core 2024.7

### What platform are you using?

ESP32

### Board

Apollo Air-1 device with ESP23 

### Component causing the issue

mics_4514

### Example YAML snippet

```yaml
n/a the issue is in the component code itself
```

### Anything in the logs that might be useful for us?

```txt
N/A
```

erkr, 2024-07-31:
Update; I was able to determine the right math for calculating the CO value by mapping points from the datasheet onto a double log graph and find a matching function (see [here](https://www.desmos.com/calculator/u9abxqqxre)).
The calculation for CO should be:
```
if (red_f > 3.4f) {
 CO = 0.0;
}
else if (red_f < 0.01) {
  CO = 1000;
}
else {
  CO = 4.2/ pow(x,1.2);
}
```
I will determine the other sensors as well

erkr, 2024-07-31:
Ammonia ([here](https://www.desmos.com/calculator/xknqara4nb)): 
if (red_f > 0.98f) {
 Ammonia = 0.0;
}
else if (red_f < 0.2532) {
  Ammonia = 500.0;
}
else {
  Ammonia = 0.9/ pow(x,4.6);
}

erkr, 2024-07-31:
Methane ([here](https://www.desmos.com/calculator/dxjf1ygh84)):

if (red_f > 0.9f) {
 Methane = 0.0;
}
else if (red_f < 0.54) {
  Methane = 0.0;  // outside the range->unlikely
}
else {
  Methane = 630/ pow(x,4.4);
}

erkr, 2024-07-31:
Ethanol ([here](https://www.desmos.com/calculator/86yjd1uvt7)):

```
if (red_f > 1.0f) {
  Ethanol = 0.0;
}
else if (red_f < 0.04) {
  Ethanol = 0.0; // outside the range->unlikely
}
else {
  Ethanol = 1.52/ pow(x,1.55);
}
```

erkr, 2024-07-31:
Last but not least. NO2 on the OX sensor ([here](https://www.desmos.com/calculator/v0w0tsvgdb)):

```
if (ox_f < 0.3f) {
  NO2 = 0.0;
}
else {
  NO2 = 0.4 * pow(x,0.74);
}
```

erkr, 2024-08-01:
Thanks for working on this🥳

> Working on a PR for this. Are they by chance backwards in some? CO example below

No, I believe I got that right.  Note in the graph the relation sensor readout of RED vs CO is negative. So smaller readouts are higher CO values. 
A readout of 3.4 represents the top left of the CO curve (1ppm), 0.01 represents the bottom right 1000ppm

erkr, 2024-08-01:
Small remark. This formula:

``` 
CO = 4.2 / pow(red_f,1.2);
``` 

It's mathematically equal to:

``` 
CO = 4.2 * pow(red_f,-1.2);
``` 

Whatever you prefer☺️

TrevorSchirmer, 2024-08-01:
Created a PR that matches. Mind taking a look @erkr 

https://github.com/esphome/esphome/pull/7173

erkr, 2024-08-01:
Looks great. How does the process work? Will this be a dev version that we can test before realising it to the public?

jesserockz, 2024-08-04:
> Looks great. How does the process work? Will this be a dev version that we can test before realising it to the public?

You can test the PR using [External Components](https://esphome.io/components/external_components.html)

Please confirm on the PR itself once tested.

closed 2024-08-30 by jesserockz

mperone, 2024-09-19:
Thanks for catching the original issue and finding curves that work better. I'm just coming off a heart attack from seeing all my sensor readings spike in the house before realizing that I had updated ESPHome a few hrs prior and following the thread here.

Anyway, wanted to chip in a thought.... your proposed updates (I'll use CO but same applies to all) use the curve you fitted across the entire sensor reading range, but then you applied piece-wise. This leaves awkward discontinuities, for instance in your CO logic:

> if (red_f > 3.4f) {
>  CO = 0.0;
> }
> else if (red_f < 0.01) {
>   CO = 1000.0;
> }
> else {
>   CO = 4.2/ pow(red_f,1.2);
> }

We get the result that a reading of more than 3.4 gets 0, but something very close to 3.4 jumps up to 4.2. That's actually my new baseline for CO it seems since I guess my sensor is reading something very close to 3.4. In fact, I'm getting the threshold value in your GAS = Threshold/ pow(red_f,EXP) formula for every gas off the MICS sensor now (ie Ethanol is reading 1.5, NO2 is reading .16, etc...)

erkr, 2024-09-20:
I think there is also an initialisation issue. The math should be correct. When red_f equals 3.4 then pow(red_f,1,2) = 4.34. So the CO=4.3/4.34 = 0.99. For lower ref_f values the CO will become higher

mperone, 2024-09-20:
Hi erkr - thanks for taking a look.... somehow I got got it backwards in my head and was calculating for red_f = 0 at the break. Will delete the original comment so as not to confuse.

Ariosinm, 2024-09-25:
Hi everyone, I don't know if your update is the cause o what may have happened but since new update, my MICS4514 reports all the sensors as nan. 
In example:
`
[18:51:46][D][sensor:094]: 'MICS4514 Carbon Monoxide': Sending state nan ppm with 2 decimals of accuracy`

TrevorSchirmer, 2024-10-03:
@Ariosinm 
Hi ,

Are you still having trouble with nan results? Your config looks very similar to ours. You are using the DFRobot MiCS-4514 sensor or a different style of MiCS-4514?

 - platform: mics_4514
    id: mics4514
    nitrogen_dioxide:
      name: Nitrogen Dioxide
    carbon_monoxide:
      name: Carbon Monoxide
    hydrogen:
      name: Hydrogen
    ethanol:
      name: Ethanol
    methane:
      name: Methane
    ammonia:
      name: Ammonia
    update_interval: 10s