esphome/issues #6107: After update to 2024.7.3 Sensor doesn't inherit from pulse_meter::PulseMeterSensor. Please double check your ID is pointing to the correct value.

anamonta, 2024-08-01:
### The problem

Updated ESPHOME in HA to latest version(2024.7.3) from previous. Most of ESP's updated nicely but my water meter does not wan't to upgrade and I cannot find the problem. Code has not been changed between the versions. I am very new to yaml so the global reading and writing has been copied from the webs.

The error I receive:  ID 'water_total' of type sensor::Sensor doesn't inherit from pulse_meter::PulseMeterSensor. Please double check your ID is pointing to the correct value.

From script window I highlight the id (water_total) with mouse and it is the same (in writing etc) on sensor and the global reading.

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.4

### What platform are you using?

ESP32

### Component causing the issue

pulse_meter

### Example YAML snippet

```yaml
`sensor:
  - platform: pulse_meter
    pin: GPIO19
    unit_of_measurement: "L/min"
    id: water_pulsecounter
    timeout: 10s
    name: "Water flow"
    filters:
      multiply: 0.1  # (1/10 pulses per l)
    total:
      unit_of_measurement: 'L'
      name: 'Total water usage'
      id: water_total
      filters:
        multiply: 0.1  # (1/10 pulses per l)

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO18
      inverted: true
    name: "Suodattimen pesu"
    device_class: running

#Let's try to save things in to globals

globals:
  - id: global_watertotal
    type: int
    restore_value: true

esphome:
  name: water-measurement
  friendly_name: water-measurement
  on_boot:
    then:
      - pulse_meter.set_total_pulses:
          id: water_total
          value: !lambda 'return id(global_watertotal);'
  on_shutdown:
    then:
      - globals.set:
          id: global_watertotal
          value: !lambda 'return id(water_total).state;'
interval:
  - interval: 10min
    then:
      - globals.set:
          id: global_watertotal
          value: !lambda 'return id(water_total).state;'
# Set pulse counter total from home assistant using this service call:
api:
  encryption:
    key: "somethings here"
  services:
    - service: set_total_water_usage
      variables:
        water_meter_reading: int
      then:
        - pulse_meter.set_total_pulses:
            id: water_total
            value: !lambda 'return water_meter_reading;'`
```

### Anything in the logs that might be useful for us?

```txt
`INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/vesimittari.yaml...
Failed config

esphome: None
  name: water-measurement
  friendly_name: water-measurement
  on_boot: 
    - then: 
        - pulse_meter.set_total_pulses: 
            
            ID 'water_total' of type sensor::Sensor doesn't inherit from pulse_meter::PulseMeterSensor. Please double check your ID is pointing to the correct value.
            id: water_total
            value: !lambda |-
              return id(global_watertotal);
      priority: 600.0
  on_shutdown: 
    - then: 
        - globals.set: 
api: [source /config/esphome/vesimittari.yaml:82]
  encryption: 
    key: hbpczNwThf040lprIHvfm+2KngmpSdnSsDQkSB2CDlg=
  services: 
    - service: set_total_water_usage
      variables: 
        water_meter_reading: int
      then: 
        - pulse_meter.set_total_pulses: 
            
            ID 'water_total' of type sensor::Sensor doesn't inherit from pulse_meter::PulseMeterSensor. Please double check your ID is pointing to the correct value.
            id: water_total
            value: !lambda |-
              return water_meter_reading;
  port: 6053`
```

ssieb, 2024-08-01:
The pulse meter is `water_pulsecounter`, so that's what you have to reference, not the sensor.

closed 2024-08-01 by ssieb

anamonta, 2024-08-01:
> The pulse meter is `water_pulsecounter`, so that's what you have to reference, not the sensor.

You are true! Thanks for fast and correct answer.
Now I just need to understand how the script has worked in the past... Anyhow this problem can be closed, sorry for all!