esphome/issues #6108: Error compiling with seeed_mr24hpc1

TomArm, 2024-08-01:
### The problem

I am having an issue compiling with the seeed studio mr24hpc1 mm wave radar sensor (*seeed_mr24hpc1* component). 

Compilation errors are raised about the class not having the correct named members (log attached)

I have tried (and got the same error)
* building for an esp32 board as well as the target D1 mini
* removing and re-installing the ESPHome HA add on
* a new docker image on a different non HA host ( ghcr.io/esphome/esphome:latest )

I checked the build folder by on the HA addon instance. The following files are present:
```
root@5c53de3b-esphome:/data/build/esphome-web-3c87db/src/esphome/components/seeed_mr24hpc1$ ls -l
total 56
-rw-r--r-- 1 root root 39394 Aug  1 11:21 seeed_mr24hpc1.cpp
-rw-r--r-- 1 root root  7052 Aug  1 11:21 seeed_mr24hpc1.h
-rw-r--r-- 1 root root  7011 Aug  1 11:21 seeed_mr24hpc1_constants.h
```
Is it odd that the folder has fewer files than the github folder( esphome/components/seeed_mr24hpc1)?

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.4

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

seeed_mr24hpc1

### Example YAML snippet

```yaml
esphome:
  name: test
  friendly_name: test

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "****"

ota:
  - platform: esphome
    password: "*****"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
[... 413 lines cut ...]
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp: In member function 'void esphome::seeed_mr24hpc1::MR24HPC1Component::set_motion_trigger_time(uint8_t)':
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:853:14: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'custom_mode_num_sensor_'
  853 |   if ((this->custom_mode_num_sensor_ != nullptr) && (this->custom_mode_num_sensor_->state == 0))
      |              ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:853:60: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'custom_mode_num_sensor_'
  853 |   if ((this->custom_mode_num_sensor_ != nullptr) && (this->custom_mode_num_sensor_->state == 0))
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp: In member function 'void esphome::seeed_mr24hpc1::MR24HPC1Component::set_motion_to_rest_time(uint16_t)':
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:863:14: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'custom_mode_num_sensor_'
  863 |   if ((this->custom_mode_num_sensor_ != nullptr) && (this->custom_mode_num_sensor_->state == 0))
      |              ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:863:60: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'custom_mode_num_sensor_'
  863 |   if ((this->custom_mode_num_sensor_ != nullptr) && (this->custom_mode_num_sensor_->state == 0))
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp: In member function 'void esphome::seeed_mr24hpc1::MR24HPC1Component::set_custom_unman_time(uint16_t)':
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:875:14: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'custom_mode_num_sensor_'
  875 |   if ((this->custom_mode_num_sensor_ != nullptr) && (this->custom_mode_num_sensor_->state == 0))
      |              ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:875:60: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'custom_mode_num_sensor_'
  875 |   if ((this->custom_mode_num_sensor_ != nullptr) && (this->custom_mode_num_sensor_->state == 0))
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/test/src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp.o] Error 1
========================== [FAILED] Took 4.26 seconds ==========================
```

TomArm, 2024-08-01:
Solved it.

I didn't realise that all the configuration sections were mandatory. 
The code below (copied from the docs) compiles fine.

```
esphome:
  name: test
  friendly_name: test

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "*****"

ota:
  - platform: esphome
    password: "*****"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "*****"

captive_portal:

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: 4 #D6
  tx_pin: 5 #  D5
  parity: NONE
  stop_bits: 1

seeed_mr24hpc1:

sensor:
  - platform: seeed_mr24hpc1
    custom_presence_of_detection:
      name: "Static Distance"
    movement_signs:
      name: "Body Movement Parameter"
    custom_motion_distance:
      name: "Motion Distance"
    custom_spatial_static_value:
      name: "Existence Energy"
    custom_spatial_motion_value:
      name: "Motion Energy"
    custom_motion_speed:
      name: "Motion Speed"
    custom_mode_num:
      name: "Current Custom Mode"

switch:
  - platform: seeed_mr24hpc1
    underlying_open_function:
      name: Underlying Open Function Info Output Switch

number:
  - platform: seeed_mr24hpc1
    sensitivity:
      name: "Sensitivity"
    custom_mode:
      name: "Custom Mode"
    existence_threshold:
      name: "Existence Energy Threshold"
    motion_threshold:
      name: "Motion Energy Threshold"
    motion_trigger:
      name: "Motion Trigger Time"
    motion_to_rest:
      name: "Motion To Rest Time"
    custom_unman_time:
      name: "Time For Entering No Person State (Custom Mode)"

button:
  - platform: seeed_mr24hpc1
    restart:
      name: "Module Restart"
    custom_set_end:
      name: "End Of Custom Mode Settings"

text_sensor:
  - platform: seeed_mr24hpc1
    heart_beat:
      name: "Heartbeat"
    product_model:
      name: "Product Model"
    product_id:
      name: "Product ID"
    hardware_model:
      name: "Hardware Model"
    hardware_version:
      name: "Hardware Version"
    keep_away:
      name: "Active Reporting Of Proximity"
    motion_status:
      name: "Motion Information"
    custom_mode_end:
      name: "Custom Mode Status"

select:
  - platform: seeed_mr24hpc1
    scene_mode:
      name: "Scene"
    unman_time:
      name: "Time For Entering No Person State (Standard Function)"
    existence_boundary:
      name: "Existence Boundary"
    motion_boundary:
      name: "Motion Boundary"   

binary_sensor:
  - platform: seeed_mr24hpc1
    has_target:
      name: "Presence Information" 
```

closed 2024-08-01 by TomArm

ssieb, 2024-08-02:
No, that's a bug in the component.  You do not have to have all those sections.

arnie580, 2024-08-05:
I'm also having this issue, RPi Pico. It works if all sections are included only.

politas, 2025-02-22:
This is still happening. I have also had to add all configuration items in order to compile.

Rudd-O, 2025-03-09:
MR60BHA2 has the same problem https://github.com/limengdu/MR60BHA2_ESPHome_external_components/issues/11

binary sensor platform should be included in this and the other component.