esphome/issues #6110: ESPHome AS3935 Lightning sensor Component - incorrect variable range restrictions

kixtarter, 2024-08-01:
### The problem

The current ESPHome A3935 Component does not allow the full range of some AS3935 settings, restricting the possible range of settings. Due to this issue the maximum sensitivity setting of the AS3935 cannot be used.
In the [AS3935 Datasheet](https://www.embeddedadventures.com/datasheets/AS3935_Datasheet_EN_v2.pdf) the noise_level variable has a possible range of 0 to 7, but in the ESPHome component it can only be set from 1 to 7, missing the most sensitive setting of 0.
Similar, in the AS3935 Datasheet the spike_rejection variable has a possible range of 0 to 11, but in the ESPHome component it can only be set from 1 to 11.
Finaly, in the AS3935 Datasheet the watchdog_threshold variable has a possible range of 0 to 10, but in the ESPHome component it can only be set from 1 to 10.

This is a snippet from the current code   [__init__.py](https://github.com/esphome/esphome/blob/dev/esphome/components/as3935/__init__.py):

```
AS3935_SCHEMA = cv.Schema(
    {
        cv.GenerateID(): cv.declare_id(AS3935),
        cv.Required(CONF_IRQ_PIN): pins.gpio_input_pin_schema,
        cv.Optional(CONF_INDOOR, default=True): cv.boolean,
        cv.Optional(CONF_NOISE_LEVEL, default=2): cv.int_range(min=1, max=7),
        cv.Optional(CONF_WATCHDOG_THRESHOLD, default=2): cv.int_range(min=1, max=10),
        cv.Optional(CONF_SPIKE_REJECTION, default=2): cv.int_range(min=1, max=11),
        cv.Optional(CONF_LIGHTNING_THRESHOLD, default=1): cv.one_of(
            1, 5, 9, 16, int=True
        ),
        cv.Optional(CONF_MASK_DISTURBER, default=False): cv.boolean,
        cv.Optional(CONF_DIV_RATIO, default=0): cv.one_of(0, 16, 32, 64, 128, int=True),
        cv.Optional(CONF_CAPACITANCE, default=0): cv.int_range(min=0, max=15),
        cv.Optional(CONF_TUNE_ANTENNA, default=False): cv.boolean,
        cv.Optional(CONF_CALIBRATION, default=True): cv.boolean,
    }
)
```
This should probably be:

```
AS3935_SCHEMA = cv.Schema(
    {
        cv.GenerateID(): cv.declare_id(AS3935),
        cv.Required(CONF_IRQ_PIN): pins.gpio_input_pin_schema,
        cv.Optional(CONF_INDOOR, default=True): cv.boolean,
        cv.Optional(CONF_NOISE_LEVEL, default=2): cv.int_range(min=0, max=7),
        cv.Optional(CONF_WATCHDOG_THRESHOLD, default=2): cv.int_range(min=0, max=10),
        cv.Optional(CONF_SPIKE_REJECTION, default=2): cv.int_range(min=0, max=11),
        cv.Optional(CONF_LIGHTNING_THRESHOLD, default=1): cv.one_of(
            1, 5, 9, 16, int=True
        ),
        cv.Optional(CONF_MASK_DISTURBER, default=False): cv.boolean,
        cv.Optional(CONF_DIV_RATIO, default=0): cv.one_of(0, 16, 32, 64, 128, int=True),
        cv.Optional(CONF_CAPACITANCE, default=0): cv.int_range(min=0, max=15),
        cv.Optional(CONF_TUNE_ANTENNA, default=False): cv.boolean,
        cv.Optional(CONF_CALIBRATION, default=True): cv.boolean,
    }
)
```
This is a snippet from the current code  [as3935.cpp](https://github.com/esphome/esphome/blob/dev/esphome/components/as3935/as3935.cpp):

```
// REG0x01, bits[3:0], manufacturer default: 0010 (2).
// This setting determines the threshold for events that trigger the
// IRQ Pin.
void AS3935Component::write_watchdog_threshold(uint8_t watchdog_threshold) {
  ESP_LOGV(TAG, "Setting watchdog sensitivity to %d", watchdog_threshold);
  if ((watchdog_threshold < 1) || (watchdog_threshold > 10))  // 10 is the max sensitivity setting
    return;
  this->write_register(THRESHOLD, THRESH_MASK, watchdog_threshold, 0);
}

// REG0x01, bits [6:4], manufacturer default: 010 (2).
// The noise floor level is compared to a known reference voltage. If this
// level is exceeded the chip will issue an interrupt to the IRQ pin,
// broadcasting that it can not operate properly due to noise (INT_NH).
// Check datasheet for specific noise level tolerances when setting this register.
void AS3935Component::write_noise_level(uint8_t noise_level) {
  ESP_LOGV(TAG, "Setting noise level to %d", noise_level);
  if ((noise_level < 1) || (noise_level > 7))
    return;

  this->write_register(THRESHOLD, NOISE_FLOOR_MASK, noise_level, 4);
}
// REG0x02, bits [3:0], manufacturer default: 0010 (2).
// This setting, like the watchdog threshold, can help determine between false
// events and actual lightning. The shape of the spike is analyzed during the
// chip's signal validation routine. Increasing this value increases robustness
// at the cost of sensitivity to distant events.
void AS3935Component::write_spike_rejection(uint8_t spike_rejection) {
  ESP_LOGV(TAG, "Setting spike rejection to %d", spike_rejection);
  if ((spike_rejection < 1) || (spike_rejection > 11))
    return;

  this->write_register(LIGHTNING_REG, SPIKE_MASK, spike_rejection, 0);
}
```
This should probably be:

```
// REG0x01, bits[3:0], manufacturer default: 0010 (2).
// This setting determines the threshold for events that trigger the
// IRQ Pin.
void AS3935Component::write_watchdog_threshold(uint8_t watchdog_threshold) {
  ESP_LOGV(TAG, "Setting watchdog sensitivity to %d", watchdog_threshold);
  if ((watchdog_threshold < 0) || (watchdog_threshold > 10))  // 10 is the max sensitivity setting
    return;
  this->write_register(THRESHOLD, THRESH_MASK, watchdog_threshold, 0);
}

// REG0x01, bits [6:4], manufacturer default: 010 (2).
// The noise floor level is compared to a known reference voltage. If this
// level is exceeded the chip will issue an interrupt to the IRQ pin,
// broadcasting that it can not operate properly due to noise (INT_NH).
// Check datasheet for specific noise level tolerances when setting this register.
void AS3935Component::write_noise_level(uint8_t noise_level) {
  ESP_LOGV(TAG, "Setting noise level to %d", noise_level);
  if ((noise_level < 0) || (noise_level > 7))
    return;

  this->write_register(THRESHOLD, NOISE_FLOOR_MASK, noise_level, 4);
}
// REG0x02, bits [3:0], manufacturer default: 0010 (2).
// This setting, like the watchdog threshold, can help determine between false
// events and actual lightning. The shape of the spike is analyzed during the
// chip's signal validation routine. Increasing this value increases robustness
// at the cost of sensitivity to distant events.
void AS3935Component::write_spike_rejection(uint8_t spike_rejection) {
  ESP_LOGV(TAG, "Setting spike rejection to %d", spike_rejection);
  if ((spike_rejection < 0) || (spike_rejection > 11))
    return;

  this->write_register(LIGHTNING_REG, SPIKE_MASK, spike_rejection, 0);
}

```

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

All, up to version 2024.7.4

### What platform are you using?

ESP32

### Board

Wemos D1 Mini V4, Wemos S2 Mini

### Component causing the issue

AS3935 I2C

### Example YAML snippet

```yaml
# AS3935 Lightning sensor Setup
as3935_i2c:
  address: 0x03
  irq_pin: GPIO14
  indoor: false              #Default: true
  noise_level: 1             #Default: 2 | Range: 0 - 7   | Currently: 1 - 7
  spike_rejection: 1         #Default: 2 | Range: 0 - 11  | Currently: 1 - 11
  lightning_threshold: 1     #Default: 1 | Range: 1, 5, 9, 16
  mask_disturber: false      #Default: false
  div_ratio: 0               #Default: 0 | Range: 0 - 3
  capacitance: 0             #Default: 0 | Range: 0 - 15
  watchdog_threshold: 1      #Default: 2 | Range: 0 - 10  | Currently: 1 - 10
  tune_antenna: false        #Default: false
  calibration: true          #Default: true
```

### Additional information

See this Home Assistant forum post for more background info about this issue:
[AS3935 Lightning sensor - some possible coding issues](https://community.home-assistant.io/t/as3935-lightning-sensor-some-possible-coding-issues/753252)

kixtarter, 2024-11-30:
With ESPHome version 2024.11.0 this issue still exists, so it would be nice if it could be taken up by the applicable code owners.

BliXem1, 2025-04-26:
Did you modify the code and does it work better?