esphome/issues #6117: ESP32-Cam http "refused to connect" after ESPHome update

ZhuDaHai, 2024-08-04:
### The problem

I was running ESP32-CAM on ESPHome with an old version of ESPHome. It was working properly.
I upgraded EPSHome to  2024.7.3, which resulted in EPSHome complaining about my ESP devices needed an update. 
After updating my 4 other devices, I then updated my ESP32-Cam. 
The update was a success. Logs connect and show output, but I can no longer connect to it via is http URL. I get "refused to connect" when trying the URL in my web browser and TinyCam cannot connect as well. Generic Camera in HA cannot connect as well.

### Which version of ESPHome has the issue?

2024..7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core: 2024.7.4, Supervisor: 2024.06.2, OS: 12.4: Frontend: 202407010.0

### What platform are you using?

ESP32

### Board

ESP32-CAM

### Component causing the issue

esp32_camera ??? IDK

### Example YAML snippet

```yaml
esphome:
  name: esp32-cam
  friendly_name: ESP32-Cam

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
#  encryption:
#    key: "T0qrCJJ9HoWm9yhPwn9gzXocAvmYRWCyAkFTsmQVH/U="

ota:
#  password: "c6083ee9877b850b9b6582cabb663c9b"
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 172.27.3.56
    gateway: 172.27.3.3
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Cam Fallback Hotspot"
    password: "JzbBlLm22Olx"

captive_portal:
[... 397 lines cut ...]
[12:23:22][C][esp32_camera:142]:   Auto Exposure Level: 1
[12:23:22][C][esp32_camera:143]:   Auto Exposure Value: 300
[12:23:22][C][esp32_camera:144]:   AGC: 1
[12:23:22][C][esp32_camera:145]:   AGC Gain: 0
[12:23:22][C][esp32_camera:146]:   Gain Ceiling: 1
[12:23:22][C][esp32_camera:152]:   Test Pattern: NO
[12:23:22][C][psram:020]: PSRAM:
[12:23:22][C][psram:021]:   Available: YES
[12:23:22][C][psram:024]:   Size: 4095 KB
[12:23:22][C][captive_portal:088]: Captive Portal:
[12:23:22][C][mdns:116]: mDNS:
[12:23:22][C][mdns:117]:   Hostname: esp32-cam
[12:23:22][C][esphome.ota:073]: Over-The-Air updates:
[12:23:22][C][esphome.ota:074]:   Address: 172.27.3.57:3232
[12:23:22][C][esphome.ota:075]:   Version: 2
[12:23:22][C][safe_mode:018]: Safe Mode:
[12:23:22][C][safe_mode:020]:   Boot considered successful after 60 seconds
[12:23:22][C][safe_mode:021]:   Invoke after 10 boot attempts
[12:23:22][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[12:23:22][C][api:139]: API Server:
[12:23:22][C][api:140]:   Address: 172.27.3.57:6053
[12:23:22][C][api:144]:   Using noise encryption: NO
[12:23:30][D][esp32_camera:196]: Got Image: len=18795
[12:23:40][D][esp32_camera:196]: Got Image: len=18866
[12:23:50][D][esp32_camera:196]: Got Image: len=18893

ZhuDaHai, 2024-08-05:
Update:
1. Changed the esp32: -> board: esp32cam. That resulted in a esp32_camera failure error 
`[12:57:39][E][esp32_camera:123]:   Setup Failed: ESP_ERR_NOT_SUPPORTED
[12:57:39][E][component:082]:   Component esp32_camera is marked FAILED`

2. set "board:" back to esp32dev - same error
3. Broke out the Arduino IDE and reflashed with the original code from Rui Santos:
https://randomnerdtutorials.com/esp32-cam-video-streaming-web-server-camera-home-assistant/
and that WORKED! VIDEO!!!

4. Reflashed the board via the ESPHome Web UI flasher, first with the Initial Install, and then with the compiled BIN file from ESPHome Addon. and......wait for it....
DEAD!!! NO VIDEO

So, yeah, this worked with a 1-year old version of ESPHome (compile, upload, VIDEO!!!). But its borked with the current version (2024.7.3)

Let me know if you need any further information.

Cheers

RubenKelevra, 2024-09-09:
I've tried to update my ESPCam from a 2021 firmware to the latest one and it now says:

```
[18:35:21][E][esp32_camera:123]:   Setup Failed: ESP_ERR_NOT_SUPPORTED
[18:35:21][E][component:082]:   Component esp32_camera is marked FAILED
```

Which I think is the same issue you encountered.

There's an older bug which a bot auto-closed - which therefore never got fixed 🙄 - here: https://github.com/esphome/issues/issues/4452

My config:

```
esphome:
  name: $devicename
  platform: ESP32
  board: esp32dev
  friendly_name: $friendly_name

substitutions:
  devicename: esp32-cam-1
  friendly_name: camera-1

# Enable logging
logger:

# Enable Home Assistant API
api:

mdns:

time:
  - platform: homeassistant

ota:
  platform: esphome
  password: !secret ota_password_esp32-cam-1

wifi:
  [...]

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  name: mycamera
  resolution: VGA
  max_framerate: 1.0 fps
  idle_framerate: 1 fps
  jpeg_quality: 10
  vertical_flip: false
  horizontal_mirror: false

output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
  - platform: gpio
    pin: GPIO33
    id: gpio_33

light:
  - platform: binary
    output: gpio_4
    name: "$friendly_name flashlight"
  - platform: binary
    output: gpio_33
    name: "$friendly_name red LED"
```

Moe555, 2024-11-09:
I have the same or at least a similar issue.

I have two ESPcam boards and I don't get any of them integrated via ESPhome.
I tried different approaches, but none of them worked.

When connecting the ESP via USB and checking the logs using the Webdashboard, 
I can see that the ESP starts up fine, connects to wifi and collects snapshots, quite similar to the log shown by the op,
but it doesn't show up in ESPhome, no adoption request or similar.

When I start the process from ESPhome (after initial setup via Webdashboard) then it always looses connection after the actual espcam config is uploaded (which doesn't contain anything special).

I didn't try to use the vanilla approach w/o esphome, to verify the boards, 
but this will be the next step and I expect the boards to work fine.

So I guess that's general bug?
Or was anyone able to fix the issue?

Jaggana69, 2025-04-18:
Same issue for me!

MiguelCosta, 2025-05-04:
I have the same issue after upgrade to the last version :(

DerEmder, 2025-05-30:
Additional Info: I flashed Tasmota on it and it worked fine with framerates up to 25fps - just for testing. Flashed it back to ESPHome and did not work at all.

DerEmder, 2025-06-03:
Updating to 2025.5.2 did not change anything - still no picture. Please fix. If you need any specific information don't hesitate to ask (and tell me exactly what I have to do to gain the information).

DerEmder, 2025-06-04:
The odd stuff is that it gets some image from the camera now and then but still does not produce an almost constant video stream. I assume the camera is initialized but reading the frames seem sto be broken somehow.

Log:

```
INFO ESPHome 2025.5.2
INFO Reading configuration /config/esphome/max.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 10.42.0.69 using esphome API
INFO Successfully connected to max @ 10.42.0.69 in 0.034s
INFO Successful handshake with max @ 10.42.0.69 in 0.132s
[21:16:28][I][app:115]: ESPHome version 2025.5.2 compiled on Jun  3 2025, 13:16:53
[21:16:28][C][wifi:600]: WiFi:
[21:16:28][C][wifi:428]:   Local MAC: 
[21:16:28][C][wifi:433]:   SSID: 
[21:16:28][C][wifi:436]:   IP Address: 10.42.0.69
[21:16:28][C][wifi:440]:   BSSID: 
[21:16:28][C][wifi:441]:   Hostname: 'max'
[21:16:28][C][wifi:443]:   Signal strength: -62 dB ▂▄▆█
[21:16:28][C][wifi:447]:   Channel: 11
[21:16:28][C][wifi:448]:   Subnet: 255.255.255.0
[21:16:28][C][wifi:449]:   Gateway: 10.42.0.1
[21:16:28][C][wifi:450]:   DNS1: 10.42.0.1
[21:16:28][C][wifi:451]:   DNS2: 0.0.0.0
[21:16:28][C][logger:224]: Logger:
[21:16:28][C][logger:225]:   Max Level: DEBUG
[21:16:28][C][logger:226]:   Initial Level: DEBUG
[21:16:28][C][logger:228]:   Log Baud Rate: 0
[21:16:28][C][logger:229]:   Hardware UART: UART0
[21:16:28][C][logger:233]:   Task Log Buffer Size: 768
[21:16:28][C][i2c.arduino:071]: I2C Bus:
[21:16:28][C][i2c.arduino:072]:   SDA Pin: GPIO14
[21:16:28][C][i2c.arduino:073]:   SCL Pin: GPIO13
[21:16:28][C][i2c.arduino:074]:   Frequency: 10000 Hz
[21:16:28][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[21:16:28][C][uart.arduino_esp32:151]: UART Bus 0:
[21:16:28][C][uart.arduino_esp32:152]:   TX Pin: GPIO1
[21:16:28][C][uart.arduino_esp32:153]:   RX Pin: GPIO3
[21:16:28][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[21:16:28][C][uart.arduino_esp32:157]:   Baud Rate: 256000 baud
[21:16:28][C][uart.arduino_esp32:158]:   Data Bits: 8
[21:16:28][C][uart.arduino_esp32:159]:   Parity: NONE
[21:16:28][C][uart.arduino_esp32:160]:   Stop bits: 1
[21:16:28][C][ledc.output:180]: LEDC Output:
[21:16:28][C][ledc.output:181]:   Pin GPIO4
[21:16:28][C][ledc.output:182]:   LEDC Channel: 2
[21:16:28][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[21:16:28][C][ledc.output:184]:   Phase angle: 0.0°
[21:16:28][C][ledc.output:185]:   Bit depth: 16
[21:16:28][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Max Motion'
[21:16:28][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[21:16:28][C][gpio.binary_sensor:016]:   Pin: GPIO16
[21:16:28][C][light:092]: Light 'Max LED'
[21:16:28][C][light:094]:   Default Transition Length: 1.0s
[21:16:28][C][light:095]:   Gamma Correct: 2.80
[21:16:28][C][esp32_camera:049]: ESP32 Camera:
[21:16:28][C][esp32_camera:050]:   Name: Max Camera
[21:16:28][C][esp32_camera:051]:   Internal: NO
[21:16:28][C][esp32_camera:053]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[21:16:28][C][esp32_camera:054]:   VSYNC Pin: 25
[21:16:28][C][esp32_camera:055]:   HREF Pin: 23
[21:16:28][C][esp32_camera:056]:   Pixel Clock Pin: 22
[21:16:28][C][esp32_camera:057]:   External Clock: Pin:0 Frequency:12000000
[21:16:28][C][esp32_camera:058]:   I2C Pins: SDA:26 SCL:27
[21:16:28][C][esp32_camera:059]:   Reset Pin: -1
[... 266 lines cut ...]
[21:16:29][C][mdns:120]: mDNS:
[21:16:29][C][mdns:121]:   Hostname: max
[21:16:29][C][esphome.ota:073]: Over-The-Air updates:
[21:16:29][C][esphome.ota:074]:   Address: 10.42.0.69:3232
[21:16:29][C][esphome.ota:075]:   Version: 2
[21:16:29][C][esphome.ota:078]:   Password configured
[21:16:29][C][safe_mode:018]: Safe Mode:
[21:16:29][C][safe_mode:020]:   Boot considered successful after 60 seconds
[21:16:29][C][safe_mode:021]:   Invoke after 10 boot attempts
[21:16:29][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[21:16:29][C][api:170]: API Server:
[21:16:29][C][api:171]:   Address: 10.42.0.69:6053
[21:16:29][C][api:173]:   Using noise encryption: YES
[21:16:29][D][sensor:094]: 'Max Still Energy': Sending state 2.00000 % with 0 decimals of accuracy
[21:16:31][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
**[21:16:34][D][esp32_camera:193]: Got Image: len=20693**
[21:16:35][D][sensor:094]: 'Max Still Energy': Sending state 2.00000 % with 0 decimals of accuracy
[21:16:36][D][htu21d:058]: Got Temperature=38.7°C
[21:16:36][D][sensor:094]: 'Max Temp': Sending state 38.67193 °C with 1 decimals of accuracy
[21:16:36][D][htu21d:079]: Got Humidity=24.4%
[21:16:36][D][sensor:094]: 'Max Hum': Sending state 24.39551 % with 1 decimals of accuracy
[21:16:36][D][htu21d:098]: Heater Level=0
[21:16:37][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[21:16:39][D][sensor:094]: 'Max Still Energy': Sending state 2.00000 % with 0 decimals of accuracy
```

RubenKelevra, 2025-06-05:
From ESPHome 2024.7 the `esp32_camera` component no longer exposes an HTTP stream by itself. If you want to view the camera in a browser or connect via Home Assistant, add the `esp32_camera_web_server` block to your YAML configuration:

```yaml
esp32_camera:
  # your existing camera configuration

esp32_camera_web_server:
  - port: 80
    mode: stream
```

DerEmder, 2025-06-05:
Yes, I know that. But using the HTTP stream will not help - the camera itself soes not work properly. I only use the API camera stream, I don't need the HTTP stream. Beside that the ESP controllers are too weak to produce API **and** HTTP stream at the same time.

But that's - as I said already - is not the problem here. Before the update to 205.5.0 the camera worked fine via HA API. After the update not anymore. And reading through the forums and Github I am not the only one.

DerEmder, 2025-06-06:
In other words (if I understand correctly) - without the HTTP server it did not work but after enabling it the camera did work?

DerEmder, 2025-06-06:
I added it, flashed it and nope - camera did not work at all. No image. Nothing.

DerEmder, 2025-06-06:
I use Arduino, here is the full yaml (nothing special, just standard stuff), worked perfectly before the update:

```
substitutions:
  firmware: '5.529'
  devicename: max
  upper_devicename: 'Max'

esphome:
  name: $devicename
  on_boot:
    then:
      - output.set_level:
          id: flash
          level: 0%
      - delay: 500ms
      - output.turn_off: flash

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  baud_rate: 0

api:
  encryption:
    key: !secret max_api

ota:
  - platform: esphome
    password: !secret max_ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  fast_connect: true
  manual_ip:
    static_ip: 10.42.0.69
    gateway: 10.42.0.1
    subnet: 255.255.255.0
    dns1: 10.42.0.1
  ap:
    ssid: '$upper_devicename'
    password: !secret init_password

esp32_camera:
  name: '$upper_devicename Camera'
  id: cam
  external_clock:
    pin: GPIO0
    frequency: 12MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  max_framerate: 15 fps
  idle_framerate: 0.05 fps
  resolution: 800x600
  jpeg_quality: 18
  vertical_flip: false
  horizontal_mirror: false
  contrast: 0
  brightness: 0
[... 253 lines cut ...]
    factory_reset:
      name: '$upper_devicename Factory Reset'
    restart:
      name: '$upper_devicename Sensor Reboot'
    query_params:
      name: '$upper_devicename Get Params'

select:
  - platform: ld2410
    distance_resolution:
      name: '$upper_devicename Resolution'
    baud_rate:
      name: '$upper_devicename Baudrate'
    light_function:
      name: '$upper_devicename Light Function'
    out_pin_level:
      name: '$upper_devicename Level'

switch:
  - platform: ld2410
    engineering_mode:
      name: '$upper_devicename Config Mode'
    bluetooth:
      name: '$upper_devicename Bluetooth'
```

DerEmder, 2025-06-06:
Additional observations:

This is what I get addressing the http stream:

[image]

Another observation:

I can read parameters from the LD2410 but not write to it. The gate parameters are empty - though the radarsensor works perfectly. I don't know if this has anything to do with the non-working camera. 

[image]

DerEmder, 2025-06-07:
> I've then removed the old ESPHome device from HA and readded the newly flashed one, and it shows the image in HA, which previously did not work.

Can you post a yaml of the working one? 

I again tested it with Tasmota esp32-webcam - works perfectly. ESPHome - nothing.

RubenKelevra, 2025-06-08:
@DerEmder 

I think the major difference between your's and mine is, that I use `board: esp32cam` not `esp32dev` and the `framework: type: esp-idf`.

Here's the full yaml:

<details>

``` 

esphome:
  name: $devicename
  friendly_name: $friendly_name

substitutions:
  devicename: outdoor-camera
  friendly_name: Outdoor Camera

esp32:
  board: esp32cam
  framework:
    type: esp-idf
# Enable logging
logger:

# Enable Home Assistant API
api:

mdns:

time:
  - platform: homeassistant

ota:
  platform: esphome
  password: !secret ota_password_esp32-cam-1

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: !secret ip-esp32-cam-1
    gateway: !secret wifi_gateway
    subnet: !secret wifi_subnet
    dns1: !secret wifi_dns1
    #dns2: !secret wifi_dns2
  fast_connect: true
  reboot_timeout: 5min
  # output_power: 15dB  

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  name: mycamera
  resolution: VGA
  max_framerate: 1.0 fps
  idle_framerate: 1 fps
  jpeg_quality: 10
  vertical_flip: false
  horizontal_mirror: false
  agc_gain_ceiling: 128x
#  ae_level: -1
  aec_mode: auto
  aec2: false

output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
  - platform: gpio
    pin: GPIO33
    id: gpio_33

light:
  - platform: binary
    output: gpio_4
    name: "$friendly_name flashlight"
  - platform: binary
    output: gpio_33
    name: "$friendly_name red LED"

sensor:
  - platform: uptime
    name: Uptime
    entity_category: "diagnostic"
    update_interval: 300s
  - platform: wifi_signal
    name: "WiFi Signal"
    accuracy_decimals: 1
    entity_category: "diagnostic"
    update_interval: 5s
    filters:
      - exponential_moving_average:
          alpha: 0.05
          send_every: 2
          send_first_at: 2
      - or:
        - throttle: 600s
        - delta: 2

esp32_camera_web_server:
  - port: 80
    mode: stream
```

</details>

DerEmder, 2025-06-08:
I tried the different board type already - still does not work. But I will try the other framework later on. I'll let you know.

RubenKelevra, 2025-06-08:
@DerEmder remember, for switching Frameworks it's highly recommended to do this *not* via OTA. It might work, but the other framework actually requires a different layout, which is why it's recommended to push this via cable instead, so the new layout can be written.

> When switching from arduino to esp-idf, make sure to update the device with a serial cable as the partition table is different between the two frameworks as Over-the-Air Updates updates will not change the partition table.

https://esphome.io/components/bluetooth_proxy.html (not sure why that's only on the Bluetooth proxy page documented 🥲

DerEmder, 2025-06-08:
Well, that is not possible for me (or rather very complicated) because the device sits behind a wall. So I can only try it via OTA. If that does not work then it will get very messy. Maybe you can imagine how much I sweated flashing Tasmota and then back to ESPHome over-the-air (twice!).

If I brick the device I will have no other option, anyway. I'll let you know.

RubenKelevra, 2025-06-08:
@DerEmder i mean I flashed unknowingly already a couple of S3 and C3 devices, just be warned, that it might not go well :)

DerEmder, 2025-06-08:
Well, it did not go well, I bricked the device - now I have to open the wall again. Darn. That might take a while ....

Anyway, that does not explain why the cam worked well until 2025.5.0 update. I never changed anything.

DerEmder, 2025-06-08:
Okidoki, I reflashed it via USB. All I got was:

```
INFO Starting log output from 10.42.0.69 using esphome API
WARNING Can't connect to ESPHome API for max @ 10.42.0.69: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.42.0.69', port=6053))]: [Errno 111] Connect call failed ('10.42.0.69', 6053) (SocketAPIError)
INFO Trying to connect to max @ 10.42.0.69 in the background
```

Reset. Power off/on. The same. Reflashed it with Arduino framework - worked again (means: API connected) except camera and now the LD2410 is also without function. I had this before and a power-cycle reanimated it again (for whatever reason).

DerEmder, 2025-06-08:
It only works with the Arduino framework (and esp32dev). What's interesting is - the LD2410 is only working half - I can read live aparameters about distance etc. but all configuration to setup the device is empty und does not do anything (see screenshot). I wonder if the problems with the camera is somehow related to the LD2410 device. But as I said already: Everything worked fine until the 2025.5.x update, so only those devs who are responsible can answer that question.

[image]

DerEmder, 2025-06-08:
Weird: I removed the LD2410 parts (incl. UART definition) from the yaml, so there is only cam and the LED light (and my I²C bus). Did not work at all, HA API did not connect at all. I put the LD2410 parts in again, reflashed and voila - worked again. What the flying hen did they do in that update?

DerEmder, 2025-06-08:
Need help for what? And what exactly should I set up? I am no programmer nor am I familar with commits and the bells & whistles behind the matrix ;) 

You’d have to exactly tell me what to do (imagine me as a curious 7yo boy instead of a 63yo hobbyist ...)

DerEmder, 2025-06-18:
Update 2025.6.0 (sigh)

Even worse, now it is not even doing anything and no api connect. And of course no cam stream. Very disappointing. the worst part is that most of my ESP8266/ESP32 controllers refuse to work after the update. So, it will be a long evening for me again. Well done, devs.

Thats all I get:

```
INFO ESPHome 2025.6.0
INFO Reading configuration /config/esphome/max.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 10.42.0.69 using esphome API
INFO Successfully resolved max @ 10.42.0.69 in 0.000s
WARNING Can't connect to ESPHome API for max @ 10.42.0.69: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.42.0.69', port=6053))]: [Errno 111] Connect call failed ('10.42.0.69', 6053) (SocketAPIError)
INFO Trying to connect to max @ 10.42.0.69 in the background
INFO Successfully resolved max @ 10.42.0.69 in 0.000s
INFO Successfully resolved max @ 10.42.0.69 in 0.000s
INFO Successfully resolved max @ 10.42.0.69 in 0.000s
INFO Successfully resolved max @ 10.42.0.69 in 0.000s

```

DerEmder, 2025-06-18:
Restored back to 2025.5.2 because also my B2500D Controller cannot be compiled anymore (massive amount red error messages) and most of my controllers are non-functioning anymore. I don't have the nerve today trying to find a solution. I only get 

> Successfully resolved max @ 10.42.0.xx in 0.000s

in the logs of most controllers. Restart does not help.

DerEmder, 2025-06-19:
@RubenKelevra Thank you. If you need any information just let me know

DerEmder, 2025-06-29:
No response for a long time now. 

Today I updated everything (again)

Installationsmethode Home Assistant OS
Core 2025.6.3
Supervisor 2025.06.2
Operating System 15.2
Frontend 20250531.4
ESPHome 2025.6.2

Now the esp32cam doesn' work at all aynmore, no data, no api, no nothing, diddly squat, so to speak. I rebooted and disconnected the power supply. Nothing.

log (thats really all I get, the last line is now repeated endlessly):

```
INFO ESPHome 2025.6.2
INFO Reading configuration /config/esphome/max.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 10.42.0.69 using esphome API
INFO Successfully resolved max @ 10.42.0.69 in 0.000s
WARNING Can't connect to ESPHome API for max @ 10.42.0.69: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.42.0.69', port=6053))]: [Errno 111] Connect call failed ('10.42.0.69', 6053) (SocketAPIError)
INFO Trying to connect to max @ 10.42.0.69 in the background
INFO Successfully resolved max @ 10.42.0.69 in 0.000s
INFO Successfully resolved max @ 10.42.0.69 in 0.000s
INFO Successfully resolved max @ 10.42.0.69 in 0.000s
INFO Successfully resolved max @ 10.42.0.69 in 0.000s
```

DerEmder, 2025-06-29:
Another controller (ESP8266) with an LD2410 got the same problem as mentioned. After clearing the build files I got a log but no connection to the API. Instead the following message:
 
[14:51:05][W][api.connection:107]: : Socket operation failed: HANDSHAKESTATE_SETUP_FAILED errno=11

Could this have anything to do with the underlaying problem? Some incompatibility between HA API and ESPHome?

DerEmder, 2025-06-29:
Log of another ESP8266 (just a LD2410 and I²C HTU21D:

```
INFO ESPHome 2025.6.2
INFO Reading configuration /config/esphome/letta.yaml...
INFO Starting log output from 10.42.0.65 using esphome API
INFO Successfully resolved letta @ 10.42.0.65 in 0.000s
INFO Successfully connected to letta @ 10.42.0.65 in 0.045s
INFO Successful handshake with letta @ 10.42.0.65 in 3.021s
[16:52:25][I][app:137]: ESPHome version 2025.6.2 compiled on Jun 29 2025, 13:24:13
[16:52:25][C][wifi:613]: WiFi:
[16:52:25][C][wifi:434]:   Local MAC:
[16:52:25][C][wifi:439]:   SSID: 
[16:52:25][C][wifi:442]:   IP Address: 10.42.0.65
[16:52:25][C][wifi:446]:   BSSID: 
[16:52:25][C][wifi:446]:   Hostname: 'letta'
[16:52:25][C][wifi:446]:   Signal strength: -73 dB ▂▄▆█
[16:52:25][C][wifi:455]:   Channel: 11
[16:52:25][C][wifi:455]:   Subnet: 255.255.255.0
[16:52:25][C][wifi:455]:   Gateway: 10.42.0.1
[16:52:25][C][wifi:455]:   DNS1: 10.42.0.1
[16:52:25][C][wifi:455]:   DNS2: 0.0.0.0
[16:52:25][C][logger:211]: Logger:
[16:52:25][C][logger:211]:   Max Level: DEBUG
[16:52:25][C][logger:211]:   Initial Level: DEBUG
[16:52:25][C][logger:217]:   Log Baud Rate: 0
[16:52:25][C][logger:217]:   Hardware UART: UART0
[16:52:25][C][i2c.arduino:072]: I2C Bus:
[16:52:25][C][i2c.arduino:073]:   SDA Pin: GPIO12
[16:52:25][C][i2c.arduino:073]:   SCL Pin: GPIO13
[16:52:25][C][i2c.arduino:073]:   Frequency: 100000 Hz
[16:52:25][C][i2c.arduino:089]:   Recovery: bus successfully recovered
[16:52:25][I][i2c.arduino:099]: Results from bus scan:
[16:52:25][I][i2c.arduino:105]: Found device at address 0x40
[16:52:25][C][uart.arduino_esp8266:118]: UART Bus:
[16:52:25][C][uart.arduino_esp8266:119]:   TX Pin: GPIO1
[16:52:25][C][uart.arduino_esp8266:120]:   RX Pin: GPIO3
[16:52:25][C][light:092]: Light 'Letta LED'
[16:52:25][C][light:094]:   Default Transition Length: 1.0s
[16:52:25][C][light:094]:   Gamma Correct: 2.80
[16:52:25][C][ld2410:022]: LD2410:
[16:52:25][C][ld2410:024]:   TargetBinarySensor 'Letta Target'
[16:52:25][C][ld2410:024]:     Device Class: 'occupancy'
[16:52:25][C][ld2410:025]:   MovingTargetBinarySensor 'Letta Moving Target'
[16:52:25][C][ld2410:025]:     Device Class: 'motion'
[16:52:25][C][ld2410:026]:   StillTargetBinarySensor 'Letta Still Target'
[16:52:25][C][ld2410:026]:     Device Class: 'occupancy'
[16:52:25][C][ld2410:079]:   EngineeringModeSwitch 'Letta Config Mode'
[16:52:25][C][ld2410:079]:     Restore Mode: always OFF
[16:52:25][C][ld2410:087]:     Icon: 'mdi:pulse'
[16:52:25][C][ld2410:096]:     Device Class: 'switch'
[16:52:25][C][ld2410:079]:   BluetoothSwitch 'Letta Bluetooth'
[16:52:25][C][ld2410:079]:     Restore Mode: always OFF
[16:52:25][C][ld2410:087]:     Icon: 'mdi:bluetooth'
[16:52:25][C][ld2410:096]:     Device Class: 'switch'
[16:52:25][C][ld2410:034]:   ResetButton 'Letta Factory Reset'
[16:52:25][C][ld2410:075]:   Throttle_ : 1000ms
[16:52:25][C][ld2410:075]:   MAC Address : unknown
[16:52:25][C][ld2410:075]:   Firmware Version : 1.07.22082218
[16:52:25][C][version.text_sensor:021]: Version Text Sensor 'Letta Firmware'
[16:52:25][C][htu21d:032]: HTU21D:
[16:52:25][C][safe_mode.button:024]: Safe Mode Button 'Letta Safe Mode'
[16:52:25][C][safe_mode.button:024]:   Icon: 'mdi:restart-alert'
[16:52:25][C][status:034]: Status Binary Sensor 'Letta Status'
[16:52:25][C][status:034]:   Device Class: 'connectivity'
[16:52:25][C][esphome.ota:073]: Over-The-Air updates:
[16:52:25][C][esphome.ota:073]:   Address: 10.42.0.65:8266
[16:52:25][C][esphome.ota:073]:   Version: 2
[16:52:25][C][esphome.ota:080]:   Password configured
[16:52:25][D][sensor:098]: 'Letta Still Energy': Sending state 20.00000 % with 0 decimals of accuracy
[... 59 lines cut ...]
[16:53:34][D][sensor:098]: 'Letta Still Energy': Sending state 8.00000 % with 0 decimals of accuracy
[16:53:35][D][sensor:098]: 'Letta Still Energy': Sending state 7.00000 % with 0 decimals of accuracy
[16:53:36][D][sensor:098]: 'Letta Still Energy': Sending state 10.00000 % with 0 decimals of accuracy
[16:53:37][D][sensor:098]: 'Letta Still Energy': Sending state 9.00000 % with 0 decimals of accuracy
[16:53:38][D][sensor:098]: 'Letta Still Energy': Sending state 8.00000 % with 0 decimals of accuracy
[16:53:41][D][sensor:098]: 'Letta Still Energy': Sending state 7.00000 % with 0 decimals of accuracy
[16:53:42][D][sensor:098]: 'Letta Still Energy': Sending state 6.00000 % with 0 decimals of accuracy
[16:53:45][D][sensor:098]: 'Letta Still Energy': Sending state 7.00000 % with 0 decimals of accuracy
[16:53:46][D][sensor:098]: 'Letta Still Energy': Sending state 8.00000 % with 0 decimals of accuracy
[16:53:48][D][sensor:098]: 'Letta Still Energy': Sending state 13.00000 % with 0 decimals of accuracy
[16:53:49][D][sensor:098]: 'Letta Still Energy': Sending state 21.00000 % with 0 decimals of accuracy
WARNING letta @ 10.42.0.65: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for letta @ 10.42.0.65
WARNING Disconnected from API
INFO Successfully resolved letta @ 10.42.0.65 in 0.000s
WARNING Can't connect to ESPHome API for letta @ 10.42.0.65: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.42.0.65', port=6053))]: [Errno 111] Connect call failed ('10.42.0.65', 6053) (SocketAPIError)
INFO Trying to connect to letta @ 10.42.0.65 in the background
INFO Successfully resolved letta @ 10.42.0.65 in 0.000s
INFO Successfully resolved letta @ 10.42.0.65 in 0.000s
INFO Successfully resolved letta @ 10.42.0.65 in 0.000s
INFO Successfully resolved letta @ 10.42.0.65 in 0.000s
INFO Successfully resolved letta @ 10.42.0.65 in 0.000s
```

At no point in time the HA API is connected. After a while the controller resets and goes into an infinite loop with the message in the last line.  Interestingly the other eight ESP8266/ESP32 with the same hardware work just fine.

DerEmder, 2025-06-29:
I reverted back to 1025.5.2 and almost everything is working again. What a mess these updates make ...

Anyway, I now removed the whole esp32cam software so I only use teh LD2410 and the I²C Bus. All works fine except the LD2410 entities are not found anymore in HA. I had a look and found that HA added the device name twice for every entity name. See image (what a stupid bug). But thats amunition for another issue. 

As soon as I add the esp32cam component the ESP32 does not work anymore. Please fix this ....

The name of the entity was _**sensor.max_distance**_ - HA makes _**sensor.max_max_distance**_ so the entities can not be found anymore. I have to manually edit all(!) entities by hand. Again ... this sucks. 

[image]

RubenKelevra, 2025-06-30:
Thanks for all the good infos. Sorry for any delay, my spare time is just currently pretty limited and divided into multiple things going on. So don't worry, I'll look into this.

DerEmder, 2025-06-30:
Cool, thank you. Right now all is working fine - except the camera sensor. And, of course with the old version. I frooze all updates now, izt is just not worth the hassle - like your time mine is as precious as well-

RubenKelevra, 2025-06-30:
I found the potential issue in the library, which is different than expected before.

I'll create a patch and send it to the library for consideration.

DerEmder, 2025-06-30:
> So the most suspicious change between 2025.4.x and 2025.5.0 is that the ESP32-camera library was changed in commit [98470d32f](https://github.com/esphome/esphome/commit/98470d32f) from Version v2.0.9 to v2.0.15, which introduced for example a ["new idf-v5 I2C driver"](https://github.com/espressif/esp32-camera/releases/tag/v2.0.11)...
> 
> > except the camera sensor.
> 
> What's the status of the camera image/sensor on 2025.5.2? Your report is a bit confusing right now, with a lot of back and forth. :)

Well, it does not work with 2025.5.2. I know  it's a bit confusing (I myself try to keep up with all the versions). In short: Everything (incl. the camera) was working until 2025.5.0. With that update the cam stopped working. Then I updated .1 and .2 and finally 2025.6.x because I had hoped it would fix the problem. So I restored back to 2025.5.2 because that was the last version where the LD2410 was working but not the cam (i had no older backup than 2025.5.2 unfortunately). The latest update 2025.6.x broke not only the cam but the LD2410 as well (API not connected and so forth). 

Anyway, right now I am on 2025.5.2 where the LD2410 is working fine - that specific one is handling the light in the main floor of the building. That's the main function and must work, especially at night. I pass on the camera right now because that is a secondary security feature.

Between all the updates/restores I also tried Tasmota and flashed a Tasmota binary with the ESPHome OTA function. The camera itself works fine up to 25fps. I flashed back OTA with the ESPHome binary I had saved before.

I'll wait for the patch then, fingers crossed it works out. Sorry for all the confusing back and forth ...

RubenKelevra, 2025-06-30:
> Sorry for all the confusing back and forth ...

No worries.

The patch might or might not help with the issues you're seeing, but [it addresses reported issues on the library update](https://github.com/esphome/esphome/pull/8832). I'll look for the other issues past 2025.5.0 next, once that patch is filed.

DerEmder, 2025-06-30:
> [@DerEmder](https://github.com/DerEmder) Please confirm something for me:
> 
> * If you remove the camera settings completely, does the i2c attached sensor (on GPIO14 / GPIO13) work?

Yes, it does.

> * If you remove the config for the i2c attached sensor, including the `i2c:` setting, does the camera work?

I will try later and report back
 
> I suspect a conflict due to the new i2c driver, which uses the same controller (0) for both the camera and the i2c port.

Hmm, thats a good point and finally makes sense. Good point, very helpful. Was the new driver introduced with 2025.5.x?

DerEmder, 2025-06-30:
Okay, I added the cam component and removed the I²C parts completely. Cam does not work (no image). The I²C part with the HTU21D was working without issues before (with or without the cam component).

Hmm. But what you say makes sense and could've been the problem. Let's try your patch and lets see what happens.

Log:

```
INFO ESPHome 2025.5.2
INFO Reading configuration /config/esphome/max.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 10.42.0.69 using esphome API
INFO Successfully connected to max @ 10.42.0.69 in 0.077s
INFO Successful handshake with max @ 10.42.0.69 in 0.119s
[19:08:46][I][app:115]: ESPHome version 2025.5.2 compiled on Jun 30 2025, 19:05:01
[19:08:46][C][wifi:600]: WiFi:
[19:08:46][C][wifi:428]:   Local MAC: 
[19:08:46][C][wifi:433]:   SSID: 
[19:08:46][C][wifi:436]:   IP Address: 10.42.0.69
[19:08:46][C][wifi:440]:   BSSID: 
[19:08:46][C][wifi:441]:   Hostname: 'max'
[19:08:46][C][wifi:443]:   Signal strength: -58 dB ▂▄▆█
[19:08:46][C][wifi:447]:   Channel: 11
[19:08:46][C][wifi:448]:   Subnet: 255.255.255.0
[19:08:46][C][wifi:449]:   Gateway: 10.42.0.1
[19:08:46][C][wifi:450]:   DNS1: 10.42.0.1
[19:08:46][C][wifi:451]:   DNS2: 0.0.0.0
[19:08:46][C][logger:224]: Logger:
[19:08:46][C][logger:225]:   Max Level: DEBUG
[19:08:46][C][logger:226]:   Initial Level: DEBUG
[19:08:46][C][logger:228]:   Log Baud Rate: 0
[19:08:47][C][logger:229]:   Hardware UART: UART0
[19:08:47][C][logger:233]:   Task Log Buffer Size: 768
[19:08:47][C][uart.arduino_esp32:151]: UART Bus 0:
[19:08:47][C][uart.arduino_esp32:152]:   TX Pin: GPIO1
[19:08:47][C][uart.arduino_esp32:153]:   RX Pin: GPIO3
[19:08:47][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[19:08:47][C][uart.arduino_esp32:157]:   Baud Rate: 256000 baud
[19:08:47][C][uart.arduino_esp32:158]:   Data Bits: 8
[19:08:47][C][uart.arduino_esp32:159]:   Parity: NONE
[19:08:47][C][uart.arduino_esp32:160]:   Stop bits: 1
[19:08:47][D][sensor:094]: 'Max Still Energy': Sending state 4.00000 % with 0 decimals of accuracy
[19:08:47][C][ledc.output:180]: LEDC Output:
[19:08:47][C][ledc.output:181]:   Pin GPIO4
[19:08:47][C][ledc.output:182]:   LEDC Channel: 2
[19:08:47][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[19:08:47][C][ledc.output:184]:   Phase angle: 0.0°
[19:08:47][C][ledc.output:185]:   Bit depth: 16
[19:08:47][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Max Motion'
[19:08:47][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[19:08:47][C][gpio.binary_sensor:016]:   Pin: GPIO16
[19:08:47][C][light:092]: Light 'Max LED'
[19:08:47][C][light:094]:   Default Transition Length: 1.0s
[19:08:47][C][light:095]:   Gamma Correct: 2.80
[19:08:47][C][esp32_camera:049]: ESP32 Camera:
[19:08:47][C][esp32_camera:050]:   Name: Max Camera
[19:08:47][C][esp32_camera:051]:   Internal: NO
[19:08:47][C][esp32_camera:053]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[19:08:47][C][esp32_camera:054]:   VSYNC Pin: 25
[19:08:47][C][esp32_camera:055]:   HREF Pin: 23
[19:08:47][C][esp32_camera:056]:   Pixel Clock Pin: 22
[19:08:47][C][esp32_camera:057]:   External Clock: Pin:0 Frequency:10000000
[19:08:47][C][esp32_camera:058]:   I2C Pins: SDA:26 SCL:27
[19:08:47][C][esp32_camera:059]:   Reset Pin: -1
[19:08:47][C][esp32_camera:080]:   Resolution: 800x600 (SVGA)
[19:08:47][C][esp32_camera:126]:   JPEG Quality: 20
[... 211 lines cut ...]
[19:09:03][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[19:09:04][D][sensor:094]: 'Max Still Energy': Sending state 4.00000 % with 0 decimals of accuracy
[19:09:07][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[19:09:10][D][sensor:094]: 'Max Still Energy': Sending state 4.00000 % with 0 decimals of accuracy
[19:09:11][D][sensor:094]: 'Max Still Energy': Sending state 6.00000 % with 0 decimals of accuracy
[19:09:15][D][sensor:094]: 'Max Still Energy': Sending state 7.00000 % with 0 decimals of accuracy
[19:09:16][D][esp32_camera:193]: Got Image: len=17262
[19:09:17][D][sensor:094]: 'Max Still Energy': Sending state 6.00000 % with 0 decimals of accuracy
[19:09:19][D][sensor:094]: 'Max Moving Distance': Sending state 370.00000 cm with 0 decimals of accuracy
[19:09:19][D][sensor:094]: 'Max Moving Energy': Sending state 17.00000 % with 0 decimals of accuracy
[19:09:19][D][sensor:094]: 'Max Still Energy': Sending state 5.00000 % with 0 decimals of accuracy
[19:09:20][D][sensor:094]: 'Max Moving Energy': Sending state 0.00000 % with 0 decimals of accuracy
[19:09:20][D][sensor:094]: 'Max Still Energy': Sending state 7.00000 % with 0 decimals of accuracy
[19:09:21][D][sensor:094]: 'Max Still Energy': Sending state 8.00000 % with 0 decimals of accuracy
[19:09:23][D][sensor:094]: 'Max Still Energy': Sending state 7.00000 % with 0 decimals of accuracy
[19:09:24][D][sensor:094]: 'Max Still Energy': Sending state 9.00000 % with 0 decimals of accuracy
[19:09:25][D][sensor:094]: 'Max Still Energy': Sending state 12.00000 % with 0 decimals of accuracy
[... cut ...]
[19:09:27][D][sensor:094]: 'Max Still Energy': Sending state 6.00000 % with 0 decimals of accuracy
[19:09:28][D][sensor:094]: 'Max Still Energy': Sending state 4.00000 % with 0 decimals of accuracy
[19:09:29][D][sensor:094]: 'Max Still Energy': Sending state 5.00000 % with 0 decimals of accuracy
[19:09:33][D][sensor:094]: 'Max Still Energy': Sending state 6.00000 % with 0 decimals of accuracy
[19:09:36][D][sensor:094]: 'Max Still Energy': Sending state 4.00000 % with 0 decimals of accuracy
[19:09:36][D][esp32_camera:193]: Got Image: len=17202
```

DerEmder, 2025-06-30:
Additional info:

Removing/attaching power supply does not help.

DerEmder, 2025-06-30:
With the I²C component out of the way there is only one component left - the LD2410. I will remove that too and use the cam component only. Let's see if there is any change.

Here's the current yaml:

``
## 69: Max - Präsenzmelder Flur EG ##

## Version = Jahr - 2020.Monat.Tag[2]

substitutions:
  firmware: '5.630'
  devicename: max
  upper_devicename: 'Max'

esphome:
  name: $devicename

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  baud_rate: 0

api:
  encryption:
    key: !secret max_api

ota:
  - platform: esphome
    password: !secret max_ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  fast_connect: true
  manual_ip:
    static_ip: 10.42.0.69
    gateway: 10.42.0.1
    subnet: 255.255.255.0
    dns1: 10.42.0.1
  ap:
    ssid: '$upper_devicename'
    password: !secret init_password

esp32_camera:
  name: '$upper_devicename Camera'
  id: espcam
  external_clock:
    pin: GPIO0
    frequency: 10MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  max_framerate: 10 fps
  idle_framerate: 0.05 fps
  resolution: 800x600
  jpeg_quality: 20
  vertical_flip: false
  horizontal_mirror: false
  contrast: 0
  brightness: 0
  saturation: -2
[... 214 lines cut ...]
    factory_reset:
      name: '$upper_devicename Factory Reset'
    restart:
      name: '$upper_devicename Sensor Reboot'
    query_params:
      name: '$upper_devicename Get Params'

select:
  - platform: ld2410
    distance_resolution:
      name: '$upper_devicename Resolution'
    baud_rate:
      name: '$upper_devicename Baudrate'
    light_function:
      name: '$upper_devicename Light Function'
    out_pin_level:
      name: '$upper_devicename Level'

switch:
  - platform: ld2410
    engineering_mode:
      name: '$upper_devicename Config Mode'
    bluetooth:
      name: '$upper_devicename Bluetooth'
``

DerEmder, 2025-06-30:
I changed the clock back to the default of 20Mhz, still no picture but the log changed significantly. But still no image in the frontend (see screenshot).

```
INFO Starting log output from 10.42.0.69 using esphome API
INFO Successfully connected to max @ 10.42.0.69 in 4.150s
INFO Successful handshake with max @ 10.42.0.69 in 0.109s
[21:03:10][I][app:115]: ESPHome version 2025.5.2 compiled on Jun 30 2025, 21:02:07
[21:03:10][C][wifi:600]: WiFi:
[21:03:10][C][wifi:428]:   Local MAC: 
[21:03:10][C][wifi:433]:   SSID: 
[21:03:10][C][wifi:436]:   IP Address: 10.42.0.69
[21:03:10][C][wifi:440]:   BSSID: 
[21:03:10][C][wifi:441]:   Hostname: 'max'
[21:03:10][C][wifi:443]:   Signal strength: -60 dB ▂▄▆█
[21:03:10][C][wifi:447]:   Channel: 11
[21:03:10][C][wifi:448]:   Subnet: 255.255.255.0
[21:03:10][C][wifi:449]:   Gateway: 10.42.0.1
[21:03:10][C][wifi:450]:   DNS1: 10.42.0.1
[21:03:10][C][wifi:451]:   DNS2: 0.0.0.0
[21:03:10][C][logger:224]: Logger:
[21:03:10][C][logger:225]:   Max Level: DEBUG
[21:03:10][C][logger:226]:   Initial Level: DEBUG
[21:03:10][C][logger:228]:   Log Baud Rate: 0
[21:03:10][C][logger:229]:   Hardware UART: UART0
[21:03:10][C][logger:233]:   Task Log Buffer Size: 768
[21:03:10][C][uart.arduino_esp32:151]: UART Bus 0:
[21:03:10][C][uart.arduino_esp32:152]:   TX Pin: GPIO1
[21:03:10][C][uart.arduino_esp32:153]:   RX Pin: GPIO3
[21:03:10][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[21:03:10][C][uart.arduino_esp32:157]:   Baud Rate: 256000 baud
[21:03:10][C][uart.arduino_esp32:158]:   Data Bits: 8
[21:03:10][C][uart.arduino_esp32:159]:   Parity: NONE
[21:03:10][C][uart.arduino_esp32:160]:   Stop bits: 1
[21:03:10][C][ledc.output:180]: LEDC Output:
[21:03:10][C][ledc.output:181]:   Pin GPIO4
[21:03:10][C][ledc.output:182]:   LEDC Channel: 2
[21:03:10][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[21:03:10][C][ledc.output:184]:   Phase angle: 0.0°
[21:03:10][C][ledc.output:185]:   Bit depth: 16
[21:03:10][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Max Motion'
[21:03:10][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[21:03:10][C][gpio.binary_sensor:016]:   Pin: GPIO16
[21:03:10][C][light:092]: Light 'Max LED'
[21:03:10][C][light:094]:   Default Transition Length: 1.0s
[21:03:10][C][light:095]:   Gamma Correct: 2.80
[21:03:10][C][esp32_camera:049]: ESP32 Camera:
[21:03:10][C][esp32_camera:050]:   Name: Max Camera
[21:03:10][C][esp32_camera:051]:   Internal: NO
[21:03:10][C][esp32_camera:053]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[21:03:10][C][esp32_camera:054]:   VSYNC Pin: 25
[21:03:10][C][esp32_camera:055]:   HREF Pin: 23
[21:03:10][C][esp32_camera:056]:   Pixel Clock Pin: 22
[21:03:10][C][esp32_camera:057]:   External Clock: Pin:0 Frequency:20000000
[21:03:10][C][esp32_camera:058]:   I2C Pins: SDA:26 SCL:27
[21:03:10][C][esp32_camera:059]:   Reset Pin: -1
[21:03:10][C][esp32_camera:080]:   Resolution: 800x600 (SVGA)
[21:03:10][C][esp32_camera:126]:   JPEG Quality: 20
[21:03:10][C][esp32_camera:127]:   Framebuffer Count: 1
[21:03:10][C][ld2410:068]:   BluetoothSwitch 'Max Bluetooth'
[21:03:10][C][ld2410:070]:     Icon: 'mdi:bluetooth'
[21:03:10][C][ld2410:079]:     Device Class: 'switch'
[21:03:10][C][ld2410:091]:     Restore Mode: always OFF
[21:03:10][C][ld2410:034]:   ResetButton 'Max Factory Reset'
[21:03:10][C][ld2410:034]:     Icon: 'mdi:restart-alert'
[21:03:10][C][ld2410:035]:   RestartButton 'Max Sensor Reboot'
[21:03:10][C][ld2410:035]:     Icon: 'mdi:restart'
[21:03:10][C][ld2410:036]:   QueryButton 'Max Get Params'
[21:03:10][C][ld2410:036]:     Icon: 'mdi:database'
[21:03:10][C][ld2410:040]:   MovingTargetDistanceSensor 'Max Moving Distance'
[21:03:10][C][ld2410:040]:     Device Class: 'distance'
[... 381 lines cut ...]
[21:04:26][D][esp32_camera:193]: Got Image: len=15332
[21:04:26][D][esp32_camera:193]: Got Image: len=15411
[21:04:26][D][esp32_camera:193]: Got Image: len=15346
[21:04:27][D][esp32_camera:193]: Got Image: len=15380
[21:04:27][D][esp32_camera:193]: Got Image: len=15416
[21:04:27][D][esp32_camera:193]: Got Image: len=15350
[21:04:27][D][esp32_camera:193]: Got Image: len=15379
[21:04:27][D][esp32_camera:193]: Got Image: len=15382
[21:04:28][D][esp32_camera:193]: Got Image: len=15351
[21:04:28][D][esp32_camera:193]: Got Image: len=15414
[21:04:28][D][esp32_camera:193]: Got Image: len=15880
[21:04:28][D][esp32_camera:193]: Got Image: len=15332
[21:04:28][D][esp32_camera:193]: Got Image: len=15373
[21:04:29][D][esp32_camera:193]: Got Image: len=15357
[21:04:29][D][esp32_camera:193]: Got Image: len=15335
[21:04:40][D][esp32_camera:193]: Got Image: len=15348
[21:04:41][D][esp32_camera:193]: Got Image: len=15068
[21:04:42][D][esp32_camera:193]: Got Image: len=14976
[21:04:43][D][esp32_camera:193]: Got Image: len=15032
[21:04:44][D][esp32_camera:193]: Got Image: len=15124
[21:04:45][D][esp32_camera:193]: Got Image: len=15093
[21:04:46][D][sensor:094]: 'Max Still Energy': Sending state 2.00000 % with 0 decimals of accuracy

```
[image]

RubenKelevra, 2025-06-30:
For reference, here's the patch for the issue I found in esp32-camera, which can cause crashes.

https://github.com/espressif/esp32-camera/pull/758

RubenKelevra, 2025-06-30:
> With the I²C component out of the way there is only one component left - the LD2410.

I tried add an I2C component like you to my ESP32-Cam, but it refuses to break the config. I always get an image. However I did not connect an actual I2C device to the board. So this might be the reason why the camera can recover the port.

Anyway, I'll dig now in the code an see if ESPHome does handle the new i2c driver correctly or not.

Btw: There's been a new version of esp32-camera released just an hour ago. That would also be worth a shot, if I don't find anything wrong with ESPHome itself.

DerEmder, 2025-06-30:
> I tried add an I2C component like you to my ESP32-Cam, but it refuses to break the config. I always get an image. However I did not connect an actual I2C device to the board. So this might be the [...]

Thanks for the info, I will try that. But then it will be bedtime for me.

RubenKelevra, 2025-06-30:
Note that this isn't expected to fix anything but random crashes. So if you don't experience random crashes, it's not worth trying :)

DerEmder, 2025-06-30:
Compiled. Flashed. No connection to the device anymore :) But OTA works, so back to Arduino Framework and the previous version.

Thats all I got:

```
INFO ESPHome 2025.5.2
INFO Reading configuration /config/esphome/max.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 10.42.0.69 using esphome API
WARNING Can't connect to ESPHome API for max @ 10.42.0.69: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.42.0.69', port=6053))]: [Errno 111] Connect call failed ('10.42.0.69', 6053) (SocketAPIError)
INFO Trying to connect to max @ 10.42.0.69 in the background

```

RubenKelevra, 2025-07-01:
> Btw, what I observed is that the esp-idf framework (opposed to the Arduino framework) compiles to much bigger binaries.

Yeah, that's to be expected. IIRC the Arduino framework cannot correctly detect the sizes of the flash. So their image sizes will be smaller, for compatibility reasons. Using the ESP-IDF framework directly however makes the full flash available, and thus the image size is bigger, because it's the full size of the flash.

By default, it's always recommended to use ESP-IDF directly. Using Arduino is much slower, uses more memory, and due to more code involved potentially more buggy.

There are only a very few components requiring you to use Arduino, and for some, say LEDs, there are alternatives under ESP-IDF which basically work the same way.

---

Sorry I invested a couple of hours into fixing bugs in the camera driver to bring support to the PY260 Camera (M5 Unit CamS3-5MP), otherwise I would have taken a look into your HA API issues. :)

DerEmder, 2025-07-02:
Don't worry, I don't have time anyway - I have to work in the cooler basement - it is to hot upstairs during the day. My brain is working slower too. So, no, no need to rush - at the moment everything (except the cam) is working fine. Take your time.

RubenKelevra, 2025-07-10:
> stub config? What's that? Please write in layman terms :)

Yeah, if you click on the button "New Device" in the ESPHome Builder and type in a name of the new device (and select a board) you end up with a small default configuration.

This gives you a clean build environment, a new name, a new Home Assistant Encryption key and a new OTA key all at once.

Then just put a new IP in there (not your `10.42.0.69` - a different one) and copy over the other important modifications you made, like the sensors.

DerEmder, 2025-07-10:
Thanks for the info, did not know that. Anyway, I did all that already one step after the other manually, I also tried a dynamic IP. Does not work. I then removed everything about the I²C Bus and tried again. Still did not work. I get the feeling my esp32cam is somewhat broken in terms of hardware - I will have a look at the hardware again later on and report back.

RubenKelevra, 2025-07-10:
Try disconnecting the camera cable and remove the Camera config.

I noticed that sometimes the ribbon connector is not really squarely seated and is connecting to more than one pin, which can lead to all kinds of weird effects.

That's obviously only true if you use a camera module with a ribbon style connector (Flachbandkabel in German).

Hope this helps.

RubenKelevra, 2025-07-10:
@DerEmder 

So let's try to remove first everything not necessary in the config, for testing:

```
esphome:
  name: max_test                        # neuer Name ⇒ keine HA-Kollisionen

esp32:
  board: esp32cam                       # gleiche HW
  framework:
    type: esp-idf                       # Problem-Framework beibehalten
  flash_size: 4MB

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  manual_ip:                             # freie IP wählen
    static_ip: 10.42.0.179
    gateway: 10.42.0.1
    subnet: 255.255.255.0
    dns1: 10.42.0.1

logger:
  level: VERY_VERBOSE                   # komplette Ausgabe
  baud_rate: 115200                     # USB-Log

debug:                                  # Heap/Stack alle 10 s
  update_interval: 10s

api:                                    # ohne Verschlüsselung → Fehler klarer
ota:                                    # OTA drin lassen, schadet nicht

```  

Please report back if this is connecting to Home Assistant. Please note that you need to have the USB-Cable connected, to record the full log after flashing.

Please upload the log either as text file or with a link to like gist.github.com, not inline here on the ticket.

Thanks!

DerEmder, 2025-07-11:
> There's _a lot_ above ;)

Yes, there are ... 

https://github.com/esphome/issues/issues/6117#issuecomment-2994259729

> Alternatively: Just grab a different ESP32 and try to replicate the issue with it? :)

I ran out of ESPs, appr.  twenty-four ESP8266 and ESP32 installed into the building (and surroundings like my own weather station) incl. one esp32cam but I will see if I can find the second esp32cam (I know its soemwhere) in my (work)shop. The problem is they have no USB, so I will have to find my USB adapter as well. Anyway, I am on it and will report back.

DerEmder, 2025-07-11:
Well, all I got was: diddly squat. Nothing. Neither of my esp32cams worked.

Both esp32cam did not even respond to USB flashing. I give up on both devices (there is something wonky about 'em) and order a new one. If the problem still persists I will open a new issue and let you know.

DerEmder, 2025-07-11:
I did revive one of the boards. I flashed my Tasmota test binary and voila' it works.

[image]

20fps while reading the LD2410! Why can't ESPHome do it that way? I'm really flabbergasted about the difference. Anyway, I will start a new topic as soon as I am finished with teh above configuration. One thing is clear - the hardware and camera is working fine.

RubenKelevra, 2025-07-12:
Yeah ESPHome isn't using the esp32-camera very well, that's one thing I like to improve on. But first I would have to improve the esp32-camera library itself a bit, and that's what I'm working on.

DerEmder, 2025-07-13:
That's a good thing and I'll try as good as I can to help. Unfortunately I have to wait 'til monday or thuesday for my new esp32cam to arrive - the two I already have are no good anymore - one is dead as a duck and the other one tends to crash now and then and USB flashing does not work at all. I ordered a new board incl. the USB converter. Then I will continue testing.

However - the Tasmota firmware seem s to have a good driver for the cam but if I increase reolution or exposure time or shutter speed it will suffer immensly and the fps goes down to maybe 8-15fps, sometimes even less. But that's ok, it is a 240Mhz microcontroller - and a max framerate of 8-15fps is something one could expect from such a slow device. I'm actually amazed the controller can handle that kind of framerate via WiFi.

RubenKelevra, 2025-07-13:
> That's a good thing and I'll try as good as I can to help.

How's your cpp today? ;)

Jokes aside: I'll mention you once I create the PRs so you can help to test it. I've heard you got some spare hardware now :)

DerEmder, 2025-07-13:
I see, that makes sense now. It is like driving with handbrake applied or .. what you say is to release the handbrake but set a speed limit while streaming the still frames. Or in other words - set the driver free and stream with variable framerate as fast as it can (but with a max limit). 

My cpp is just fine :)  ... and very limited. Back in the day I used to program myself in ... Assembler and Machine language. For me C++ and yaml are the devils work :)

And yes, I will have a working cam for the building and one for testing. I am looking forward to it.

[1 more reply]