esphome/issues #6119: 2024.7.3 Broke Platfrom Dallas_temp

ESP32andmore, 2024-08-04:
### The problem

Release 2024.7.3, Dallas_temp on ESP32 gets "Scratch pad checksum invalid!" Works fine in 2024.6.6

13:55:05 | [W] | [component:170] | Component dallas_temp.sensor cleared Warning flag
13:55:05 | [W] | [dallas.temp.sensor:139] | 'AVfan1 Temperature 3' - Scratch pad checksum invalid!
13:55:05 | [W] | [component:157] | Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
13:55:05 | [W] | [dallas.temp.sensor:139] | 'AVfan1 Temperature 1' - Scratch pad checksum invalid!
13:55:05 | [W] | [component:157] | Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
13:55:05 | [W] | [dallas.temp.sensor:139] | 'AVfan1 Temperature 2' - Scratch pad checksum invalid!
13:55:05 | [W] | [component:157] | Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

No relevant but latest

### What platform are you using?

ESP32

### Board

ESP32andmore.com ECB

### Component causing the issue

One-wire: Dallas_temp

### Example YAML snippet

```yaml
- platform: dallas_temp
    address: ${dallasaddress1}
    update_interval: 2s
    #index: 0
    id: temp1
    filters:
      - filter_out: NaN
      - offset: 0.0
      #- lambda: return x * (9.0/5.0) + 32.0;
    unit_of_measurement: "°C"
    name: ${fdevicename} Temperature 1
```

### Anything in the logs that might be useful for us?

```txt
13:55:05 | [W] | [component:170] | Component dallas_temp.sensor cleared Warning flag
13:55:05 | [W] | [dallas.temp.sensor:139] | 'AVfan1 Temperature 3' - Scratch pad checksum invalid!
13:55:05 | [W] | [component:157] | Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
13:55:05 | [W] | [dallas.temp.sensor:139] | 'AVfan1 Temperature 1' - Scratch pad checksum invalid!
13:55:05 | [W] | [component:157] | Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
13:55:05 | [W] | [dallas.temp.sensor:139] | 'AVfan1 Temperature 2' - Scratch pad checksum invalid!
13:55:05 | [W] | [component:157] | Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
```

ssieb, 2024-08-04:
Nothing changed in .7.x.  What was the last version that worked?
What is in the logs for the `one_wire:` section?

ESP32andmore, 2024-08-05:
Samuel,
Attached are two logs from same board and sensors programmed with .6.x and .7.x back to back
.7.x , the one-wire component finds no devices:

[08:42:06][C][gpio.one_wire:020]: GPIO 1-wire bus:
[08:42:06][C][gpio.one_wire:021]:   Pin: GPIO27
[08:42:06][W][gpio.one_wire:078]:   Found no devices!

.6.x, does find the sensors:

[08:59:57][C][gpio.one_wire:020]: GPIO 1-wire bus:
[08:59:57][C][gpio.one_wire:021]:   Pin: GPIO27
[08:59:57][C][gpio.one_wire:080]:   Found devices:
[08:59:57][C][gpio.one_wire:082]:     0x8b3ca8f64935b228 (DS18B20)
[08:59:57][C][gpio.one_wire:082]:     0x783c0af6490d3628 (DS18B20)
[08:59:57][C][gpio.one_wire:082]:     0xc13c01f09642c928 (DS18B20)
[08:59:57][C][gpio.one_wire:082]:     0x043c6ef649772f28 (DS18B20)

[08:59:57][I][app:100]: ESPHome version 2024.6.6 compiled on Jul  7 2024, 23:46:33
[08:59:57][C][wifi:599]: WiFi:
[08:59:57][C][wifi:427]:   Local MAC: 08:A6:F7:54:0E:BC
[08:59:57][C][wifi:432]:   SSID: 'hill'[redacted]
[08:59:57][C][wifi:435]:   IP Address: 192.168.0.67
[08:59:57][C][wifi:439]:   BSSID: 68:7F:F0:00:80:B3[redacted]
[08:59:57][C][wifi:440]:   Hostname: 'avfan1'
[08:59:57][C][wifi:442]:   Signal strength: -53 dB ▂▄▆█
[08:59:57][C][wifi:446]:   Channel: 11
[08:59:57][C][wifi:447]:   Subnet: 255.255.255.0
[08:59:57][C][wifi:448]:   Gateway: 192.168.0.1
[08:59:57][C][wifi:449]:   DNS1: 192.168.0.1
[08:59:57][C][wifi:450]:   DNS2: 8.8.8.8
[08:59:57][C][logger:185]: Logger:
[08:59:57][C][logger:186]:   Level: DEBUG
[08:59:57][C][logger:188]:   Log Baud Rate: 115200
[08:59:57][C][logger:189]:   Hardware UART: UART0
[08:59:57][C][i2c.arduino:071]: I2C Bus:
[08:59:57][C][i2c.arduino:072]:   SDA Pin: GPIO21
[08:59:57][C][i2c.arduino:073]:   SCL Pin: GPIO22
[08:59:57][C][i2c.arduino:074]:   Frequency: 50000 Hz
[08:59:57][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[08:59:57][C][i2c.arduino:071]: I2C Bus:
[08:59:57][C][i2c.arduino:072]:   SDA Pin: GPIO16
[08:59:57][C][i2c.arduino:073]:   SCL Pin: GPIO17
[08:59:57][C][i2c.arduino:074]:   Frequency: 800000 Hz
[08:59:57][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[08:59:57][C][gpio.one_wire:020]: GPIO 1-wire bus:
[08:59:57][C][gpio.one_wire:021]:   Pin: GPIO27
[08:59:57][C][gpio.one_wire:080]:   Found devices:
[08:59:57][C][gpio.one_wire:082]:     0x8b3ca8f64935b228 (DS18B20)
[08:59:57][C][gpio.one_wire:082]:     0x783c0af6490d3628 (DS18B20)
[08:59:57][C][gpio.one_wire:082]:     0xc13c01f09642c928 (DS18B20)
[08:59:57][C][gpio.one_wire:082]:     0x043c6ef649772f28 (DS18B20)
[08:59:57][C][template.number:050]: Template Number 'AVfan1 Target Temp 1'
[08:59:57][C][template.number:050]:   Unit of Measurement: '°C'
[08:59:57][C][template.number:050]:   Device Class: 'temperature'
[08:59:57][C][template.number:051]:   Optimistic: YES
[08:59:57][C][template.number:052]:   Update Interval: 60.0s
[08:59:57][C][template.number:050]: Template Number 'AVfan1 Target Temp 2'
[08:59:57][C][template.number:050]:   Unit of Measurement: '°C'
[08:59:57][C][template.number:050]:   Device Class: 'temperature'
[08:59:57][C][template.number:051]:   Optimistic: YES
[08:59:57][C][template.number:052]:   Update Interval: 60.0s
[08:59:57][C][template.number:050]: Template Number 'AVfan1 Target Temp 3'
[08:59:57][C][template.number:050]:   Unit of Measurement: '°C'
[08:59:57][C][template.number:050]:   Device Class: 'temperature'
[08:59:57][C][template.number:051]:   Optimistic: YES
[08:59:57][C][template.number:052]:   Update Interval: 60.0s
[08:59:57][C][template.number:050]: Template Number 'AVfan1 Target Temp 4'
[08:59:57][C][template.number:050]:   Unit of Measurement: '°C'
[08:59:57][C][template.number:050]:   Device Class: 'temperature'
[... 652 lines cut ...]
[08:42:09][W][dallas.temp.sensor:139]: 'AVfan1 Temperature 2' - Scratch pad checksum invalid!
[08:42:09][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[08:42:09][W][dallas.temp.sensor:139]: 'AVfan1 Temperature 4' - Scratch pad checksum invalid!
[08:42:09][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[08:42:09][D][pulse_counter:174]: 'AVfan1 Mon 4': Retrieved counter: 0.00 pulses/min
[08:42:09][D][sensor:094]: 'AVfan1 Mon 4': Sending state 0.00000 rpm with 2 decimals of accuracy
[08:42:09][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:42:10][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:42:10][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:42:10][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:42:10][W][dallas.temp.sensor:139]: 'AVfan1 Temperature 1' - Scratch pad checksum invalid!
[08:42:10][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[08:42:11][W][dallas.temp.sensor:139]: 'AVfan1 Temperature 3' - Scratch pad checksum invalid!
[08:42:11][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[08:42:11][W][dallas.temp.sensor:139]: 'AVfan1 Temperature 2' - Scratch pad checksum invalid!
[08:42:11][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[08:42:11][W][dallas.temp.sensor:139]: 'AVfan1 Temperature 4' - Scratch pad checksum invalid!
[08:42:11][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[... cut ...]
[08:42:11][D][sensor:094]: 'AVfan1 Loop Time': Sending state 145.00000 ms with 0 decimals of accuracy
[08:42:11][D][sensor:094]: 'AVfan1 Heap Max Block': Sending state 110580.00000 B with 0 decimals of accuracy
[08:42:11][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:42:12][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:42:12][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:42:12][W][component:170]: Component dallas_temp.sensor cleared Warning flag

ssieb, 2024-08-05:
I've double-checked and there were no changes that would directly affect this between those versions.
How are the sensors wired, including lengths?
I've done some tweaking of the timings, so you can try this:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ one_wire, gpio ]
    refresh: 1min
```

ESP32andmore, 2024-08-07:
Samuel,
The hardware and cables were verified and are fine.  I also verified other models of boards and their .yamls and dallsa_temp functioned fine.  I had done a "clean build files" on the device in question named "avfan1" assuming it would pull the everything  anew. It must not have due to reasons I do not understand causing the one_wire issue.
That lead me to do the following: I saved the .yaml under a temp name, DELETED the device, made a new one with the same name and copied in the original code, built, and the dallas sensors worked fine.
Since build directories are hidden it is hard to observe the build directories. Possibly, it  might be caused by  me keeping different versions (for example different sensor configurations) of the .yamls (and therefore different file names) under one device name within the Dashboard.
And likewise the issue I had with SH4x sensors when not connected causing aggravated looptimes, might be the same phenomenon as that "cured itself" which may not have been a lib fix by you all.
Thanks for responding and at a minimum I have a work around if I encounter these strange errors again. Keep up the good work and support despite my prior comment made under frustration.

ESP32andmore, 2024-08-13:
Samuel, Again today I am encountering the same issues with the one-wire/dallas_temp. 2034.6.6 works, current version does not. I'll enter the same issue

sakrlog, 2024-08-15:
Hey folks, I'm facing the exact same issue which I first reported about in this [thread](https://github.com/esphome/issues/issues/6045). This is still an issue for me, I'm thinking of reverting the version until this is solved

swifty99, 2024-09-10:
I have trouble on IDF ESPHome version 2024.8.3 with various ESP32-S2 chips this time. My hardware is unchanged too ;-)

patfelst, 2024-09-18:
I found that disabling the web server component made the Dallas sensors work again.

ESP32andmore, 2024-09-18:
Patrick,
If I get the chance this week I will try  some different configurations to see if it is memory related with the high usage components such as the web server component.  Nonetheless, there appears to be quite a bit of memory available with the server enabled.  Bluetooth also is a hog and I'll add those components to see if memory usage is at issue or something related to the build and libraries being improperly updated.   I did see this when I had multiple devices in the dashboard with the "device name" being the identical.

closed 2024-12-13 by ssieb