esphome/issues #6123: on_time sometimes not triggering

Fusseldieb, 2024-08-06:
### The problem

Seems to be related to #573.

When I use on_time, to turn on a relay at 08:00:00, it happens that it doesn't trigger - sometimes. The funny thing is that the time is synchronized, and it has days in which it does and others which it doesn't - without any reboots in between (meaning that the time is always on sync).

Im starting to think that the code skips from 07:59:59 to 08:00:01 due to blocking action such as reading sensors and whatnot, and misses the on_time condition. 

---

How is the on_time handled? Does it only trigger on exact times, or is there some sort of scheduler that knows that it should still trigger, even if one second has passed? If the former, I think some sort of the latter should be implemented.

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.4

### What platform are you using?

ESP32

### Board

ESP32 dev board

### Component causing the issue

GPIO out

### Example YAML snippet

```yaml
# on_time combined with blocking sensors
```

SeByDocKy, 2024-08-12:
Not sure it's linked but recently I observed that total daily energy sensors are no more resetting à 00h00.....  at least on ESP32 S3 w ESP-IDF recommanded

Ceralor, 2024-10-24:
I'm running into this as well making an alarm clock.