esphome/issues #6124: error: conflicting declaration of 'u32_t sys_now()' with 'C' linkage

SkyeLowry, 2024-08-06:
### The problem

When trying to compile a Generic - RTL8720CF (2M/992k) board with the default yaml from the UI I get the following error when trying to build:
/data/cache/platformio/packages/library-lwip/src/include/lwip/sys.h:447:7: error: conflicting declaration of 'u32_t sys_now()' with 'C' linkage
  447 | u32_t sys_now(void);
      |       ^~~~~~~
In file included from /data/cache/platformio/packages/library-lwip/src/include/lwip/opt.h:51,
                 from /data/cache/platformio/packages/library-lwip/src/include/lwip/ip_addr.h:40,
                 from .piolibdeps/test-config/AsyncTCP-esphome/src/AsyncTCP.h:28,
                 from .piolibdeps/test-config/AsyncTCP-esphome/src/AsyncTCP.cpp:24:
/data/cache/platformio/platforms/libretiny/cores/realtek-ambz2/base/config/lwipopts.h:10:22: note: previous declaration with 'C++' linkage
   10 | extern unsigned long sys_now(void);
      |                      ^~~~~~~
*** [.pioenvs/test-config/lib45a/AsyncTCP-esphome/AsyncTCP.cpp.o] Error 1
========================= [FAILED] Took 13.79 seconds =========================

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.4

### What platform are you using?

RTL87XX

### Board

Generic - RTL8720CF (2M/992k)

### Example YAML snippet

```yaml
esphome:
  name: test-config
  friendly_name: Test Config

rtl87xx:
  board: generic-rtl8720cf-2mb-992k

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "+9wtvaMfbewBuhrgaIPCb6OE7p4WEKfQ+KWmuahbpRc="

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-Config Fallback Hotspot"
    password: "xxx"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/test-config.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test-config (board: generic-rtl8720cf-2mb-992k; framework: arduino; platform: libretiny @ 1.5.1)
--------------------------------------------------------------------------------
Warning! Non-Git installations are NOT SUPPORTED.
HARDWARE: RTL8720CF 100MHz, 256KB RAM, 992KB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-realtek-ambz2 @ 0.0.0+v2022.11.17.sha.5f7b937 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 10.0.1+sha.752c89a 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.1.3-ambz2+sha.929d2dd 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.5.1
 - ltchiptool @ 4.11.1
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2024.7.3
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/test-config/lib45a/AsyncTCP-esphome/AsyncTCP.cpp.o
Compiling .pioenvs/test-config/libc9f/ESPAsyncWebServer-esphome/WebRequest.cpp.o
Compiling .pioenvs/test-config/libc9f/ESPAsyncWebServer-esphome/WebResponses.cpp.o
Compiling .pioenvs/test-config/libc9f/ESPAsyncWebServer-esphome/WebServer.cpp.o
Compiling .pioenvs/test-config/libb92/DNSServer/DNSServer.cpp.o
Compiling .pioenvs/test-config/lib11c/libsodium/crypto_aead/chacha20poly1305/sodium/aead_chacha20poly1305.c.o
Compiling .pioenvs/test-config/lib11c/libsodium/crypto_core/ed25519/core_ed25519.c.o
In file included from /data/cache/platformio/packages/library-lwip/src/include/lwip/timeouts.h:44,
                 from /data/cache/platformio/packages/library-lwip/src/include/lwip/tcpip.h:45,
                 from /data/cache/platformio/packages/library-lwip/src/include/lwip/priv/tcpip_priv.h:44,
                 from .piolibdeps/test-config/AsyncTCP-esphome/src/AsyncTCP.cpp:369:
/data/cache/platformio/packages/library-lwip/src/include/lwip/sys.h:447:7: error: conflicting declaration of 'u32_t sys_now()' with 'C' linkage
  447 | u32_t sys_now(void);
      |       ^~~~~~~
In file included from /data/cache/platformio/packages/library-lwip/src/include/lwip/opt.h:51,
                 from /data/cache/platformio/packages/library-lwip/src/include/lwip/ip_addr.h:40,
                 from .piolibdeps/test-config/AsyncTCP-esphome/src/AsyncTCP.h:28,
                 from .piolibdeps/test-config/AsyncTCP-esphome/src/AsyncTCP.cpp:24:
/data/cache/platformio/platforms/libretiny/cores/realtek-ambz2/base/config/lwipopts.h:10:22: note: previous declaration with 'C++' linkage
   10 | extern unsigned long sys_now(void);
      |                      ^~~~~~~
Compiling .pioenvs/test-config/lib11c/libsodium/crypto_core/ed25519/core_ristretto255.c.o
*** [.pioenvs/test-config/lib45a/AsyncTCP-esphome/AsyncTCP.cpp.o] Error 1
========================== [FAILED] Took 3.65 seconds ==========================
```

jwreford, 2024-08-29:
I ran into the same issue today (2024.8.0).
It compiles correctly for me if I remove the captive_portal component.

Not ideal for my case as the board I'm working on needs to be reassembled deep into the device it came from, so if something happens it will be harder to troubleshoot. But at least it builds that way.

k-w-1, 2024-11-28:
Another datapoint: I was not able to compile with `captive_portal` removed, unless I also removed `web_server`. However, once flashed, it still failed to boot. Although ironically it turns out that I actually personally have an RTL87**1**0CF, I did discover that it actually boots with code targetting the RTL8720CF. Furthermore, I was able to resolve the boot error by using the `realtek-update` branch of LibreTiny, which has the added value of fixing the `captive_portal` and `web_server` issues.

YMMV, but the code I used was:
```yaml
rtl87xx:
  board: wbr3
  family: rtl8720c
  framework:
    version: 0.0.0
    source: https://github.com/libretiny-eu/libretiny#feature/realtek-update
```
You might however want to try simply adding the framework override to LibreTiny's realtek-update branch (e.g. the wbr3 board & family parts might not be required).
All of that said, it's obviously under heavy development (particularly, OTA isn't yet supported); you may want to watch libretiny-eu/libretiny#44.

fcol95, 2025-03-21:
Had the same error recently, seems to be fixed with the latest release (v1.8.0) of libretiny!
However my esphome builder UI still pulls v1.7.0 by default, so had to override it:
```
rtl87xx:
  board: generic-rtl8720cf-2mb-992k
  framework:
    version: 1.8.0
    source: https://github.com/libretiny-eu/libretiny

```