esphome/issues #6128: GROW R503 fingerprint reader can't recognize fingeprints after waking up with error Timed out waiting for sensor wake-up

Paclanc, 2024-08-07:
### The problem

I've started using the R503 grow fingerprint sensor on a ESP32-ETH01 board, alongside some binary sensor for the door, the doorbell and the door lock. 

I starded noticing that sometimes, when I touch the sensor, it doesn't recognize my fingerprint, not until it goes to sleep and then wakes up again. 

```
[16:05:44][E][fingerprint_grow:509]: Timed out waiting for sensor wake-up
[16:05:45][E][fingerprint_grow:467]: No response received from reader
[16:05:45][E][fingerprint_grow:522]: Wrong password
[16:05:45][E][fingerprint_grow:457]: Unknown response received from reader: 0x21
[16:05:45][D][fingerprint_grow:179]: Unknown Scan Error: 33
[16:05:45][W][component:237]: Component fingerprint_grow took a long time for an operation (1032 ms).
[16:05:45][W][component:238]: Components should block for at most 30 ms.
```

I have configured the sensor as the doc says in order to use sleep mode. (However, I couldn't find a BC327 transistor, so I used a BC558)

These are the full logs: On these, the fingerprints recognizes my finger. Then, after going to sleep it does not. Afther another wake up, it works again.

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/esp32-casa-puerta-eth.yaml...
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.31.185 using esphome API
INFO Successfully connected to esp32-casa-puerta-eth @ 192.168.31.185 in 0.124s
INFO Successful handshake with esp32-casa-puerta-eth @ 192.168.31.185 in 0.070s
[16:05:36][I][app:100]: ESPHome version 2024.7.3 compiled on Aug  1 2024, 09:11:43
[16:05:36][C][wifi:599]: WiFi:
[16:05:36][C][wifi:427]:   Local MAC: BC:DD:C2:CC:6B:E0
[16:05:36][C][wifi:432]:   SSID: [redacted]
[16:05:36][C][wifi:435]:   IP Address: 192.168.31.185
[16:05:36][C][wifi:439]:   BSSID: [redacted]
[16:05:36][C][wifi:440]:   Hostname: 'esp32-casa-puerta-eth'
[16:05:36][C][wifi:442]:   Signal strength: -75 dB ▂▄▆█
[16:05:36][C][wifi:446]:   Channel: 1
[16:05:36][C][wifi:447]:   Subnet: 255.255.255.0
[16:05:36][C][wifi:448]:   Gateway: 192.168.31.1
[16:05:36][C][wifi:449]:   DNS1: 192.168.31.1
[16:05:36][C][wifi:450]:   DNS2: 0.0.0.0
[16:05:36][C][logger:185]: Logger:
[16:05:36][C][logger:186]:   Level: DEBUG
[16:05:36][C][logger:188]:   Log Baud Rate: 115200
[16:05:36][C][logger:189]:   Hardware UART: UART0
[16:05:36][C][uart.arduino_esp32:151]: UART Bus 1:
[16:05:36][C][uart.arduino_esp32:152]:   TX Pin: GPIO17
[16:05:36][C][uart.arduino_esp32:153]:   RX Pin: GPIO5
[16:05:36][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[16:05:36][C][uart.arduino_esp32:157]:   Baud Rate: 57600 baud
[16:05:36][C][uart.arduino_esp32:158]:   Data Bits: 8
[16:05:36][C][uart.arduino_esp32:159]:   Parity: NONE
[16:05:36][C][uart.arduino_esp32:160]:   Stop bits: 1
[16:05:36][C][gpio.output:010]: GPIO Binary Output:
[16:05:36][C][gpio.output:011]:   Pin: GPIO2
[16:05:36][C][gpio.output:012]:   Inverted: YES
[16:05:36][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Puerta de casa'
[16:05:36][C][gpio.binary_sensor:015]:   Device Class: 'door'
[16:05:36][C][gpio.binary_sensor:016]:   Pin: GPIO33
[16:05:36][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Timbre de casa'
[16:05:36][C][gpio.binary_sensor:016]:   Pin: GPIO32
[16:05:36][C][template.text_sensor:020]: Template Sensor 'Estado del lector de huellas'
[16:05:36][C][template.text_sensor:020]: Template Sensor 'Modo del lector de huellas'
[16:05:36][C][template.number:050]: Template Number 'ID seleccionado'
[16:05:36][C][template.number:051]:   Optimistic: YES
[16:05:36][C][template.number:052]:   Update Interval: 60.0s
[... 478 lines cut ...]
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
  - service: enroll_next # Idea by Ralf Klüber (thanks!)
    variables:
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return id(fingerprint_count).state;'
          num_scans: !lambda 'return num_scans;'
  - service: cancel_enroll
    then:
      - fingerprint_grow.cancel_enroll:
  - service: delete
    variables:
      finger_id: int
    then:
      - fingerprint_grow.delete:
          finger_id: !lambda 'return finger_id;'
  - service: delete_all
    then:
      - fingerprint_grow.delete_all:
```

Paclanc, 2024-08-14:
Any comments?

nguyenquangduc, 2024-08-23:
Have you solved this problem? How to solve it? I have the same problem

Paclanc, 2024-08-23:
No, I haven't solved it yet. I don't know if it is software related or hardware related.

Are you using the BC327 transistor? Or extending your reader with Ethernet cable?