esphome/issues #6130: Deprecated flag "color_mode"

robercy, 2024-08-08:
### The problem

Warning messages in the log file. 

### Which version of ESPHome has the issue?

all

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

all

### What platform are you using?

ESP8266

### Board

sonoff

### Anything in the logs that might be useful for us?

```txt
Logger: homeassistant.components.mqtt.light.schema_json
Source: components/mqtt/light/schema_json.py:156
integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 09:12:29 (6 occurrences)
Last logged: 09:12:29

Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'schema': 'json', 'supported_color_modes': ['color_temp'], 'brightness': True, 'effect': True, 'icon': 'mdi:led-strip', 'name': 'Output', 'effect_list': ['Random', 'My Slow Random Effect', 'My Fast Random Effect', 'None'], 'color_mode': True, 'command_topic': 'b612/dining-room-stripled-inner/cmd/out', 'availability_topic': 'b612/dining-room-stripled-inner/status', 'state_topic': 'b612/dining-room-stripled-inner/out', 'unique_id': '4c11ae0e3acc-light-ef48dc26', 'device': {'name': 'Dining room Stripled inner', 'sw_version': 'esphome v2024.5.5 Jun 16 2024, 21:08:11', 'identifiers': '4c11ae0e3acc', 'suggested_area': 'Dining room', 'model': 'esp01_1m', 'manufacturer': 'espressif'}, 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3

Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'schema': 'json', 'supported_color_modes': ['color_temp'], 'brightness': True, 'effect': True, 'icon': 'mdi:led-strip', 'name': 'Output', 'effect_list': ['Random', 'My Slow Random Effect', 'My Fast Random Effect', 'None'], 'color_mode': True, 'command_topic': 'b612/dining-room-stripled-outer/cmd/out', 'availability_topic': 'b612/dining-room-stripled-outer/status', 'state_topic': 'b612/dining-room-stripled-outer/out', 'unique_id': '3c71bf2ab1c8-light-ef48dc26', 'device': {'name': 'Dining room Stripled outer', 'sw_version': 'esphome v2024.5.5 Jun 16 2024, 21:14:43', 'identifiers': '3c71bf2ab1c8', 'suggested_area': 'Dining room', 'model': 'esp01_1m', 'manufacturer': 'espressif'}, 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3

Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'schema': 'json', 'supported_color_modes': ['color_temp'], 'brightness': True, 'effect': True, 'icon': 'mdi:led-strip', 'name': 'Output CWWW', 'effect_list': ['Strobe', 'Pulse', 'Flicker', 'None'], 'color_mode': True, 'command_topic': 'b612/home-theater-stripled/cmd/out_cwww', 'availability_topic': 'b612/home-theater-stripled/status', 'state_topic': 'b612/home-theater-stripled/out_cwww', 'unique_id': 'cc7b5c361590-light-b3c4d33a', 'device': {'name': 'Home theater stripLED', 'sw_version': 'esphome v2024.7.2 Jul 30 2024, 16:49:49', 'identifiers': 'cc7b5c361590', 'suggested_area': 'Home theater', 'model': 'nodemcu-32s', 'manufacturer': 'espressif'}, 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3

Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'schema': 'json', 'supported_color_modes': ['rgb'], 'brightness': True, 'effect': True, 'icon': 'mdi:led-strip', 'name': 'Output RGB', 'effect_list': ['Strobe', 'Pulse', 'Flicker', 'Rainbow', 'Color Wipe', 'Scan', 'Twinkle', 'Random Twinkle', 'Fireworks', 'Addressable Flicker', 'WLED', 'DDP', 'None'], 'color_mode': True, 'command_topic': 'b612/home-theater-stripled/cmd/out_rgb', 'availability_topic': 'b612/home-theater-stripled/status', 'state_topic': 'b612/home-theater-stripled/out_rgb', 'unique_id': 'cc7b5c361590-light-77978f33', 'device': {'name': 'Home theater stripLED', 'sw_version': 'esphome v2024.7.2 Jul 30 2024, 16:49:49', 'identifiers': 'cc7b5c361590', 'suggested_area': 'Home theater', 'model': 'nodemcu-32s', 'manufacturer': 'espressif'}, 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3

Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'schema': 'json', 'supported_color_modes': ['color_temp'], 'brightness': True, 'effect': True, 'icon': 'mdi:led-strip', 'name': 'Output', 'effect_list': ['Random', 'My Slow Random Effect', 'My Fast Random Effect', 'None'], 'color_mode': True, 'command_topic': 'b612/living-room-stripled/cmd/out', 'availability_topic': 'b612/living-room-stripled/status', 'state_topic': 'b612/living-room-stripled/out', 'unique_id': 'cc50e35d6cd2-light-ef48dc26', 'device': {'name': 'Living room Stripled', 'sw_version': 'esphome v2024.5.5 Jun 17 2024, 15:15:10', 'identifiers': 'cc50e35d6cd2', 'suggested_area': 'Living room', 'model': 'esp01_1m', 'manufacturer': 'espressif'}, 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3`
```

jensweimann, 2024-10-21:
Still occurring as of Esphome version 20.10.0. I have this config:

`light:
  - platform: neopixelbus
    variant: WS2812
    name: light
    id: my_light
    default_transition_length: 0s
    gamma_correct: 1.0
    pin: GPIO3
    num_leds: 7
    icon: mdi:moon-full
    restore_mode: RESTORE_DEFAULT_OFF`

sbrk-r, 2024-11-14:
I also get these warnings in HA the last few months, I don't even have lights per se - just the status_led component triggers it.

I guess the mqtt component needs changing to fit new home assistant.

nima-1102, 2025-01-08:
I still get these message:
```
Logger: homeassistant.components.mqtt.light.schema_json
Source: components/mqtt/light/schema_json.py:158
integration: MQTT (documentation, issues)
First occurred: 21:53:03 (1 occurrences)
Last logged: 21:53:03

Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'schema': 'json', 'supported_color_modes': ['rgb'], 'brightness': True, 'effect': True, 'unique_id': '4417930daac4-light-d5c7b082', 'availability_topic': 'klo/status', 'name': 'LED Strip', 'effect_list': ['Rainbow', 'Scan', 'Twinkle', 'Random Twinkle', 'Fireworks', 'Addressable Flicker', 'None'], 'device': {'identifiers': '4417930daac4', 'name': 'klo', 'sw_version': '2024.12.2 (Jan 8 2025, 21:51:54)', 'model': 'd1_mini', 'connections': [['mac', '4417930daac4']], 'manufacturer': 'Espressif'}, 'state_topic': 'klo/light/led_strip/state', 'color_mode': True, 'command_topic': 'klo/light/led_strip/command', 'icon': 'mdi:led-strip-variant'}. This will stop working in Home Assistant Core 2025.3
```

These are my config:
```
light:
  - platform: neopixelbus
    name: "LED Strip"
    id: led_strip
    icon: mdi:led-strip-variant
    pin: D3
    restore_mode: ALWAYS_OFF
    type:  GRB
    variant: WS2812
    num_leds: 150
    effects:
      - addressable_rainbow:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
```

HA Core: 2025.1.1
ESPHome Firmware: 2024.12.2

nima-1102, 2025-03-07:
Since version Home Assistant Core 2025.3 is "color_mode" now an error and no longer a warning and should be removed from ESPHome.

```
Log details (ERROR)
Logger: homeassistant.components.mqtt
Source: helpers/config_validation.py:990
integration: MQTT (documentation, issues)
First occurred: March 7, 2025 at 22:54:07 (15 occurrences)
Last logged: March 7, 2025 at 22:54:07

The 'color_mode' option has been removed, please remove it from your configuration
```

DirkBaumeister, 2025-04-23:
So, any progress on this? 🙂