esphome/issues #6132: Missing library ld2450 since update ESPhome, cannot compile previously working yaml

m90att, 2024-08-08:
### The problem

Since upgrading Home Assistant to 2024.8.0, I am seeing an error when installing yaml that works in previous versions.

YAML is provided below.
ld2450 is underlined in red in Home Assistant ESPHome addon 2024.7.3

[image]

When installing, the following error is seen.

`INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/presence3.yaml...
Failed config

external_components: [source /config/esphome/presence3.yaml:12]
  - source: 
      type: git
      url: https://github.com/hareeshmu/esphome
      ref: ld2450
    components: 
      
      Could not find __init__.py file for component ld2450. Please check the component is defined by this source (search path: /data/external_components/46c9292e/esphome/components/ld2450/__init__.py).
      - ld2450`

I am not certain in which version this started but I rolled back ESPHome to 2024.6.6 and none of the issues above are seen and the installation succeeds.

### Which version of ESPHome has the issue?

2024.7.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.0

### What platform are you using?

ESP32

### Component causing the issue

HLK-LD2450

### Example YAML snippet

```yaml
esphome:
  name: mmwave-sensor
  friendly_name: mmWave Sensor
  name_add_mac_suffix: True

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source:
      type: git
      url: https://github.com/hareeshmu/esphome
      ref: ld2450
    components: [ ld2450 ]

uart:
  id: uart_bus
  rx_pin: 
[... 331 lines cut ...]
        name: Zone-3 Moving Target Count

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Presence3 Fallback Hotspot"
    password: "9TOtbZGaUgwR"

captive_portal:
```

randybb, 2024-08-08:
try to Clear build files

m90att, 2024-08-08:
Hi @randybb, thanks for the reply.  I've tried 'Clean Build Files' but get the same error...
[image]

ssieb, 2024-08-09:
I just tried your config and it compiled with no problems.
Try adding a `refresh: 1min` option to the external component config.  Your cached copy seems to be messed up somehow.

randybb, 2024-08-09:
This is why I proposed the clearing of build files as I added it to my config and I got the same validation error, but once I cleared build files it compiled without any issues.

m90att, 2024-08-09:
Thank you, I tried it again this morning and it's working.  I can't begin to imagine what the cause was but I accept that it is likely to be local to my setup.  Thank you for your support.  Closing.

closed 2024-08-09 by m90att