esphome/issues #6133: Assert failed when using type: BINARY for online_image

jbcurran, 2024-08-09:
### The problem

I am attempting to use online_image with my [Adafruit Magtag](https://www.adafruit.com/magtag).
I have successfully displayed the image once but all attempts now results in an error. 

If I use any other type (i.e. GRAYSCALE), I do not get the errors, but this image is not displayed properly.

I included a image component to show that the Magtag is capable of displaying the correct image.
I've tried a number of image hosting locations, to check if that's the problem.

### Which version of ESPHome has the issue?

2024.8.0-dev20240808

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.4

### What platform are you using?

ESP32-IDF

### Board

adafruit_magtag29_esp32s2

### Component causing the issue

online_image

### Example YAML snippet

```yaml
esphome:
  name: kitchendisplay
  friendly_name: kitchendisplay
  # on_boot:
  #   then:
  #     - wait_until: api.connected
  #     - component.update: example_image

esp32:
  board: adafruit_magtag29_esp32s2
  framework:
    type: esp-idf

logger:

api:
  encryption:
    key: !secret kd_key

ota:
  - platform: esphome
    password: !secret kd_pass

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

substitutions:
  # image_url: "https://dummyimage.com/100x100/000/fff.png"
  # image_url: "https://placehold.co/100x100/000000/FFFFFF.png"
  image_url: "https://i.imghippo.com/files/edXQc1723215915.png"

http_request:

online_image:
  - url: ${image_url}
    format: png
    id: example_image
    on_download_finished:
      - logger.log: "Download Finished"
      - component.update: kit_disp
    on_error:
      - logger.log: "Could not download the image"

image:
  - file: ${image_url}
    id: example_image2

spi:
  clk_pin: GPIO36
  mosi_pin: GPIO35
  miso_pin: GPIO37

display:
  - platform: waveshare_epaper
    id: kit_disp
    cs_pin: GPIO8
    dc_pin: GPIO7
    busy_pin: GPIO5
    reset_pin: GPIO6
    model: gdew029t5
    reset_duration: 200ms
    update_interval: never
    rotation: 270
    lambda: |-
      ESP_LOGI("display", "Updating display");
      it.image( 20, 10, id(example_image ));
      it.image(140, 10, id(example_image2));

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO15
      inverted: true
      mode:
        input: true
        pullup: true
    name: Button
    on_press:
      then:
        - logger.log: "button press"
        - component.update: kit_disp
  - platform: gpio
    pin:
      number: GPIO14
      inverted: true
      mode:
        input: true
        pullup: true
    name: Button
    on_press:
      then:
        - logger.log: "button press"
        - component.update: example_image
  - platform: gpio
    pin:
      number: GPIO12
      inverted: true
      mode:
        input: true
        pullup: true
    name: Button
    on_press:
      then:
        - logger.log: "button press"
        - online_image.release: example_image
```

### Anything in the logs that might be useful for us?

```txt
[11:07:53][D][binary_sensor:036]: 'Button': Sending state ON
[11:07:53][D][main:174]: button press
[11:07:53][I][online_image:095]: Updating image
[11:07:55][D][esp-idf:000]: I (58058) esp-x509-crt-bundle: Certificate validated
[11:07:55]
[11:07:56][D][esp-idf:000]: I (59238) HTTP_CLIENT: Body received in fetch header state, 0x3ffe528e, 176
[11:07:56]
[11:07:56][D][online_image:120]: Starting download
[11:07:56][I][online_image:136]: Downloading image
[11:07:56][W][component:237]: Component gpio.binary_sensor took a long time for an operation (2734 ms).
[11:07:56][W][component:238]: Components should block for at most 30 ms.
[11:07:56][D][online_image:066]: Allocating new buffer of 1250 Bytes...
[11:07:56][D][online_image:073]: New size: (100, 100)
[11:07:56][W][component:237]: Component online_image took a long time for an operation (95 ms).
[11:07:56][W][component:238]: Components should block for at most 30 ms.
[11:07:56][D][binary_sensor:036]: 'Button': Sending state OFF
[11:07:56][D][online_image:145]: Image fully downloaded
[11:07:56][D][main:312]: Download Finished
[11:07:56][I][display:067]: Updating display
[11:07:56][W][component:237]: Component online_image took a long time for an operation (179 ms).
[11:07:56][W][component:238]: Components should block for at most 30 ms.
[11:07:56]
[11:07:56]assert failed: block_locate_free heap_tlsf.c:441 (block_size(block) >= size)
[11:07:56]
[11:07:56]
[11:07:56]Backtrace: 0x40024a4a:0x3ffd6690 0x4002af5d:0x3ffd66b0 0x4003174d:0x3ffd66d0 0x4002face:0x3ffd67f0 0x40030171:0x3ffd6810 0x400253bf:0x3ffd6830 0x400253d1:0x3ffd6860 0x400253fa:0x3ffd6880 0x4003175d:0x3ffd68a0 0x400af200:0x3ffd68c0 0x400afa23:0x3ffd68e0 0x400b327d:0x3ffd6900 0x400b8a46:0x3ffd6950 0x400b92eb:0x3ffd6970 0x400ae50b:0x3ffd6990
[11:07:56]
[11:07:56]
[11:07:56]
[11:07:56]
[11:07:56]ELF file SHA256: 78996621e1184a9a
[11:07:56]
Terminal disconnected: NetworkError: The device has been lost.
```

### Additional information

type: BINARY (Before errors)

[image]

type: GRAYSCALE

[image]

guillempages, 2024-08-19:
I've taken a brief look at that, and I cannot reproduce it :-(

At a first glance, it looks like it has a problem allocating memory (it cannot find a block, that is big enough).
Does your device have PSRAM? Is it enabled?

jbcurran, 2024-08-20:
Thanks for taking a look, and thank you for all the effort towards this component! It's really awesome.

Yes, it has a ESP32-S2-WROVER (4 MByte of Flash and 2 MByte of PSRAM). I don't actually know how to confirm the psram is enabled other than what is displayed during installation:
[image]

I took a dive into the [sdkconfig_options](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#compiler-options), but wasn't able to fix the issue.

Do you know if your component needs one of these options?
[image]

jbcurran, 2024-08-20:
I switched back to framework: type: arduino and it's working!
So, not sure where there error lies in esp-idf, but I'll keep digging.