esphome/issues #6136: JSN-SR04T component spamming the sensor unreasonably

Pzdrs, 2024-08-10:
### The problem

I want to use the JSN-SR04T water level sensor, but when I connect it to the microcontroller, it seems like the sensor is being queried (sending 0x55) for a reading many times a second. I expected that the sensor would read every so often (the defined *update_interval*) and make the almost inaudible click sound every time. Instead its pretty much a constant beep/whine because its clicking so fast.

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

jsn_sr04t

### Example YAML snippet

```yaml
uart:
  - baud_rate: 9600
    tx_pin: 1
    rx_pin: 3

sensor:
  - platform: "jsn_sr04t"
    name: "Distance"
    update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
With the serial lines disconnected
---
[18:37:59][W][component:237]: Component jsn_sr04t.sensor took a long time for an operation (127 ms).
[18:37:59][W][component:238]: Components should block for at most 30 ms.
[18:38:04][VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=98632 (now=103632)
[18:38:04][V][jsn_sr04t.sensor:014]: Request read out from sensor

With them connected
---
[18:38:14][W][component:237]: Component jsn_sr04t.sensor took a long time for an operation (127 ms).
[18:38:14][W][component:238]: Components should block for at most 30 ms.
[18:38:14][V][jsn_sr04t.sensor:022]: Read byte from sensor: ff
[18:38:14][V][jsn_sr04t.sensor:022]: Read byte from sensor: 1
[18:38:14][V][jsn_sr04t.sensor:022]: Read byte from sensor: 6
[18:38:14][V][jsn_sr04t.sensor:022]: Read byte from sensor: 6
[18:38:14][V][jsn_sr04t.sensor:039]: Distance from sensor: 262mm, 0.262m
[18:38:14][V][sensor:043]: 'Distance': Received new state 0.262000
[18:38:14][D][sensor:094]: 'Distance': Sending state 0.26200 m with 3 decimals of accuracy
[18:38:14][W][component:237]: Component jsn_sr04t.sensor took a long time for an operation (398 ms).
[18:38:15][W][component:238]: Components should block for at most 30 ms.
[18:38:15][V][jsn_sr04t.sensor:022]: Read byte from sensor: ff
[18:38:15][V][jsn_sr04t.sensor:022]: Read byte from sensor: 1
[18:38:15][V][jsn_sr04t.sensor:022]: Read byte from sensor: 6
[18:38:15][V][jsn_sr04t.sensor:022]: Read byte from sensor: 6
[18:38:15][V][jsn_sr04t.sensor:039]: Distance from sensor: 262mm, 0.262m
[18:38:15][V][sensor:043]: 'Distance': Received new state 0.262000
[18:38:15][D][sensor:094]: 'Distance': Sending state 0.26200 m with 3 decimals of accuracy
```

### Additional information

When the serial lines are not connected, according to the logs, reading happens every 5s, as defined in the YAML. But when I connect them, both the logs and the sensor itself go crazy. Important to note that the readings seem to be correct so that is not the issue at hand.

ssieb, 2024-08-11:
Did you disable serial logging?

Pzdrs, 2024-08-11:
No, quite the opposite actually. I upped the level to see the component logging every time it pings the sensor.

ssieb, 2024-08-11:
The **serial** logging.  If you haven't disabled that, then you're spamming the sensor with data and probably triggering it endlessly.
Since this is an ESP32, you should be using other pins for the sensor uart.

Pzdrs, 2024-08-11:
Disabling logging over serial by setting the _baud_rate_ on the logger component did the trick. Thank you so much. I don't think I would ever figure this out myself.

closed 2024-08-11 by Pzdrs

ssieb, 2024-08-11:
But you shouldn't be using those pins anyway on an ESP32.

Pzdrs, 2024-08-11:
Yeah I reread your comment a few minutes after the initial fix and changed the pin pair. Thank you