esphome/issues #6137: Bluetooth Proxys unavailable since 2024.8

Lux315, 2024-08-10:
### The problem

All bluetooth proxys and their devices are offline and unavailable since 2024.8
[image]

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8

### What platform are you using?

ESP32

ssieb, 2024-08-11:
How did you install it?
The message tells you what the issue is.

fjavimer, 2024-08-16:
I have exactly the same problem. Has anyone been able to solve it?

Lux315, 2024-08-16:
I reinstalled the proxy and now it works fine for me

closed 2024-08-16 by Lux315

fjavimer, 2024-08-17:
working, thank you very much