esphome/issues #6140: Voice Assistant works once

Coldness00, 2024-08-11:
### The problem

Hello,

I'm setting up an ESP Voice Assistant and it works only once.
The only solution I found is to either restart the ESP or to reconfigure the device from the integration panel (API reconnection).

It s configured to work on push to talk, when I push the first time, I have normal log output however, on the second press, nothing except the push button being pressed.

Issue is same while using Wake Word -> Works only once then stop working until reboot

I use adf_pipeline, but with regular confif I have same issue
I have no issues on other devices

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.08

### What platform are you using?

ESP32-IDF

### Board

ESP-WROOM-32

### Component causing the issue

voice assistant

### Example YAML snippet

```yaml
esphome:
  name: esp32-voiceass1
  friendly_name: ESP32-VoiceAss1
  on_boot:
    - priority: -100
      then:
        - wait_until: api.connected
        - delay: 1s
        - if:
            condition:
              switch.is_on: use_wake_word
            then:
              - voice_assistant.start_continuous:

esp32:
  board: denky32
  framework:
    type: esp-idf
    version: recommended
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  port: 6053
  encryption:
    key: plopploploploplo

ota:
[... 302 lines cut ...]
[15:46:46][D][esp-idf:000][http]: W (1987426) HTTP_STREAM: No more data,errno:0, total_bytes:52060, rlen = 0

[15:46:46][D][esp-idf:000][http]: I (1987429) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0

[15:46:46][I][esp_adf_pipeline:214]: [ http ] status: 15
[15:46:46][D][esp_adf_pipeline:302]: State changed from RUNNING to STOPPING
[15:46:46][I][adf_media_player:135]: got new pipeline state: 4
[15:46:46][D][esp-idf:000][decoder]: I (1987618) AUDIO_ELEMENT: IN-[decoder] AEL_IO_DONE,-2

[15:46:46][D][esp-idf:000][decoder]: I (1987744) MP3_DECODER: Closed

[15:46:46][D][esp-idf:000][i2s_out]: I (1987839) AUDIO_ELEMENT: IN-[i2s_out] AEL_IO_DONE,-2

[15:46:46][D][esp_adf_pipeline:302]: State changed from STOPPING to STOPPED
[15:46:46][I][adf_media_player:135]: got new pipeline state: 5

-------
Then it s not working as can see.
-------

[15:46:49][D][binary_sensor:036]: 'VoicePushToTalk': Sending state ON
[15:46:50][D][binary_sensor:036]: 'VoicePushToTalk': Sending state OFF
[15:46:51][D][binary_sensor:036]: 'VoicePushToTalk': Sending state ON
[15:46:52][D][binary_sensor:036]: 'VoicePushToTalk': Sending state OFF
```

rgnyldz, 2024-09-18:
Have the exact setup and exact problem. Because I too want to use media player I use adf. Below are my a bit cleaned logs and there I have two requests. First one is successful, but the second one is halting on "State changed from STOPPING to STOPPED" 

So I believe it is a problem related to using ADF. and the state of it not changing. 

```
[23:41:21][D][micro_wake_word:399]: Resetting buffers and probabilities
[23:41:21][D][micro_wake_word:195]: State changed from IDLE to START_MICROPHONE
[23:41:21][D][micro_wake_word:107]: Starting Microphone
[23:41:21][D][esp_adf_pipeline:050]: Starting request, current state UNINITIALIZED
[23:41:21][D][esp-idf:000]: I (5483) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[23:41:21][D][i2s_audio:072]: Installing driver : yes
[23:41:21][D][esp_adf_pipeline:358]: pipeline tag 0, i2s_in
[23:41:21][D][esp_adf_pipeline:358]: pipeline tag 1, pcm_reader
[23:41:21][D][esp-idf:000]: I (5493) AUDIO_PIPELINE: link el->rb, el:0x3d80c358, tag:i2s_in, rb:0x3d80c634

[23:41:21][D][esp_adf_pipeline:370]: Setting up event listener.
[23:41:21][D][esp_adf_pipeline:302]: State changed from UNINITIALIZED to PREPARING
[23:41:21][D][micro_wake_word:195]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[23:41:21][D][esp_audio_sinks:053]: Set bitdepth to 32
[23:41:21][D][esp_adf_pipeline:302]: State changed from PREPARING to STARTING
[23:41:21][D][esp-idf:000]: I (5510) AUDIO_ELEMENT: [i2s_in-0x3d80c358] Element task created

[23:41:21][D][esp-idf:000]: I (5512) AUDIO_ELEMENT: [pcm_reader-0x3d80c4ec] Element task created

[23:41:21][D][esp-idf:000]: I (5514) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:8479499 Bytes, Inter:144328 Bytes, Dram:144328 Bytes

[23:41:21][D][esp-idf:000][i2s_in]: I (5518) AUDIO_ELEMENT: [i2s_in] AEL_MSG_CMD_RESUME,state:1

[23:41:21][D][esp-idf:000]: I (5520) AUDIO_PIPELINE: Pipeline started

[23:41:21][I][esp_adf_pipeline:214]: [ pcm_reader ] status: 12
[23:41:21][D][esp_adf_pipeline:131]: Check element [i2s_in] status, 3
[23:41:21][D][esp_adf_pipeline:131]: Check element [pcm_reader] status, 3
[23:41:21][D][esp_adf_pipeline:302]: State changed from STARTING to RUNNING

[23:41:21][D][micro_wake_word:195]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
[23:41:21][I][esp_adf_pipeline:214]: [ pcm_reader ] status: 12
[23:41:21][I][esp_adf_pipeline:214]: [ i2s_in ] status: 12

[23:41:21][C][template.switch:068]: Template Switch 'SL - Mute assist'
[23:41:21][C][template.switch:091]:   Restore Mode: always OFF
[23:41:21][C][template.switch:057]:   Optimistic: YES

[23:41:21][C][i2s_audio:028]: I2SController:
[23:41:21][C][i2s_audio:029]:   AccessMode: exclusive
[23:41:21][C][i2s_audio:030]:   Port: 0
[23:41:21][C][i2s_audio:032]:   Reader registered.
[23:41:21][C][i2s_audio:028]: I2SController:
[23:41:21][C][i2s_audio:029]:   AccessMode: exclusive
[23:41:21][C][i2s_audio:030]:   Port: 1
[23:41:21][C][i2s_audio:035]:   Writer registered.
[23:41:21][C][i2s_audio:138]: I2S-Writer (Initial-CFG):
[23:41:21][C][i2s_audio:140]:   sample-rate: 16000 bits_per_sample: 32
[23:41:21][C][i2s_audio:141]:   channel_fmt: 0 channels: 2
[23:41:21][C][i2s_audio:142]:   use_apll: no, use_pdm: no
[23:41:21][C][i2s_audio:135]: I2S-Reader (Initial-CFG):
[23:41:21][C][i2s_audio:140]:   sample-rate: 16000 bits_per_sample: 32
[23:41:21][C][i2s_audio:141]:   channel_fmt: 4 channels: 1
[23:41:21][C][i2s_audio:142]:   use_apll: no, use_pdm: no

[23:41:21][C][micro_wake_word:051]: microWakeWord:
[23:41:21][C][micro_wake_word:052]:   models:
[23:41:21][C][micro_wake_word:015]:     - Wake Word: okay nabu
[23:41:21][C][micro_wake_word:016]:       Probability cutoff: 0.970
[23:41:21][C][micro_wake_word:017]:       Sliding window size: 5
[23:41:21][C][esp_adf_pipeline.microphone:020]: ADF-Microphone
[23:41:21][C][adf_media_player:016]: ESP-ADF-MediaPlayer:
[23:41:21][C][adf_media_player:018]:   Number of ASPComponents: 3

[... 192 lines cut ...]
[23:41:39][D][esp-idf:000][decoder]: I (23920) AUDIO_ELEMENT: IN-[decoder] AEL_IO_DONE,-2

[23:41:40][D][esp-idf:000][decoder]: I (24281) MP3_DECODER: Closed

[23:41:40][D][esp-idf:000][resampler]: I (24387) AUDIO_ELEMENT: IN-[resampler] AEL_IO_DONE,-2

[23:41:40][D][esp-idf:000][i2s_out]: I (24451) AUDIO_ELEMENT: IN-[i2s_out] AEL_IO_DONE,-2

[23:41:40][D][esp_adf_pipeline:302]: State changed from STOPPING to STOPPED
[23:41:40][I][adf_media_player:135]: got new pipeline state: 5
[23:41:43][D][micro_wake_word:162]: The 'okay nabu' model sliding average probability is 0.996 and most recent probability is 1.000
[23:41:43][D][micro_wake_word:123]: Wake Word 'okay nabu' Detected
[23:41:43][D][micro_wake_word:195]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[23:41:43][D][micro_wake_word:129]: Stopping Microphone
[23:41:43][D][esp_adf_pipeline:302]: State changed from RUNNING to STOPPING
[23:41:43][D][micro_wake_word:195]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[23:41:43][D][esp_adf_pipeline:302]: State changed from STOPPING to STOPPED
[23:41:43][D][micro_wake_word:195]: State changed from STOPPING_MICROPHONE to IDLE
[23:41:43][D][light:036]: 'SL - On board light' Setting:
[23:41:43][D][light:047]:   State: ON
[23:41:43][D][light:051]:   Brightness: 60%
[23:41:43][D][light:059]:   Red: 43%, Green: 43%, Blue: 100%
[23:41:43][D][light:109]:   Effect: 'Fast Pulse'

```

Xe138, 2024-10-12:
I'm having the same issue.  I believe it is related to the audio output not working.  Are you getting any audio feedback when the first command is completed?  It seems like the replying phase is failing and the device never returns to the listening phase.

For testing I removed the voice assistant module and just used it as a speaker for media playback from home assistant, which is not working either.  I believe it has to do with the following errors:

```
[15:46:43][D][esp-idf:000][decoder]: W (1985143) AUDIO_ELEMENT: IN-[decoder] AEL_IO_ABORT

[15:46:43][D][esp-idf:000][decoder]: W (1985147) AUDIO_ELEMENT: OUT-[decoder] AEL_IO_ABORT

[15:46:43][D][esp-idf:000][decoder]: W (1985150) MP3_DECODER: output aborted -3

[15:46:43][D][esp-idf:000][decoder]: I (1985170) MP3_DECODER: Closed
```

lterfloth, 2024-10-20:
I do have the same problem and unable to find a solution. Would be happy if someone can point in the right direction.

plplaaa2, 2024-10-26:
[image]

I have the same problem. The voice assistant operates only once and the assistant does not return from responding to idle after that. As a result of the automation test, even if you order a voice assistant stop immediately after TTS, it does not return from responding to idle. I think it is a problem that will be solved when the assistant satellite goes back to idle.

AnaCoda, 2025-03-29:
I also have this problem and I am using ESPHome 2025.2.2, with Arduino framework. Doesn't seem to work at all (no voice data) when using ESP-IDF framework

CataCluj, 2025-03-29:
I also had this problem, for a long time, after some update. Very annoying, as I couldn't find why for a long time.
It seems (but I have no idea if that's true) that before the said update, "voice_assistant.stop" allowed the pipeline to finish, but after the update it would stop it before it had a chance to do anything.
I may have found a solution, at least for my situation:
You have:
`    on_release:
      - delay: 200ms
      - voice_assistant.stop:
`
My first "fix" was to have a 7s delay instead of 200ms. That might be enough for most cases, but I don't like hard coded delays, so I tried the following:
`    on_release:
#      - delay: 7s  # Add a small delay to ensure audio processing is complete
      - wait_until:
          not:
            voice_assistant.is_running:
      - voice_assistant.stop
`
Note that this is for PTT (Push To Talk) as I'm not interested in wake words.

v12345vtm, 2025-04-06:
Wont even compile  : i have this error 

Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/c6/.pioenvs/c6/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:309 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)