esphome/issues #6142: Bought a DHTC12 sensor. Similair to DHT12 sensor but different i2c bus adress? New component?

AussiSG, 2024-08-11:
### The problem

So I bought a DHTC12 sensor from Aliexpress.
It looks like a DHT12 but it reports it self under a different I2C adress. It states that it finds address 0x44.
Which should be suitable for the `sht3xd component` but when using that, the values reported back are off the chart.

130 degrees celsius and 100% humidity.

I got this in the logs when using the sht3xd component:

```
[21:44:56][I][i2c.arduino:096]: Results from i2c bus scan:
[21:44:56][I][i2c.arduino:102]: Found i2c device at address 0x23
[21:44:56][I][i2c.arduino:102]: Found i2c device at address 0x44
[21:44:56][I][i2c.arduino:102]: Found i2c device at address 0x53
[21:44:56][I][i2c.arduino:102]: Found i2c device at address 0x76
```

```
[21:44:56][C][sht3xd:049]: SHT3xD:
[21:44:56][D][sht3xd:064]:   Setup successful
[21:44:56][D][sht3xd:065]:   Serial Number: 0xFFFFFFFF
[21:44:56][D][sht3xd:066]:   Heater Enabled: false
[21:44:56][C][sht3xd:068]:   Address: 0x44
[21:44:56][C][sht3xd:069]:   Update Interval: 30.0s
[21:44:56][C][sht3xd:071]:   Temperature 'Achtertuin interne temperatuur 2'
[21:44:56][C][sht3xd:071]:     Device Class: 'temperature'
[21:44:56][C][sht3xd:071]:     State Class: 'measurement'
[21:44:56][C][sht3xd:071]:     Unit of Measurement: '°C'
[21:44:56][C][sht3xd:071]:     Accuracy Decimals: 1
[21:44:56][C][sht3xd:072]:   Humidity 'Achtertuin Luchtvochtigheid'
[21:44:56][C][sht3xd:072]:     Device Class: 'humidity'
[21:44:56][C][sht3xd:072]:     State Class: 'measurement'
[21:44:56][C][sht3xd:072]:     Unit of Measurement: '%'
[21:44:56][C][sht3xd:072]:     Accuracy Decimals: 1
```

```
[21:45:09][D][sht3xd:097]: Got temperature=130.00°C humidity=100.00%
[21:45:09][D][sensor:094]: 'Achtertuin interne temperatuur 2': Sending state 130.00000 °C with 1 decimals of accuracy
[21:45:09][D][sensor:094]: 'Achtertuin Luchtvochtigheid': Sending state 100.00000 % with 1 decimals of accuracy
```

### Which version of ESPHome has the issue?

2024.7.2 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.4

### What platform are you using?

ESP32

### Board

esp32-wroom ( full board )

### Component causing the issue

i2c

ssieb, 2024-08-11:
If it's compatible with the DHT12, then use that component and just set the address.

AussiSG, 2024-08-11:
> If it's compatible with the DHT12, then use that component and just set the address.

I didn't know that was possible.
I will investigate this tomorrow.

Thanks for pointing this out !

AussiSG, 2024-08-12:
Sadly enough that does not seem to do the trick:
```
#Temperature and humidity sensor DHTC12
  - platform: dht12
    address: 0x44
    temperature:
      name: ${device_name} interne temperatuur 2
    humidity:
      name: ${device_name} Luchtvochtigheid
    update_interval: 30s
```

```
[17:31:11][D][dht12:045]: DHT12:
[17:31:11][C][dht12:046]:   Address: 0x44
[17:31:11][E][dht12:048]: Communication with DHT12 failed!
[17:31:11][C][dht12:050]:   Temperature 'Achtertuin interne temperatuur 2'
[17:31:11][C][dht12:050]:     Device Class: 'temperature'
[17:31:11][C][dht12:050]:     State Class: 'measurement'
[17:31:11][C][dht12:050]:     Unit of Measurement: '°C'
[17:31:11][C][dht12:050]:     Accuracy Decimals: 1
[17:31:11][C][dht12:051]:   Humidity 'Achtertuin Luchtvochtigheid'
[17:31:11][C][dht12:051]:     Device Class: 'humidity'
[17:31:11][C][dht12:051]:     State Class: 'measurement'
[17:31:11][C][dht12:051]:     Unit of Measurement: '%'
[17:31:11][C][dht12:051]:     Accuracy Decimals: 1
[17:31:11][E][component:082]:   Component dht12.sensor is marked FAILED
```

And when using the SHT3xD component it seems to be setting up / starting up but with wrong values
```
[21:44:56][C][sht3xd:049]: SHT3xD:
[21:44:56][D][sht3xd:064]:   Setup successful
```

randybb, 2024-08-12:
Some sellers are mentioning that it is a replacement for "DHT20AM2320" - maybe platform: am2320 or dht12?

AussiSG, 2024-08-12:
Tried both and they both failed to setup communication. 
Or am doing something wrong with forcing the i2c address?

randybb, 2024-08-12:
Nope, just there is no datasheet for that part. You can ask the supplier to provide it and then we can do something. Without it it would be just trial and error... Just get something that is supported.

AussiSG, 2024-08-12:
Yes , I agree with you on this :) Will contact the seller right away. Curious if there is any data of this typical thing.

randybb, 2024-08-12:
Based on the SHT3xD datasheet it is not only simply reading data from a register, some init is needed to start measurement and then the data can be read from a register with crc. So I guess it is pretty similar or the component is not validating data.
Lets wait for the datasheet :)

AussiSG, 2024-08-15:
Yes after some chatting around with the seller he sended me some file.
Which includes a chinese data sheet hehe. I tried to translate it via a online program.
And also a zip file which contains programming files.

[黑色DHTC12.zh-CN.en.pdf](https://github.com/user-attachments/files/16628534/DHTC12.zh-CN.en.pdf)

[黑色DHTC12.pdf](https://github.com/user-attachments/files/16628485/DHTC12.pdf)

[DHTC12_IIC_OneWire210816.zip](https://github.com/user-attachments/files/16628469/DHTC12_IIC_OneWire210816.zip)

AussiSG, 2024-08-26:
No one had a chance to peak at the above data ?