esphome/issues #6144: IPv6 can't be enabled for libretiny

dwmw2, 2024-08-12:
### The problem

```
Failed config

network: [source /home/dwmw/esphome/base.yaml:74]
  
  This feature is only available on ['esp32', 'esp8266', 'rp2040'].
  enable_ipv6: True
```
I have IPv6 enabled in my base config; I can turn it off for specific devices by overriding it to `False` again if I have to, but even *that* doesn't work here as it refuses even to parse the default configuration.

But aside from the premature failure, *why* doesn't it work with libretiny? Looking at libretiny documentation, I see hints that there is IPv6 support: https://docs.libretiny.eu/ltapi/class_wi_fi_class/#function-enableipv6

Is this specific to BK72xx rather than libretiny itself? https://github.com/esphome/esphome/pull/6387 seemed to block this configuration without any link to an upstream bug or feature request.

### Which version of ESPHome has the issue?

dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

BK72XX

### Board

Tongou DIN rail power meter

### Component causing the issue

network

### Example YAML snippet

```yaml
network:
  enable_ipv6: true
```

dwmw2, 2024-08-12:
@HeMan do you have any context? Thanks.

HeMan, 2024-08-12:
> @HeMan do you have any context? Thanks.

I wasn't able to get IPv6 to work with Libretiny, that's why it is blocked. I could have another go now.

dwmw2, 2024-08-12:
Thanks. I may also change it to error if IPv6 is enabled at the *end* of parsing the configuration.

If it doesn't work, having a bug filed against libretiny would be helpful.

jesserockz, 2024-08-13:
This is not an issue/bug, this would be a feature request.

dwmw2, 2024-08-13:
I consider at least the part where it refuses to parse a configuration which *ends* with IPv6 disabled, to be a bug. But no need to debate that point; I'll submit a PR for that.

closed 2024-08-16 by jesserockz

dwmw2, 2024-08-16:
Feature request for actually it *work*, with early attempts at actually doing so: 
https://github.com/esphome/feature-requests/issues/2853