esphome/issues #6145: docker build does not include mdns but relies on the host

makuser, 2024-08-12:
### The problem

The docker build is not able to resolve .local addresses, without setting `network_mode: host`.

Including the zeroconfig/mdns resolver (eg. avahi) in the docker build solves this and then does not require but also not interfere with `network_mode: host`.

If we just include that (eg as per https://github.com/esphome/issues/issues/3517#issuecomment-1531212602), in the Dockerfile, that would be awesome and no longer require `network_mode: host`.

### Which version of ESPHome has the issue?

2024.XX

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

makuser, 2024-08-12:
Related: #3517

domoritz, 2025-01-03:
Having avahi would be great as it might help resolve issues with resolving local addresses better than it does right now.