esphome/issues #6149: Swing not working with Gree climate_ir component with YAC1FB9

enheragu, 2024-08-13:
### The problem

The component is able to change both temperature and fan speed. That works correctly. 
But the swing is blocked and does nothing. 

The AC has many options that can be set using its remote for the vertical swing (not horizontal).  When controlled with the climate_ir the vertical swing is just blocked in the middle.

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.1

### What platform are you using?

ESP8266

### Board

nodemcu CH340 v3

### Component causing the issue

climate

### Example YAML snippet

```yaml
remote_transmitter:
  id: ac_salon_transmitter
  pin: D1
  carrier_duty_percent: 50%

climate:
  # Works all but not swinging
  - platform: gree
    model: yac1fb9
    supports_cool: true
    supports_heat: true
    name: "Living Room AC"
    sensor: temperature_sensor
    transmitter_id: ac_salon_transmitter
```

### Additional information

In case it's useful, in the HA integration I'm making use of the  thermostat card
[image]

As said both temperature and fan control works, its the swing button that does nothing.
```yaml
- type: thermostat
    features:
      - style: icons
        type: climate-hvac-modes
        hvac_modes:
          - 'off'
          - cool
      - style: dropdown
        type: climate-swing-modes
        swing_modes:
          - 'off'
          - vertical
      - style: dropdown
        type: climate-fan-modes
        fan_modes:
          - auto
          - low
          - medium
          - high
    entity: climate.cortina_salon_living_room_ac
    name: ' '
    show_current_as_primary: false
```

sarthurdev, 2024-09-06:
Noted a checksum issue in my PR https://github.com/esphome/esphome/pull/7418

Possibly related to your bug.

dantesbr, 2024-12-06:
This PR https://github.com/esphome/esphome/pull/7418 worked with my Gree YB1F2 remote control.