esphome/issues #6153: Please provide a ota migration path for the breaking change coming with 2024.6.0

sureal, 2024-08-14:
### The problem

Starting with 2024.6.0 you added a breaking change regarding the yaml configuration of oat (over the air updates)
I'm currently stuck to 2024.5.5 as my old configuration was simply 

`ota: `

When I try to update the configuration to a state like
`ota:
  - platform: esphome`

while using esp home 2024.5.5 the (ota) install failes as it expects a dictionary instead.

When I try using 2024.6.0 or above I cannot update wirelessly at all (as my esp32 still has the old config without platform: esphome)

Hence one gets stuck in a chicken egg problem.

Could you please add instructions to the docs in order to have a wirelessly (!) working migration path so one can keep up to date (especially regarding security concerns)? That would be nice.

### Which version of ESPHome has the issue?

2024.6.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

ota

### Example YAML snippet

```yaml
ota:
  - platform: esphome
```

guillempages, 2024-08-19:
I'm not exactly sure what you are requesting here...

The migration path is actually pretty easy: you add the ` - platform: esphome` line in your `ota` section, and just do "esphome run xxx.yaml" (on the CLI; I'm not sure about the commands in the dashboerd; should work the same nevertheless).

The old configuration does not compile with esphome >= 2024.6, and the new one does not compile wit esphome < 2024.6, as you mention. but you the new binary can be perfectly updated over the air on an older device; it's just a "configuration" change at compile time, but there were no changes done on the OTA protocol iself, so nothing should break.

sureal, 2024-08-19:
Hi,
thanks for your answer. May be I should clarify my setup: I use ESPHome via Docker as part of my docker based home assistant instance. I was used to update my ESP32 device which has ESPHome installed using the web interface (http://192.168.zyx.xyz:6052/) . This way I ran into the before mentioned chicken egg problem.
I now realised that there is a config dir on my raspi keeping the yaml file. The solution for me (aka migration path) was to shut down esphome docker, edit the config/myfile.yaml adding  - platform: esphome line in ota, start esphome docker and click the update button in the web interface.  This way it seems to work (still compiling) Thank you very much for you impulse. Anyway I think other people may stumble over the same thing, so may be its worth extending the docs a bit.

closed 2024-08-24 by guillempages