esphome/issues #6158: IR Sender/Receiver crash loop

Miq1, 2024-08-15:
### The problem

I am building a fake remote for my VU+ satellite receiver. After adding some more buttons the ESP-01M device used is crashing repeatedly. Memory seems to be sufficient for updates over Serial. I am not aware of anything different in the YAML, so I attached both the complete YAML and logs below for your reference.

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

not relevant

### What platform are you using?

ESP8266

### Board

ESP-01M

### Example YAML snippet

```yaml
esphome:
  name: irremote2

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  platform: esphome
  password: ""

wifi:
  ssid: "FritzGandhi"
  password: "MahatmaNetzwerk"

remote_receiver:
  pin: GPIO14
  dump: all
  idle: 40ms

remote_transmitter:
  pin: GPIO04
  carrier_duty_percent: 50%

climate:
  - platform: fujitsu_general
    name: "Klima Wohnzimmer"
    supports_cool: true
    supports_heat: true
    visual:
      min_temperature: 18
      max_temperature: 25
      temperature_step: 0.5

button:
  - platform: template
    name: "VU Up"
[... 594 lines cut ...]
[12:29:35][C][climate_ir:077]:   Min. Temperature: 16.0°C
[12:29:35][C][climate_ir:078]:   Max. Temperature: 30.0°C
[12:29:35][C][climate_ir:079]:   Supports HEAT: YES
[12:29:35][C][climate_ir:080]:   Supports COOL: YES
[12:29:35][C][remote_receiver.esp8266:060]: Remote Receiver:
[12:29:35][C][remote_receiver.esp8266:061]:   Pin: GPIO14
[12:29:35][W][remote_receiver.esp8266:063]: Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True' in the pin schema!
[12:29:35][C][remote_receiver.esp8266:066]:   Buffer Size: 1000
[12:29:35][C][remote_receiver.esp8266:067]:   Tolerance: 25%
[12:29:35][C][remote_receiver.esp8266:069]:   Filter out pulses shorter than: 50 us
[12:29:35][C][remote_receiver.esp8266:070]:   Signal is done after 40000 us of no changes
[12:29:35][C][mdns:116]: mDNS:
[12:29:35][C][mdns:117]:   Hostname: irremote2
[12:29:35][C][esphome.ota:073]: Over-The-Air updates:
[12:29:35][C][esphome.ota:074]:   Address: irremote2.local:8266
[12:29:35][C][esphome.ota:075]:   Version: 2
[12:29:35][C][safe_mode:018]: Safe Mode:
[12:29:35][C][safe_mode:019]:   Boot considered successful after 60 seconds
[12:29:35][C][safe_mode:021]:   Invoke after 10 boot attempts
[12:29:35][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[12:29:35][C][api:139]: API Server:
[12:29:35][C][api:140]:   Address: irremote2.local:6053
[12:29:35][C][api:144]:   Using noise encryption: NO
```
```

Miq1, 2024-08-18:
I experimented a bit and found that removing some of the button templates will reduce the number of crashes, so it might be a memory issue. But stiil, the Pronto data blocks are known at compile time and are part of the statically allocated RAM (``RAM:   [=====     ]  54.7% (used 44844 bytes from 81920 bytes)``). 

Hence there must be something else using heap memory until it is exhausted.

ssieb, 2024-08-18:
`Unhandled C++ exception: OOM`, so it is definitely running out of RAM for some reason.  It appears to be crashing when HA connects.  Can you remove it from HA and add the `debug:` component with the heap sensors enabled to see what the state it?
But also, this is not an issue for github.  Can you please come ask about this on the esphome discord server?

closed 2024-08-18 by ssieb

Miq1, 2024-08-20:
I added the ``debug:`` component and it shows a mere 1168 bytes of free heap immediately upon startup:
```
[15:15:17][D][debug:035]: ESPHome version 2024.7.3
[15:15:17][D][debug:039]: Free Heap Size: 1168 bytes
[15:15:17][D][debug:041]: Flash Chip: Size=2048kB Speed=40MHz Mode=DOUT
[15:15:17][D][debug:050]: Chip ID: 0x004E3838
[15:15:17][D][debug:051]: SDK Version: 2.2.2-dev(38a443e)
[15:15:17][D][debug:052]: Core Version: 3.1.2
[15:15:17][D][debug:053]: Boot Version=7 Mode=1
[15:15:17][D][debug:054]: CPU Frequency: 80
[15:15:17][D][debug:055]: Flash Chip ID=0x001540A1
[15:15:17][D][debug:056]: Reset Reason: Exception
[15:15:17][D][debug:057]: Reset Info: Fatal exception:29 flag:2 (Exception) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[15:15:17][D][text_sensor:064]: 'Device Info': Sending state '2024.7.3|Flash: 2048kB Speed:40MHz Mode:DOUT|Chip: 0x004e3838|SDK: 2.2.2-dev(38a443e)|Core: 3.1.2|Boot: 7|Mode: 1|CPU: 80|Flash: 0x001540a1|Reset: Exception|Fatal exception:29 flag:2 (Exception) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x0'
[15:15:17]
[15:15:17]User exception (panic/abort/assert)
[15:15:17]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[15:15:17]
[15:15:17]Unhandled C++ exception: OOM
[15:15:17]
```
Is there any means to remove components or parts of components to decrease heap usage? Else I will have to switch hardware and build upon an ESP32 I am afraid.

ssieb, 2024-08-20:
You have a lot of things in there.  I would highly recommend using an ESP32 anyway.

Miq1, 2024-08-21:
Yeah. Did it, works. Thank you for your assistance!